probe-rs/probe-rs #466: WB55 seems to have a bug in the range of its flashing algorithm

rubdos, 2020-12-04:
**Describe the bug**

Testing out probe-rs on a WB55RG

`probe-rs`, just installed from crates.io:

```
       DEBUG probe_rs::flashing::download           > Found loadable segment.
       DEBUG probe_rs::flashing::download           > Found loadable segment.
       DEBUG probe_rs::flashing::download           > Found loadable segment.
        INFO probe_rs::flashing::download           > Found 3 loadable sections:
        INFO probe_rs::flashing::download           >     .vector_table at 08000000 (1024 byte0)
        INFO probe_rs::flashing::download           >     .text at 08000400 (1360 byte0)
        INFO probe_rs::flashing::download           >     .rodata at 08000950 (384 byte0)
       DEBUG probe_rs::flashing::loader             > Using builder for region (0x08000000..0x09000000)
       DEBUG probe_rs::flashing::loader             > Algorithm stm32wb_m4 - start: 0x8000000 - size: 0x100000
       DEBUG probe_rs::flashing::loader             > Algorithms: []
       DEBUG probe_rs::probe::stlink                > Read mem 32 bit, address=e0002000, length=4
SNIP
     Erasing sectors ⠁ [00:00:00] [#########################################]       0B/      0B @       0B/s (eta 0s )
       Error failed to flash /home/rsmet/src/waveshare-test/target/thumbv7em-none-eabi/debug/waveshare-test

Caused by:
    0: Error while flashing
        INFO cargo_flash                            > Metadata {
    chip: Some(
        "Unspecified(\"STM32WB55RGVx\")",
    ),
    probe: Some(
        "STLink",
    ),
    speed: Some(
        "1800",
    ),
    release: "0.10.2",
    commit: "crates.io",
}
     Erasing sectors ⠁ [00:00:00] [#########################################]       0B/      0B @       0B/s (eta 0s )
```

... given `memory.x`:

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 512K
  RAM : ORIGIN = 0x20000000, LENGTH = 128K
}
```

N.b., this chip is supposed to have 1M of RAM.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach a Nucleo WB55
2. `cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart`
3. Alter the `memory.x` as per above
4. `cargo flash --chip STM32WB55RGVx`

**Expected behavior**
It flashes fine

**Stacktrace**
n.a.

**Desktop (please complete the following information):**
Linux, Arch

**Additional context**

When applying this patch:

```diff
diff --git a/probe-rs/targets/STM32WB Series.yaml b/probe-rs/targets/STM32WB Series.yaml
index 4c36de8..5435bba 100644
--- a/probe-rs/targets/STM32WB Series.yaml
+++ b/probe-rs/targets/STM32WB Series.yaml
@@ -154,7 +154,7 @@ flash_algorithms:
     flash_properties:
       address_range:
         start: 134217728
-        end: 135266304
+        end: 150994944
       page_size: 1024
       erased_byte_value: 255
       program_page_timeout: 400
```

the Nucleo flashes fine. The difference between the two `end` values is 0xF00000, which makes me think that the original value was missing a trailing zero. I'm not inclined to submit it as a merge-request though, since I'm not knowledgeable enough to make this claim. @Tiwalun seems to have done the initial import, so pinging!

Tiwalun, 2020-12-04:
Thanks for your report!

The target descriptions are generated automatically from XML files provided by ARM, and it seems that they released a newer version, which fixes this issue.

The addresses for the algorithm are actually correct, but the flash address in the chip description was wrong: 
https://github.com/probe-rs/probe-rs/blob/e527946b5c0c723e837ccdf23c39afff475806d9/probe-rs/targets/STM32WB%20Series.yaml#L93 

This meant that probe-rs was unable to find the correct flash algorithm.

closed 2020-12-04 by bors[bot]

rubdos, 2020-12-04:
> The target descriptions are generated automatically from XML files provided by ARM, and it seems that they released a newer version, which fixes this issue.

Exactly why I didn't want to file a PR, you guys know better! Thanks :-)

rubdos, 2020-12-04:
Can confirm, this works!

```
        INFO probe_rs::flashing::download           > Found 3 loadable sections:
        INFO probe_rs::flashing::download           >     .vector_table at 08000000 (1024 byte0)
        INFO probe_rs::flashing::download           >     .text at 08000400 (7460 byte0)
        INFO probe_rs::flashing::download           >     .rodata at 08002124 (1580 byte0)
       DEBUG probe_rs::flashing::loader             > Using builder for region (0x08000000..0x08100000)
       DEBUG probe_rs::flashing::loader             > Algorithm stm32wb_m4 - start: 0x8000000 - size: 0x100000
       DEBUG probe_rs::flashing::loader             > Algorithms: [RawFlashAlgorithm { name: "stm32wb_m4",
```

stappersg, 2020-12-05:
On Fri, Dec 04, 2020 at 01:02:19PM -0800, Ruben De Smet wrote:
> Can confirm, this works!

Thanks for the feedback.

commit de3277f2 (2020-12-04): Update target description for STM32WB55 (fix #466)

The flash address for some chips were wrong, which
meant that probe-rs was unable to find the correct
flash algorithm.
commit 416b7485 (2020-12-04): Merge #467

467: Update target description for STM32WB55 (fix #466) r=Yatekii a=Tiwalun

The flash address for some chips were wrong, which
meant that probe-rs was unable to find the correct
flash algorithm.