probe-rs/probe-rs #470: DAP read timeout on nRF9160

macthestack, 2020-12-06:
Cannot use `cargo embed --release --target thumbv8m.main-none-eabihf`. Background: #404 (cannot re-open)

I'm using cargo-embed:6728e8a > probe-rs:d65f31d

```
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 2690580481.
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 8c, 2, 9, 0, 80, 91, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 587202578.
       DEBUG probe_rs::flashing::flasher                                  > Waiting for routine call completion.
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 8c, 2, 9, 0, 80, 91, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 587202578.
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=TAR { address: e000edf0 }
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 2c, 2, f8, 76, 0, f0, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 3758157296.
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, 7c, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, 24, 40, 0, 0, 64, 0]
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, e8, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP WAIT, retries remaining 5.
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, 8, 0, 0, 80, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 16.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, e8, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP WAIT, retries remaining 4.
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, 8, 0, 0, 80, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 16.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, e8, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP WAIT, retries remaining 3.
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, 8, 0, 0, 80, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 16.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, e8, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP WAIT, retries remaining 2.
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, 8, 0, 0, 80, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 16.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, e8, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP WAIT, retries remaining 1.
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, 8, 0, 0, 80, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 64, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 16.
       ERROR probe_rs::probe::jlink                                       > DAP read timeout.
     Erasing sectors ✔ [00:00:01] [--------------------------------------------------------]       0B/132.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:01] [--------------------------------------------------------]       0B/132.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/martin/repos/viva/edge/beebox/target/thumbv8m.main-none-eabihf/release/beebox

             Caused by:
                 0: Error while flashing
                 1: Something during the interaction with the core went wrong
                 2: A core architecture specific error occured
                 3: Failed to read register DRW at address 0x0000000c
                 4: Operation timed out
```

macthestack, 2020-12-22:
Finally got it to work with latest embed from master but only with speeds below 10kHz and still, it's very unpredictable.

When it fails it fails predictably - in the beginning - leading me to believe it could support higher speeds if we can get past whatever it gets stuck on.

Tiwalun, 2020-12-22:
Good to hear!

Could you please post the full debug output again? If possible, both from a working case as well as from a failing case.

macthestack, 2020-12-22:
Certainly.

[log - fail.txt](https://github.com/probe-rs/probe-rs/files/5730192/log.-.fail.txt)
[log - success.txt](https://github.com/probe-rs/probe-rs/files/5730193/log.-.success.txt)

Tiwalun, 2020-12-22:
Hmm, I don't see any obvious problems. The detection of APs in the beginning looks a bit weird, but I don't think that's an issue.

Could you please set the debug level to trace (i.e. `RUST_LOG=trace`) and post that output?

macthestack, 2021-01-02:
Sorry for the delay. Here's the output (from a different machine/os but same behaviour):

[log - fail.txt](https://github.com/probe-rs/probe-rs/files/5760529/log.-.fail.txt)
[log - success.txt](https://github.com/probe-rs/probe-rs/files/5760530/log.-.success.txt)

macthestack, 2021-01-02:
I noticed that I only get the first 600 lines when piping to a file. (strangely I'm getting this both on linux and mac in different terminals)

I have copied the outputs directly from the terminal. The two files are both failing but differently it seems.

[log6.txt](https://github.com/probe-rs/probe-rs/files/5760595/log6.txt)
[log7.txt](https://github.com/probe-rs/probe-rs/files/5760596/log7.txt)

macthestack, 2021-01-03:
By inserting a ```std::thread::sleep(Duration::from_millis(100));``` at line 528 in the flasher.rs (right after ```self.call_function(registers, init)?;```). I have finally managed to get it to reliably work. I don't know why it doesn't wait for completion but maybe you guys have an idea. It works for me for now which is great but if I get more time I can try to contribute.

Dirbaio, 2021-08-20:
@macthestack could you confirm whether this is still an issue? there's been quite a bit of jlink fixes, and some nrf91-specific fixes in #619 too.

macthestack, 2021-08-21:
It works. Thanks guys!

closed 2021-08-21 by macthestack