probe-rs/probe-rs #473: Offer openocd's flash padded_value num value

jacobrosenthal, 2020-12-07:
In programming a nordic softdevice with openocd, then cargo flashing a normal rust project(avoiding softdevice) and everything works great.

I have a [patch](https://github.com/probe-rs/cargo-flash/pull/149) for precompiled hex support so I can program the softdevice from cargo flash `cargo flash --hex s140_nrf52_7.2.0_softdevice.hex`
Then again cargo flash as normal (avoiding the softdevice) and it wont load.

The only binary difference openocd says are different on device after programming is there are 2 padded regions, one in the center, and one at the end. openocd leaves both of these bytes as 0xff, where cargo flash sets them all to 0x00

[image]

:100AE0007703000000000020F00A0000040000006E
:100AF000001000000000000000FFFFFF0090D00386 D003 here is the other last byte before gap
:10100000C0130020E15F0200A1C10000475F0200A1

normal address diff between lines 
0AF-0AE = 0x10
1000-0AF = 0x510 the gap line
Yeah so thats a 'gap' in the hex

[image]

And last ihex is
:046630003C11C40055
so last data is C4 00 in both

So that differs at 157236 or 26634 I think. the start of the application is 27000
So thats padding after the end of the hex

It turns out openocd has a command for this
> Command: flash padded_value num value
> Sets the default value used for padding any image sections, This should normally match the flash bank erased value. If not specified by this comamnd or the flash driver then it defaults to 0xff. 
http://openocd.org/doc/html/Flash-Commands.html

ill set that to 0 in openocd to match cargo flash to see if that would 'break' as well
`openocd -f interface/cmsis-dap.cfg -f target/nrf52.cfg -c "init; reset halt; nrf51 mass_erase; flash padded_value 0 0x00; program s140_nrf52_7.2.0_softdevice.hex verify exit"`
And yes it does fail, though oddly it only pads the 'center', but not the 157236 location so its not a perfect test

But it certainly seems like openocd prefers leaving/writing open bytes and bytes padding out the end of a sector or whatever to 0xff and the softdevice needs one or both of those to be true for some reason

adamgreig, 2020-12-07:
Instead of making it a configuration option, would it be simpler to just make sure we always fill with FF instead? It seems like that should be pretty well universally more useful.

Yatekii, 2020-12-07:
Yes, I agree with that sentiment. Also it merely requires changing the default value we currently set, I believe.

Tiwalun, 2020-12-07:
The target definitions also contain the default value for the flash, for example:

https://github.com/probe-rs/probe-rs/blob/4f2ef53b33548b59670469502288e4f8330ee9c8/probe-rs/targets/HT32F0006%20Series.yaml#L88

commit e8368057 (2020-12-07): Merge #475

475: default FlashPage value to chips erased_byte_value r=Tiwalun a=jacobrosenthal

Closes #473
PR #475 (2020-12-07): default FlashPage value to chips erased_byte_value
Closes #473
closed 2020-12-07 by bors[bot]