probe-rs/probe-rs #474: Is "flashing" to RAM supposed to work?

taruti, 2020-12-07:
Hi,

Probe-rs (via probe-run) seems unable to "flash" an ELF image to RAM. Is this supposed to work via the flashing API (that high-level tools use), or should the memory writing API be used?

If I wanted to patch this to work - would adding support for this to the flashing API be acceptable? (I know ram is not flash). Or is this supposed to work and I'm just stumbling upon user error/bug.

```
$ probe-run --chip nrf52 target/thumbv7em-none-eabihf/release/hello
  (HOST) INFO  flashing program (1.40 KiB)
Error: Error while flashing

Caused by:
    No flash memory contains the entire requested memory range 0x20000000..0x20000100.
```

The openocd workflow I was trying to replicate was:
```
...
reset init
load_image $bin 0 elf 0x20000000 262144
resume 0x20000000
```

I have memory.x setup and things work with openocd+gdb.

Tiwalun, 2020-12-07:
The flash API currently only supports actual flash, so using it to load data into RAM is expected to fail right now. 

What's your use case for this? Do you want to load code into flash and RAM at the same time, or only into RAM?

I think extending the flash API to cover this case would be fine, as long as it doesn't cause issues with the "flash only" loading.

taruti, 2020-12-07:
Thank you for the answers!

I have been using a testing workflow with gdb+openocd that is purely in RAM and have been tinkering with switching to using more rust tooling.

I'll tinker with the code and make a PR if all goes well.

commit bee9913a (2020-12-11): Merge #480

480: Support "flashing" program data to RAM r=Tiwalun a=taruti

Fixes #474 

Due to probe-rs-rtt vs probe-rs git versioning testing with actual hardware hard (probe-run has unrelated issues compiling).
PR #480 (2020-12-11): Support "flashing" program data to RAM
Fixes #474 

Due to probe-rs-rtt vs probe-rs git versioning testing with actual hardware hard (probe-run has unrelated issues compiling).
closed 2020-12-11 by bors[bot]