probe-rs/probe-rs #490: Connect under reset not working with stlinkv2 + nrf52840

Dirbaio, 2020-12-14:
```
     Running `probe-run --chip nRF52840_xxAA --defmt --input target/thumbv7em-none-eabi/release/device_gwc --probe '0483:3748:C2xxxx' --connect-under-reset`
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out
```

Trace logs: https://gist.github.com/Dirbaio/455d98823d19066bffaa1c1028a8de26

probe-run using probe-rs v0.10.1

Dirbaio, 2021-08-20:
This is working as intended: nRF chips don't do stuff while nRST is asserted, they're not capable of any SWD communication at all. It seems `connect-under-reset` is quite ST-specific.

closed 2021-08-20 by Dirbaio

Yatekii, 2021-08-20:
'connect-under-reset' is generally required for chips that support remapping reset and swd pins. its also present on the atsam devices. Just to clarify this. But I agree that this works as intended and can be closed.