probe-rs/probe-rs #496: Weird error message when a target is not connected or unpowered

korken89, 2020-12-15:
**Describe the bug**
Probe-rs generates the following error when a target is not connected (or un-powered):

```
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to the selected architecture occured
    1: An error occured in the SWD communication between DAPlink and device.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Have no target connected
2. Use a download command eg `download_file_with_options`

**Expected behavior**
An error message that says that the target cannot be found.

**Desktop (please complete the following information):**
 - Arch Linux

Yatekii, 2024-04-27:
Hmm I am aware of this message, but reading it with the 2024 me, the message seems correct 🤔

noppej, 2024-04-27:
Definitely correct from an accuracy perspective. Is it helpful from a UX perspective?

BartMassey, 2024-06-02:
A more specific "cannot find a device" error message would definitely be more helpful, if possible.