probe-rs/probe-rs #507: Unable to flash SAM chip with bootloader

henrikssn, 2020-12-23:
**Describe the bug**
When trying to flash a Feather M0 (SAMD21G18AU) with 8K bootloader, the command fails (see log output below).

**To Reproduce**
Steps to reproduce the behavior:
1. `cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart`
2. Edit `memory.x` to make space for bootloader:
```
MEMORY
{
  FLASH : ORIGIN = 0x00000000 + 8K, LENGTH = 256K - 8K
  RAM : ORIGIN = 0x20000000, LENGTH = 64K
}
```
3. Edit `.cargo/config`:
```
[build]
target = "thumbv6m-none-eabi" 
```
4. Run `cargo flash --chip atsamd21g18au`

**Expected behavior**
The program is flashed to the MCU.

**Stacktrace**
```
DEBUG probe_rs_cli_util > Running '/home/erik/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo' in directory /code/bug-reproduce-samd-bootprot
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
    Flashing /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot
       DEBUG jaylink           > libusb 1.0.22.11312
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: true
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: true
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 001: ID 1d6b:0003
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 005: ID 1366:0101
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 007: ID 0483:3748
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 006: ID 0403:6001
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 002: ID 8087:0a2b
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 001: ID 1d6b:0002
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
[... 510 lines cut ...]
       DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000EDF0
       DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x02000001
       DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000EDF0
       DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x02000001
     Erasing sectors ✔ [00:00:00] [------------------------------------------------------------------------]       0B/ 16.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [------------------------------------------------------------------------]       0B/  2.00KB @       0B/s (eta 0s )
       Error failed to flash /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot

Caused by:
    0: Error while flashing
    1: Something during the interaction with the core went wrong
    2: An error with the usage of the probe occured
    3: Operation timed out
```
**Desktop (please complete the following information):**
 - Linux (Debian Buster)

**Additional context**
* The board has a 8K bootloader which I am trying to preserve.
* The first flash pages are protected by BOOTPROT.
* Flashing the board directly with JFlash starting at address `0x2000` works fine, so does `bossac`.
* I am trying to use `probe-rs` instead of the bootloader due to the RTT integration, but would like to keep the option of later flashing with bootloader.

Yatekii, 2020-12-23:
Ohhh I think I know what's the issue! Could you try and pass `--nmagic` to the linker?

henrikssn, 2020-12-23:
`--nmagic` was already set. As suggested in the Matrix room I added the `--connect-under-reset` flag to cargo flash which gave the following error message:

```
$ cargo flash --chip atsamd21g18au --log debug
       DEBUG probe_rs_cli_util > Running '/home/erik/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo' in directory /code/bug-reproduce-samd-bootprot
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
    Flashing /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot
       DEBUG jaylink           > libusb 1.0.22.11312
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: true
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: true
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 001: ID 1d6b:0003
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 011: ID 1366:0101
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 007: ID 0483:3748
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 010: ID 0403:6001
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 002: ID 8087:0a2b
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 001: ID 1d6b:0002
       DEBUG probe_rs::probe::daplink::tools > Device open error Access
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       DEBUG jaylink                                > legacy caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG jaylink                                > extended caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO cargo_flash                            > Protocol speed 0 kHz
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 801020640
       DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
[... 474 lines cut ...]
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 1, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=TAR { address: e000edf0 }
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0xDRW { data: 1030003 }
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 1, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=TAR { address: e000edf4 }
[... cut ...]
 Programming pages   ✔ [00:00:00] [------------------------------------------------------------------------]       0B/  2.00KB @       0B/s (eta 0s )
       Error failed to flash /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot

Caused by:
    0: Error while flashing
    1: The execution of 'erase_sector' failed with code 1. Perhaps your chip has write protected sectors that need to be cleared? Perhaps you need the --nmagic linker arg https://github.com/rust-embedded/cortex-m-quickstart/pull/95
```

henrikssn, 2020-12-23:
I cloned probe-rs and cargo-embed, and edited the `SAM D.yaml` flash properties to set sector size to 2K (IIUC, it should really be 256b but page size is already set to 1K).

New error message:
```
cargo flash --probe 1366:0101 --chip atsamd21g18au --log info
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
    Flashing /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot
        INFO cargo_flash > Protocol speed 0 kHz
        INFO probe_rs::probe::jlink > J-Link: S/N: 801020640
        INFO probe_rs::probe::jlink > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
        INFO probe_rs::probe::jlink > J-Link: Hardware version: JLink 1.0.0
        INFO probe_rs::probe::jlink > J-Link: Target voltage: 3.30 V
        INFO probe_rs::flashing::download > Found loadable segment, address: 0x00002000
        INFO probe_rs::flashing::download > Matching section: ".vector_table"
        INFO probe_rs::flashing::download > Found loadable segment, address: 0x000020c0
        INFO probe_rs::flashing::download > Matching section: ".text"
        INFO probe_rs::flashing::download > Found loadable segment, address: 0x00002480
        INFO probe_rs::flashing::download > Matching section: ".rodata"
        INFO probe_rs::flashing::download > Found 3 loadable sections:
        INFO probe_rs::flashing::download >     .vector_table at 00002000 (192 byte0)
        INFO probe_rs::flashing::download >     .text at 000020C0 (960 byte0)
        INFO probe_rs::flashing::download >     .rodata at 00002480 (376 byte0)
     Erasing sectors ⠁ [00:00:00] [########################################################################]       0B/      0B @       0B/s (eta 0s )
       Error failed to flash /code/bug-reproduce-samd-bootprot/target/thumbv6m-none-eabi/debug/bug-reproduce-samd-bootprot

Caused by:
    0: Error while flashing
        INFO cargo_flash                  > Metadata {
    chip: Some(
        "Unspecified(\"atsamd21g18au\")",
    ),
    probe: None,
    speed: Some(
        "0",
    ),
    release: "0.10.2",
    commit: "b0a02d9-modified",
}
     Erasing sectors ⠁ [00:00:00] [########################################################################]       0B/      0B @       0B/s (eta 0s )
```

LongLiveCHIEF, 2021-04-24:
I was reviewing @henrikssn 's updates via #542, and testing them against `atsamd21g18au`, and have come up with the following error:

```
Target /home/chief/code/led-backlighting-controller/target/thumbv6m-none-eabi/release/led-backlighting-controller
        WARN probe_rs::flashing::flasher > Chip erase was the selected method to erase the sectors but this chip does not support chip erases (yet).
        WARN probe_rs::flashing::flasher > A manual sector erase will be performed.
     Reading flash   ✔ [00:01:52] [###########################################################################################################################################]  21.35KB/ 21.35KB @     193B/s (eta 0s )
     Erasing sectors ✔ [00:00:00] [-------------------------------------------------------------------------------------------------------------------------------------------]       0B/ 32.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:01:53] [-------------------------------------------------------------------------------------------------------------------------------------------]       0B/ 32.00KB @       0B/s (eta 0s )
       Error failed to flash /home/chief/code/led-backlighting-controller/target/thumbv6m-none-eabi/release/led-backlighting-controller
             
             Caused by:
                 0: Error while flashing
                 1: Failed to erase flash sector at address 0x00000000.
                 2: The execution of 'erase_sector' failed with code 1. This might indicate a problem with the flash algorithm.
```

I took a look, and I think this may be caused by invalid `flash_properties` in the `atsamd21_128` algo at https://github.com/probe-rs/probe-rs/blob/08d9fa2a1ae42d7c2ac230bf4e6c8c8adadeecf4/probe-rs/targets/SAMD21.yaml#L589-L610

LongLiveCHIEF, 2021-04-24:
so, strangely enough, even though I'm _fairly_ sure my chip is a `atsamd21g18au`, if I change my chip in `.embed.toml` and `probe-run` commands to `atsamd21e17a`, everything works flawlessly... but only if I use my stlinkv2-1 probe (jlink edu mini still has issues).

I am currently testing on a seeeduino xiao, which the manufacturer states is a "ARM Cortex-M0+ CPU(SAMD21G18)" ([seeeduino xiao specs](https://wiki.seeedstudio.com/Seeeduino-XIAO/#specification).

The actual chip on the xiao board is shielded, so I cant verify the chip without destroying the board, but I have some new boards coming in a few days that supposedly also have the atsamd21g18au (and not shielded), and I'll re-run some of these tests with those boards as well to see what results I can replicate.

If things go strangely at that point, I'll rip the shielding off one of these seeeduino's to verify the chip.

LongLiveCHIEF, 2021-04-25:
btw, the flash aglo is 2556 not 128, at least according to the CMSIS pack. (this matches what is currently in the targets spec of probe-rs for the chip.)

wjbuys, 2021-11-13:
I'm running into the same issue while trying to flash an ATSAMD21G18AU (Adafruit Feather M0 Express); probe-run tries to erase flash at address 0x0000, even though `memory.x` reserves 8k for the bootloader and `--nmagic` is passed to the linker:

```
   Flashing atsamd/boards/feather_m0/target/thumbv6m-none-eabi/debug/examples/blinky_basic
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x00002000, virtual address: 0x00002000, flags: 0x4
        INFO probe_rs::flashing::download > Matching section: ".vector_table"
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x000020b0, virtual address: 0x000020b0, flags: 0x5
        INFO probe_rs::flashing::download > Matching section: ".text"
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x00005ba0, virtual address: 0x00005ba0, flags: 0x4
        INFO probe_rs::flashing::download > Matching section: ".rodata"
        INFO probe_rs::flashing::loader   > Found 3 loadable sections:
        INFO probe_rs::flashing::loader   >     .vector_table at 00002000 (176 bytes)
        INFO probe_rs::flashing::loader   >     .text at 000020B0 (15088 bytes)
        INFO probe_rs::flashing::loader   >     .rodata at 00005BA0 (1716 bytes)
        INFO cargo_flash                  > Protocol speed 0 kHz
        INFO probe_rs::probe::jlink       > J-Link: S/N: 801038796
        INFO probe_rs::probe::jlink       > J-Link: Firmware version: J-Link EDU Mini V1 compiled Nov  2 2021 11:12:01
        INFO probe_rs::probe::jlink       > J-Link: Hardware version: J-Link 1.0.0
        INFO probe_rs::probe::jlink       > J-Link: Target voltage: 3.30 V
        INFO probe_rs::flashing::flasher  > Erasing sector at address 0x00000000
        INFO probe_rs::flashing::flasher  > Done erasing sector. Result is 1. This took 15.732822ms
     Erasing sectors ✔ [00:00:00] [-------------------------------------------------------------------------------------------------------------]       0B/32.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [-------------------------------------------------------------------------------------------------------------]       0B/16.62KiB @       0B/s (eta 0s )
       Error Failed to erase flash sector at address 0x00000000.
```

This is trying to flash the unmodified blinky_basic example from [here](https://github.com/atsamd-rs/atsamd/tree/master/boards/feather_m0)

kaidokert, 2022-05-29:
Was also playing with Feather-M0 just now and passing `atsamd21e17a` made it magically work.

Yatekii, 2022-05-29:
So to clarify: You intentinally passed the wrong chip to make it work? Could you specify what Chip you have?

We should diff the config files against eachother and see what the differences are :)

LongLiveCHIEF, 2022-05-29:
@Yatekii this is only happening when the board uses the `ATSAMD21G18AU` chip.  I've tested on both the feather M0 and the seeeduino xiao.  The only way to flash the chips on these boards is to use the `atsamd21e17a`.

I did some digging a little over a year ago, and determined that it came down to the flash region size being incorrect for the CMSIS pack for the 21G18au.  I compared the flash algo's for that chip and the 21e17a, and the only differences were the flash region size. The region size for the 21e17a seems to be right size for 21d18au, so that why I flashed it using the "wrong chip" so long ago. 

Since then I've seen a few dozen people with the same problem for these 2 chips, and this fix always works for them.

I think there's another issue thread in one of the probe-rs repos where I had an even more detailed description and analysis, but can't remember where that is.

Yatekii, 2022-05-29:
Hmm then we should fix the target description file :/ Or see if there is an updated CMSIS-Pack :/

LongLiveCHIEF, 2022-06-11:
If I get a chance, I'll look into whether or not the CMSIS-Pack has been updated this weekend. If not, I can submit a PR to update the target description file for the g18au.

ianrrees, 2022-06-13:
I went ahead and ran target-gen on the current SAMD21 [dev pack](https://github.com/Microchip-MPLAB-Harmony/dev_packs) - [here](https://github.com/probe-rs/probe-rs/compare/master...ianrrees:update-samd21?expand=1)- it looks like the tool now generates decimal rather than hex values, so the diff isn't particularly enlightening, but the spot checking I did doesn't indicate any changes including to the ATSAMD21G18AU flash algorithm.

The [ATSAMD HAL](https://github.com/atsamd-rs/atsamd) has made a PR or two to the dev packs before, but I'm not actually sure what exactly the error is.  Per the SAMD21 datasheet 22.6.3 "Region Lock Bits", the region size is determined by the total flash size - ATSAMD21G18AU has 256kB flash so should have 16kB region size, ATSAMD21E17A is 128kB/8kB.  Those look to be the same as the "sector" values in the generated yaml.

I do notice the page size for all these parts is 64B per the datasheet table 10-3, and the flash algos differ between these two parts: ATSAMD21G18AU uses flash algo atsamd21_256 and that uses 64B page size, The ATSAMD21E17A uses algo atsamd21_128, **1024B** page size.  However, I don't understand how that would relate to the bootloader specifically...

ianrrees, 2023-10-13:
After some time away from SAMD and Rust firmware, this has tripped me up again so I've dug in to the cause:

probe-rs has a concept of flash "sectors", and before programming it erases any sectors that will later be programmed.  The sector boundaries are calculated based on the flash algorithm, from SAMD21.yaml which is generated from the Microchip dev pack.  ATSAMD21G18AU sectors start at 0x0 with size 0x4000 (16kB), while ATSAMD21E17A sectors start at 0x0 and have size 0x2000 (8kB).

When probe-rs attempts to program a SAMD21G18AU that has an 8kB bootloader (as is typical for Adafruit boards, at least), it decides to erase the sector that goes from 0x0 to 0x4000-1 because the program starts halfway through that sector.  But, the chip has the bootloader protection enabled for the lower 8kB (this setting can be queried).

The naive fix of turning down the sector size in SAMD21.yaml fails, I presume because the loader can't handle the finer grained sector addresses.

@Yatekii do you have any thoughts on the best approach to resolve this?

Yatekii, 2023-10-13:
The erase always happens per sector. The only way is to erase the bootloader and write it again (there is a flag for it). Maybe the chip has a weird finer grained sector set that can be enabled with some commands; in that case we can fix it with a new flashloader. But thats a rather rare thing afaik. Bootloaders are typically the size of N sectors for this reason.

ianrrees, 2023-10-13:
From a look at the datasheet tonight, I believe the SAMD21 flash can be erased in units called "rows" which are 256 bytes, each row is four 64 byte "pages".  SAMD21 has two flash protection schemes which relate to this issue, one via BOOTPROT and another via "Region Lock Bits":

BOOTPROT is meant to protect a bootloader at the beginning of flash, it's a 3-bit value which maps on to the same set of sizes for all flash size parts, either 0B, 512B, 1024B, 2048B, ... 32768B.

The Region Lock Bits provide a more general-purpose protection scheme, which divides the flash in to 16 equally-sized regions and allows disabling writes per-region.   I believe these regions are what we're treating as "sectors".

Yatekii, 2023-10-14:
What if you try to make the sector size 256 bytes?

I mean best would be if there was a dynamic sector erase size in our flashing code, but for now we can do it with a smaller sector maybe. It will take much longer to erase like this tho.

ianrrees, 2023-10-16:
> What if you try to make the sector size 256 bytes?

Changing the sector size in SAMD21.yaml fails (that's what I did in [this comment](https://github.com/probe-rs/probe-rs/issues/507#issuecomment-1760839797)).  My guess is that the loader program doesn't handle finer grained erases, but that is just a guess - is that program provided by the pack?

I also tried setting the sector size to 0x800, which is the same used for ATSAMD21E15A - that fails similarly to 0x10, but setting the device to ATSAMD21E15A succeeds.

> I mean best would be if there was a dynamic sector erase size in our flashing code, but for now we can do it with a smaller sector maybe. It will take much longer to erase like this tho.

Understood; I'm happy to look at concatenating adjacent sectors if that would help.

ianrrees, 2023-10-21:
I've decompiled the `EraseSector` and `ProgramPage` functions for the loaders used for [ATSAMD21E15A](https://gist.github.com/ianrrees/86c298a58d4c53157eb6b312e5fa888f) and [ATSAMD21G18AU](https://gist.github.com/ianrrees/c01c777cbe2c1d33350305b4f891b811) (added comments to this `EraseSector`).  In short - `EraseSector` provided in the dev_packs works in terms of the regions (and unlocks the region lock bits accordingly), and internally does the erases per 256B row.  I don't see a more appropriate function to use for finer-grained erase.

It seems like there are a couple options to proceed:

Take the bootloader protection in to account, which would look something like
  1. Save existing bootloader protection value
  2. Disable bootloader protection
  3. Use existing flash algo to program (erasing the bootloader and writing it back, as necessary)
  4. Restore the bootloader protection value

Or, create a new erase function, that can work per-row.  Ideally, the new erase function would take a start address and size parameter, so we don't need to make separate transactions for each row erase - but that would require more extensive changes.

ianrrees, 2023-10-23:
After a bit more thought about this - there's an option in-between those two:

C) Modify the flash algorithm to backup, disable, and re-enable the bootloader protection.

I tend to think that A) is the best option, because it would give the user more visibility and control over the state of the chip.  Automatically turning off a protection feature seems a bit surprising too.

ianrrees, 2023-10-26:
>  what's more interesting, when I flash this chip with platformio, which uses the same derived algos from the cmsis packs, it works fine.

I replicated this using platformio and a J-Link - from the logs it appears that pio uses Segger's software to drive the flashing process, and it doesn't seem to suffer the same problem as here.  With the chip's bootloader protection set to 8kB, the Segger tool flashes the application, and if I set the bootloader protection to 16kB, it fails with an error `((sector is locked))`.  So, it's not fiddling with the bootloader protection, and I guess it isn't using the loader from Microchip's dev_packs.

@LongLiveCHIEF - what probe are you using?

LongLiveCHIEF, 2023-10-26:
I believe I tried with a a rust probe, and an st-link V3... and I recall
doing the same tests with a jlink and pio that you did. I think I even
documented some of it in another related issue here somewhere.

Yatekii, 2023-11-05:
@ianrrees Did we want to still come up with a fix for this or do we require a chip erase? Maybe we can at least fix the diagnostics on this so it tells us what's wrong.

ianrrees, 2023-11-06:
Yes, I think this is still a significant issue.  With https://github.com/probe-rs/probe-rs/pull/1855, probe-rs now does have a chip erase for ATSAM, however that doesn't really help here since the chip erase functionality leaves the "user row" intact, and that includes the bootloader protection setting...

With a wider view, I think we need a generic reading/writing functionality for field like the BOOTPROT, where there's more to it than simply reading/writing a register, and there are also good reasons for developers to want to access just that specific field.  For another instance, in a recent project with i.MX8 (not-Rust and proprietary), I've encountered similar need for programming crypto keys.

I totally agree that better diagnostics messages would be great - in this case it would be difficult to make the diagnostic very helpful without the ability to read the BOOTPROT setting.  The logs in the attached bugs do start with the same failure to erase sector 0x0000, however in other combinations of flash size and BOOTPROT setting, those erasing errors will manifest at different addresses.

ianrrees, 2024-08-11:
@Yatekii - apologies for digging this up after so long away!  I've got some time for OSS again, and am curious whether there have been developments on this?

9names, 2024-08-11:
@ianrrees as long as the memory is addressable this should be resolved by #2654