probe-rs/probe-rs #517: --chip argument not recognised without --elf

thejpster, 2021-01-04:
**Describe the bug**

When I try to run `cargo flash` for a non-standard target using a pack file, it tells me the chip argument doesn't exist. If I give the `--elf` argument, it's fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate a YAML file for the MKE06Z4 pack (http://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.MKE06Z4_DFP.12.2.0.pack) with the latest target-gen (0.5.0)
2. Run `cargo flash --release --chip-description-path ./MKE06Z4.yaml --chip MKE06Z128VLK4`
3. See the error:
```
error: Found argument '--chip' which wasn't expected, or isn't valid in this context

USAGE:
    cargo build --release

For more information try --help
       Error Failed to run cargo build: exit code = Some(1)
```

**Expected behavior**

I expected it to flash my CPU, like it does if I run:
```
cargo flash --release --elf ./target/thumbv6m-none-eabi/release/frdm-ke06-demo --chip-description-path ./MKE06Z4.yaml --chip MKE06Z128VLK4
    Erasing sectors ✔ [00:00:00] [#################################################################################################################]     512B/    512B @   2.94KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [#################################################################################################################]     512B/    512B @     861B/s (eta 0s )
    Finished in 0.457s
```

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - macOS Mojave
 - Windows 10
 - Linux (Ubuntu, Gentoo, Debian, Fedora, etc.)

**Additional context**

My Cargo.toml is:

```
cat Cargo.toml 
[package]
name = "frdm-ke06-demo"
version = "0.1.0"
authors = ["Jonathan Pallant <github@thejpster.org.uk>"]
edition = "2018"
readme = "README.md"

[dependencies]
ke06z4-pac = { path = "../ke06z4-pac", features=["rt"] }
cortex-m = "0.7.0"
cortex-m-rt = "0.6.10"
panic-halt = "0.2.0"

[[bin]]
name = "frdm-ke06-demo"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
```

Yatekii, 2021-01-05:
The source of this issue is that we check if the argument needs to be removed with `starts_with()` which gives a false positive. There is an easy fix for this I can make tomorrow :)

thejpster, 2021-01-05:
Oh! You mean `"chip-description-path".starts_with("chip") == true` ?

Yatekii, 2021-01-05:
Yes ;) It only fails with your order of arguments. if you put --chip first it works ofc which is why our tests didn't spot it. The choice of starts_with has an actual reason, but it was shortsighted lazyness in hindsight ...

commit e614317a (2021-01-08): Fix #517
closed 2021-01-08 by bors[bot]