probe-rs/probe-rs #530: Report target voltage in CLI when available

barnabywalters, 2021-01-25:
ST-Link probes support reporting the target voltage detected on their VAPP pin, which is a useful problem-solving tool when solving connection issues. openocd prints this information by default for every operation:

    Info : Target voltage: 1.529843

I propose that probe-rs-cli and related tools would benefit from reporting this information to the user whenever available.

Additionally, there are several QoL improvements which could easily be built on top of this functionality. Firstly, warning in the case of a voltage which is clearly too low, again in the style of openocd:

    Error: target voltage may be too low for reliable debugging

I can also imagine it being easy to detect a disconnected VAPP pin by taking several consecutive measurements and comparing them. When left disconnected, the measured target voltage seems to flicker between several values, in my case:

    0.003141
    1.556545
    0.001571

whereas a correctly connected VAPP fluctuates by around 0.02v. When handling errors like `SwdDpFault`, `SwdApError`, `JtagGetIdcodeError` and `SwdDpParityError` (for which little-to-no useful debugging information is currently available), automatically detecting a disconnected VAPP from the fluctuating target voltage would be a useful diagnostic.

barnabywalters, 2021-01-25:
I’d be happy to try working on any of these enhancements, but I’m new to rust and would need some guidance. I already browsed through the code and found a few places which seem relevant, but it isn’t obvious to me exactly where the features I propose would best fit into the codebase.

barnabywalters, 2021-01-25:
Oh, I forgot to mention: the same error case caused by leaving VAPP disconnected can sometimes cause `probe-rs-cli info` to print an empty `Available Access Ports:` rather than an error. Whatever codepath leads to that outcome would therefore also be a candidate for low target voltage or disconnected-VAPP diagnostics, but it’s also possible that printing an empty `Available Access Ports:` rather than an error should be considered a separate bug.

Tiwalun, 2021-01-25:
I think it's a great idea to add this.

I would suggest to add a new function to the `DebugProbe` trait, which can be used to read the target voltage.
The function can then be implemented by the different probes.

On the `Probe` struct you can then add a similar function, which forwards to the actual probe. Similar to how the `Probe::speed_khz()` function works.

For reporting the voltage, it would probably be easiest in the beginnnig to add some log output to the `Probe::attach` method, which prints the detected voltage.

commit 69c4a54e (2021-01-26): Implemented Target Voltage Measurement, fixes #530

Added `get_target_voltage()` to `DebugProbe`, based off the original implementation of it
on `STLink`, but generalised to return `Result<Option<f32>, DebugProbeError>` to allow for
probes which don’t support reading target voltage. Added a default implementation on the
trait which returns `Ok(None)`.

Added a `LOW_TARGET_VOLTAGE_WARNING_THRESHOLD` constant to the `probe` module, currently
set to 1.4V as an initial guideline.

Implemented `get_target_voltage()` on `STLink` and `JLink`, based off existing code which
did a similar thing.

Altere [... cut]
closed 2021-01-26 by bors[bot]