probe-rs/probe-rs #544: Flashing >128kByte fails for STM32L452CCUx using an STLINK v2

jhbruhn, 2021-02-12:
**Describe the bug**
When trying to flash a binary with more than 128kByte on a STM32L452CCUx (256kByte flash), the `cargo flash` command fails with an `The execution of 'program_page' failed with code 1` error. Using the 512kByte variant STM32L452CEUx as a --target, but still flashing to the aforementioned chip, the flashing is successful.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a firmware with a size of more than 128kByte for STM32L452CCUx
2. Try to flash it onto that chip via STLINK v2
3. Observe flashing fail after 128kByte

**Expected behavior**
Flashing should succeed even with a binary larger than 128kByte

**Stacktrace**

The last two command calls in program_page before flashing fails:

```
        INFO probe_rs::flashing::flasher                                  > Flashing page at address 0x801fc00 with size: 1024
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write block with total size 1024 bytes to address 0x200003bc
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0x200003BC
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write first block with len 68 at address 0x200003bc
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, block with len=17 words
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0x20000400
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Writing chunk with len 956 at address 0x20000400
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, block with len=239 words
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Finished writing block
       DEBUG probe_rs::flashing::flasher                                  > Calling routine 20000337(Some(134347776), Some(1024), Some(536871868), None, init=false)
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF8
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x20000337
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF4
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x0001000F
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF0
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x00030003
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF4
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x0000000F
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF0
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x00030003
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF8
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x20000336
       DEBUG probe_rs::flashing::flasher                                  > content of PC 0xf: 0x20000336 should be: 0x20000337
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF8
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x0801FC00
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF4
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x00010000
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF0
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x00030003
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF4
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, value=0x00000000
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
[... 370 lines cut ...]
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x00030003
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0xE000EDF8
       DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface         > Read register    DRW, value=0x00000001
        INFO probe_rs::flashing::flasher                                  > Flashing took: 53.549209ms
     Erasing sectors ✔ [00:00:11] [#############################################################################################################################################################################################] 130.00KB/130.00KB @  50.43KB/s (eta 0s )
 Programming pages   ✔ [00:00:08] [###########################################################################################################################################################################################--] 128.00KB/130.00KB @  11.50KB/s (eta 0s )
       Error failed to flash /home/jbruhn/dev/sensing-puck-firmware/target/thumbv7em-none-eabihf/debug/sensing-puck

Caused by:
    0: Error while flashing
    1: The execution of 'program_page' failed with code 1
```

**Desktop (please complete the following information):**
 - Linux, Ubuntu 20.04

**Additional context**
This only happened after I ran rustup update today.
rustc 1.50.0 (cb75ad5db 2021-02-10)
cargo-flash 0.8.0

Unfortunately I do not know which version I had previously, but I remember installing it around December 2020 if that helps

Yatekii, 2021-02-12:
So do I understand that this happened after you updated rustc and was working properly before that?

If this is correct try passing `--nmagic` to the linker :)

jhbruhn, 2021-02-12:
Still fails with the same error.

Yatekii, 2021-02-12:
Could you paste the full log? That also shows what sections it tries to flash.

jhbruhn, 2021-02-12:
Here you go, I hope the debug option was detailed enough: https://gist.github.com/jhbruhn/36208d632e96c3cfc7113e724e4b5391 I somehow could not get the flash command to log to a file successfully, even when redirecting both stdout and stderr.

Yatekii, 2021-02-12:
> I somehow could not get the flash command to log to a file successfully, even when redirecting both stdout and stderr.

Oh, sorry, I forgot that: you can use `--disable-progressbars` so that will not disturb the streams :)

Yatekii, 2021-02-12:
There was some bug in the past with the ST algos for the L4 chips which prevented flashing more than the first half of the memory. @korken89 wrote a fixed implementation which we are using tho if I am not mistaken. So I am curious why this seems to be happening again.
Maybe he can shed some light on it.

So before your update to rustc 1.50 you were able to flash more than 128kB?

Yatekii, 2021-02-12:
The impl is here btw: https://github.com/korken89/stm32l4x2-flashloader/blob/master/src/main.rs. I cannot spot the error tho.

jhbruhn, 2021-02-14:
I just tried it with my firmware compiled with rust v1.46.0 (arbitrary older version), With that flashing succeeds, but not with v1.48.0 . Interestingly enough, flashing succeeds with the newest rust version 1.50 when passing --chip-id STM32L452CEUx as the chip, which is exactly the same but has 512kByte of Flash instead of 256kByte, so it must be some internal (or flashloader-internal?) bound checking.

Yatekii, 2021-02-14:
Yes, I talked to @korken89 and he says that the erase for the upper pages might not work above 128kB because he didn't need it and didn't put it there. And maybe the writing checks if the pages are erased before it writes and thus fails. So the algo most likely needs a small extension :)

jhbruhn, 2021-02-14:
Am I missing something or is the target Flash Size hardcoded here? https://github.com/korken89/stm32l4x2-flashloader/blob/707ba7e5839489afd5a69e11bf16030de9da9003/src/main.rs#L254 I assume the different algorithm variants in the .yaml file for the STM32L4 are compiled with different values for that variable, right? Maybe somewhere there a mistake happened? probe-rs is definitely choosing the right (stm32l4_256) flashing algorithm for my chip, and as it works with the 512-kbyte algorithm, maybe that is the issue? Unfortunately I can't test with a real 512kByte chip.

Yatekii, 2021-02-14:
Oh, I believe @korken89's algotithm is for the _128 variant: https://github.com/probe-rs/probe-rs/commit/42d3100f8e703537336f2443e407c54e92075c7c#diff-6c153a9db71e6e637941c530b65eb84048517a0614318d1a8b967167d2f8425b
I somehow did not look this up properly :/

So I believe the _256 variant is still the broken algo from ST. What we can do here is take Emil's algorithm, adapt it for the _256 variant and replace the ST one with it. Sounds easy enough :)

Yatekii, 2021-02-17:
@jhbruhn this is fixed now, right?

Closing :)

closed 2021-02-17 by Yatekii