probe-rs/probe-rs #550: Listing probes should yield `Result`s

jonas-schievink, 2021-02-15:
Currently, J-Link probes that fail to open do not show up as inaccessible in the probe list. Instead, they just have their serial number and device identifier left blank.

Listing probes should instead yield `Result`s, so that the calling code can act properly, by:
- selecting the only accessible probe when there is more than 1 (but only 1 can be opened successfully)
- providing guidance to the user when the probe is detected but inaccessible (eg. by pointing at a suitable set of udev rules)

This should be relatively easy to fix, by changing how the `Probe::list_all` interface works. I haven't thought about how an ideal interface would look though.

Yatekii, 2021-02-21:
I agree to this.

Though I do not believe it is that easy to fix as all devices are opened a bit differently and some have proper USB descriptors and some do not, etc. The code there is quite messy at the moment so a little refactoring is definitely worthwhile either ways.

Yatekii, 2024-04-27:
I consider this sufficiently fixed in #1269.
Yielding results would make the interface way less convenience and acting on the error is really hard.
I think a log is very good for debugging :)

closed 2024-04-27 by Yatekii