probe-rs/probe-rs #554: Cannot flash or debug STM32G4 when using JTDI as IO pin with SWD

kevswims, 2021-02-24:
**Describe the bug**
When my user code configures the PA15 to a different alternate function, the session fails with the following error:
```    
Finished release [optimized + debuginfo] target(s) in 0.19s
     Running `probe-run --chip STM32G473RETx --defmt target\thumbv7em-none-eabihf\release\hello`
  (HOST) INFO  flashing program (8.12 KiB)
  (HOST) INFO  success!
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to a probe type occured
    1: Command failed with status SwdDpError
error: process didn't exit successfully: `probe-run --chip STM32G473RETx --defmt target\thumbv7em-none-eabihf\release\hello` (exit code: 1)
```
To recover the board I have to flash another program on it using a different tool such as STM32Cube.

Running `$env:RUST_LOG='probe_rs=debug'; probe-rs-cli info` gives the following output:
```
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 004: ID 0bda:5411
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 000: ID 8086:a12f
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 002: ID 046d:c52b
 DEBUG probe_rs::probe::daplink::tools > No device matches
 DEBUG probe_rs::probe::daplink::tools > Attempting to open 0483:374b in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 150, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14155, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 37)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 WARN  probe_rs::probe::stlink                > send_jtag_command 242 failed: JtagNoDeviceConnected
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to a probe type occured
    1: Command failed with status JtagNoDeviceConnected
```

**To Reproduce**
I am flashing using a SWD connection from an ST-Link debug probe.

Here is the minimal code that I have to reproduce the issue:
```
#![no_main]
#![no_std]

use stm32g4xx_hal::stm32g4::stm32g473;

use cortex_m::asm;

#[cortex_m_rt::entry]
fn main() -> ! {
    defmt::info!("Hello, world!");

    let peripherals = stm32g473::Peripherals::take().unwrap();

    let rcc = &peripherals.RCC;

    rcc.ahb2enr.modify(|_, w| w.gpioaen().set_bit());

    let gpioa = peripherals.GPIOA;

    unsafe {
        gpioa.moder.modify(|_,w| {
            w.moder15().bits(0b10)
        });

        gpioa.afrh.modify(|_, w| {
            w.afrh15().bits(11)
        });
    }

    loop {
        for _ in 0..10000000 {
            asm::nop();
        }
        defmt::info!("Hello, world!");
    }
    stm32g4_usb::exit()
}

```

**Expected behavior**
I would expect the probe to configure the STM32G4 into SW-DP mode only so that the JTAG pins can be released. The procedure is in section 47.4.4 in the reference manual:
```
47.4.4 Using serial wire and releasing the unused debug pins as GPIOs
To use the serial wire DP to release some GPIOs, the user software must change the GPIO
(PA15, PB3 and PB4) configuration mode in the GPIO_MODER register.This releases
PA15, PB3 and PB4 which now become available as GPIOs.
When debugging, the host performs the following actions:
• Under system reset, all SWJ pins are assigned (JTAG-DP + SW-DP).
• Under system reset, the debugger host sends the JTAG sequence to switch from the
JTAG-DP to the SW-DP.
• Still under system reset, the debugger sets a breakpoint on vector reset.
• The system reset is released and the Core halts.
• All the debug communications from this point are done using the SW-DP. The other
JTAG pins can then be reassigned as GPIOs by the user software.
Note: For user software designs, note that:
To release the debug pins, remember that they are first configured either in input-pull-up
(nTRST, TMS, TDI) or pull-down (TCK) or output tristate (TDO) for a certain duration after
reset until the instant when the user software releases the pins.
When debug pins (JTAG or SW or TRACE) are mapped, changing the corresponding IO pin
configuration in the IOPORT controller has no effect.
```

**Desktop (please complete the following information):**
 - Windows 10

Yatekii, 2021-02-24:
Hi!

Try using '--connect-under-reset' which does exactly what the refman specifies.

Generally it's a bad idea to reconfigure the programmer pins as the features are kinda wonky and troublesome. Sometimes you ofc have to. Just a general hint :)

Good Luck!

kevswims, 2021-02-24:
Thanks @Yatekii 
I tried the `--connect-under-reset` flag and that does allow me to flash when that pin is in the non-working state.

It does not seem to configure the IC to only SW-DP mode so code that uses the JTDI pin still breaks the debugger connection.

Understood on it being difficult to reconfigure the programmer pins however this is the whole point of using SWD is that is allows the JTAG pins to be freed for other uses. On the particular part it is not really possible to use multiple CAN channels without using that JTDI pin for one of them.

adamgreig, 2021-02-24:
probe-rs definitely uses SWD by default, not JTAG - I don't think we even support JTAG operation for Cortex-M - but something else seems to be going wrong here, possibly with the stlink interface. The IC doesn't need special configuring to SW-DP mode I don't believe; the debug port is a SWJ-DP and the stlink should transmit the sequence to go into SWD mode, which is what happens in normal operation.

In this case, though, the stlink is reporting an error after trying to swap to SWD (confusingly when it says 'jtag' some of that also means 'swd', just how stlink names things):

```
DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 37)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 WARN  probe_rs::probe::stlink                > send_jtag_command 242 failed: JtagNoDeviceConnected
```

I don't know enough about the stlink's operation to know what's going wrong exactly, but perhaps something else in your code is causing the issue. If you keep everything the same but disable just one or both of the GPIOA register writes, does that help? I also noticed you have `w.afrh15().bits(11)` which I guess should be `0b11` but I don't think that's the problem.

kevswims, 2021-02-24:
@adamgreig 

I'm not quite understanding what the reference manual has to say on this. Section 47.4.2 seems to indicate that a specific action needs to be done to disable JTAG and only use SWD (https://www.st.com/resource/en/reference_manual/dm00355726-stm32g4-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) but it is not clear to me what that action is.

Writing `11` for the `afrh15` is intentional. I am trying to switch that pin from AF0 to AF11. AF0 is the `JTDI` function and I am trying to use the `CAN3_TX` function.

adamgreig, 2021-02-24:
> Section 47.4.2 seems to indicate that a specific action needs to be done to disable JTAG and only use SWD

The action to activate the SW-DP is described in 47.3.1: the probe sends a special sequence over TMS/TCK, which probe-rs always does (it always opens in SWD mode). To release PA15 etc from JTAG use you just remap it in the GPIO registers as you're already doing (described in 47.4.4). I guess you could try also releasing PB3 and PB4 in case they all need to be changed... it would also be worth double checking the right bits are being written to the registers, inc as there's an error in the PAC.

The only thing that does stand out to me is that 47.4.4 specifically calls out that the debugger would attach under reset and then the user firmware would release those ports. In general I don't think this is required (i.e. it should be fine to connect not-under-reset) but I don't know why the manual calls this out. You said you can recover using STM32Cube: does that also connect under reset, or is it able to connect while running?

Basically this isn't something where probe-rs usually has to do anything differently; it already only uses the SWD lines. It _could_ be some weird interaction with the STlink (since their protocol is undocumented), if you had another probe you could try that, or it could be some weird errata on the chip (like needing all the JTAG pins released, or something else... there's an errata about nJTRST but that shouldn't be affecting this). Or it could be an error in the PAC causing the wrong bits to be written.

> Writing `11` for the `afrh15` is intentional. I am trying to switch that pin from AF0 to AF11. AF0 is the `JTDI` function and I am trying to use the `CAN3_TX` function.

Sorry, yep, of course.

kevswims, 2021-02-24:
I tested with also reassigning PB3 and PB4 and the behavior is the same. 

I did notice while testing that with the connect-under-reset flag I am getting a slightly different error on the disconnect failure where it gives SwdApError:

```
    Finished release [optimized + debuginfo] target(s) in 1.30s
     Running `probe-run --chip STM32G473RETx --defmt --connect-under-reset target\thumbv7em-none-eabihf\release\hello`
  (HOST) INFO  flashing program (8.31 KiB)
  (HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Error: Error communicating with probe: An error with the usage of the probe occured

Caused by:
    0: An error with the usage of the probe occured
    1: An error specific to a probe type occured
    2: Command failed with status SwdApError
error: process didn't exit successfully: `probe-run --chip STM32G473RETx --defmt --connect-under-reset target\thumbv7em-none-eabihf\release\hello` (exit code: 1)
```

kevswims, 2021-02-24:
Well this is fun. It looks like my setup fails in the same way if I configure PA15 to other modes from within STM32Cube as well. I guess I will be filing this issue with ST and seeing what they come back with.

Yatekii, 2021-02-24:
I am curious to hear what they say :)

Also, I'll implement a hint for this error case for cargo-flash, such that future users will be able to know what's going on :)

Also, @adamgreig I have never looked at the SWD line of a ST-Link. I'll do once I get my glasgow ... What could be is that the ST-Link API call we do actually does unlock the combined interface, but with the new info from @kevswims I do not believe that is the issue. Still, with ST-Links I'd never rule out us holding it wrong :P (see issue from Sunday ... was great fun ... 2 lines of code in 16hrs or something :D)

kevswims, 2021-02-24:
Here's a link to my post to ST. https://community.st.com/s/question/0D53W00000arR83SAE/stm32g473-unable-to-use-pa15-as-gpio-when-debugging-in-swd-mode

I'll update when I get a response back. If I don't hear anything back from them soon I'll see if I can get in contact with an Applications Engineer through work.

Disasm, 2021-02-25:
I noticed that errata for this chip contains the following section:

[image]

This may be somehow relevant to the problem in the issue. Maybe JTAG pins should be reconfigured in a specific sequence.

Yatekii, 2021-02-25:
But he is not using JTAG, but SWD :)

Disasm, 2021-02-25:
Sure, I mean, they screwed up somewhere nearby. The same thing was with memory reads during sleep: I found a different issue in errata which happened to be relevant.

kevswims, 2021-02-25:
Well it seems that this was much more simple than it seemed. I had a hidden short between PA15 (JTDI) and the SWCLOCK pin so when I modified the pin mode PA15 became an output instead of an input and things stopped working... With the short removed everything works great.

🤦 

Thanks so much for the help and suggestions everyone and at least I got to learn a bit more about how the debug interface works. 😀

closed 2021-02-25 by kevswims