probe-rs/probe-rs #555: probe-rs timed out when reading hardware breakpoints

spcan, 2021-02-24:
**Describe the bug**
When running the latest v0.2.0 `probe-run` with the STM32-F411-VET discovery board, the process fails with:
```
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out
```
`probe-run` logs show that this timeout happens when the function `core.set_hw_breakpoint(main)?` is called.

**To Reproduce**
Steps to reproduce the behavior:
1. Run a project with the STM32F411VET and with `probe-run`
2. Probe gets flashed correctly
3. Probe hangs 

**Expected behavior**
The probe does not hang and it starts normal function

**Stacktrace**
```
  (HOST) DEBUG RAM region: 0x20000000-0x2001FFFF
  (HOST) DEBUG section `.data` is in RAM at 0x20000000-0x200001C7
  (HOST) DEBUG section `.bss` is in RAM at 0x200001C8-0x200001D7
  (HOST) DEBUG section `.uninit` is in RAM at 0x200001D8-0x200005D7
  (HOST) DEBUG vector table: VectorTable { location: 8000000, initial_sp: 20020000, reset: 8000199, hard_fault: 8000b19 }
  (HOST) DEBUG found 1 probes
  (HOST) DEBUG opened probe
  (HOST) DEBUG started session
  (HOST) INFO  flashing program (4.59 KiB)
  (HOST) INFO  success!
  (HOST) DEBUG 129576 bytes of stack available (0x200005D8-0x20020000), using 1024 byte canary to detect overflows
  (HOST) DEBUG starting device
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out
```

**Desktop (please complete the following information):**
 - Linux Ubuntu 20.04 LTS

**Additional context**

I do not know how to enable logging for the `probe-rs` crate when used through `probe-run`, if someone could tell me, I can enable it and update with the logs.

spcan, 2021-02-25:
I checked a bit further and apparently it is actually timing out at the `core.wait_for_core_halted()` function. From what I can see there is no way to know what it is reading without enabling logging inside the crate. 

I'm running `probe-run` but it does not log anything from the `probe-rs` crate. Is there any way to enable this functionality from my project, or would I have to use `probe-rs` directly?

Yatekii, 2021-02-25:
I am not familiar with the probe-run crate, but you could try and prepend RUST_LOG=DEBUG to your command and see if that helps :)

ssnover, 2021-06-03:
@spcan I was running into this issue for a little while as well. The error occurred due to the configuration of my `memory.x` file. When thumbing through the reference manual I had grabbed the RAM and FLASH size for the STM32F302xB chip instead of the STM32F302x8 chip on my board and this was the symptom that something was wrong. I'd double check those.

Yatekii, 2021-08-21:
I will consider this closed as this was on a very old version of probe-rs. Please reopen if it still persists.

closed 2021-08-21 by Yatekii