probe-rs/probe-rs #557: Unable to connect to STM32F303x Discovery board

smhalp, 2021-02-26:
**Describe the bug**
Cant probe my STM32F303x Discovery board. Keep hitting a JtagGetIdcodError message. Using a Mac running Catalina 10.15.7.

**STACKTRACE**
$ RUST_LOG=probe_rs=debug probe-rs-cli info
DEBUG probe_rs::probe::daplink::tools > Attempting to open 0483:374b in CMSIS-DAP v1 mode
DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 200, extra: None }
DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14155, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
DEBUG probe_rs::probe::stlink                > Initializing STLink...
DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
DEBUG probe_rs::probe::stlink                > STLink version: (2, 37)
DEBUG probe_rs::probe::stlink                > attach(Swd)
DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
WARN  probe_rs::probe::stlink                > send_jtag_command 242 failed: JtagGetIdcodeError
DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to a probe type occured
    1: Command failed with status JtagGetIdcodeError

jspencer, 2021-04-04:
I have the same setup, STM32F303x Discovery on macOS 10.15.7 and mine is working, so maybe I can help. 
It looks like it's failing while attempting to switch from MassStorage mode to SWD. I had re-flashed the STLink using the "stsw-link007" package from the ST website to upgrade to the latest firmware and at that time I chose the bundle without MassStorage. I did it mainly to stop the filesystem from automounting when I plugged it in but it may help with your issue. There's a warning about having their latest usb driver that I ignored, assuming it was intended for linux users using the same java-based flash util, and there was no issue.

Maybe obvious, but make sure CN4 is bridged, so your probe is connected to the target.

You might also make sure you have solder bridge 10 bridged on the Discovery board. It doesn't seem like it would cause this error, but it's another potential difference.

bugadani, 2024-04-01:
After 3 years of inactivity I'm tempted to close this issue. A lot has changed since then, so if there was an issue, it might have gone away "naturally". Let me know if you encounter this again.

closed 2024-04-01 by bugadani