probe-rs/probe-rs #558: segmentation fault with 'probe-rs-cli list'

noppej, 2021-02-26:
Hi,
I suspect something in my development environment is broken, and would appreciate any pointers to help me troubleshoot.

My problem is that probe-rs-cli fails with segmentation fault (with board connected to USB, as well as when board is not connected)

**Desktop **
 - macOS Catalina

**Additional context**
This used to work for me as recently as yesterday. This morning, nothing!

At first I thought it was a hardware error on either my Nucleo board, or my Mac, but ...
- STM32CubeProgrammer can connect to, program and inspect the MCU
- probe-run --list-probes shows the available probes
- cargo-embed can program and run my embedded program successfully, and even shows RTT output as expected.

Yatekii, 2021-02-26:
Hey @noppej could you please provide some more info?

- When exactly does it segfault? The bulletpoints you listed seem to indicate that it works without segfault sometimes. Does it only work on another OS, or does it sometimes work on OSX too?
- What is the command that segfaults?
- What version of probe-rs do you run? Is this master or 0.10.1?

noppej, 2021-02-26:
Great questions. I only have the one Mac, so can't test anywhere else, but it happens consistently since this morning. I've been testing with this for 2 weeks, and it has worked without fail ... until today. Other tools (ST, cargo-embed, probe-run) still work, but probe-rs library faults on the same place, _every time_, and `probe-rs-cli list` is an easy way to reproduce (also to test that it wasn't my code (use of probe-rs api) that behaved uniquely.

I am working with the latest master of probe-rs, and the attached is my VSCode screenshot when the problem happens. Is there a better way I can capture a stack trace to send to you?
[image]

Yatekii, 2021-02-26:
Hmm this seems like a bug in the libusb bindings :/ But I can't remember that we did any specific update of this crate.

I guess if you use cargo embed/flash from master (or well the PR branches that are open to fix master) it will crash too? I could try on my brothers mac maybe. Or maybe @Tiwalun is in the mood too.

noppej, 2021-02-26:
To test the latest cargo-embed, I just did a `cargo install --force --git https://github.com/probe-rs/cargo-embed`

It failed to build with the following output ... am I doing something wrong?
```

    Updating git repository `https://github.com/probe-rs/cargo-embed`
  Installing cargo-embed v0.10.1 (https://github.com/probe-rs/cargo-embed#a2f82165)
    Updating crates.io index
    Updating git repository `https://github.com/probe-rs/probe-rs`
    Updating git repository `https://github.com/probe-rs/probe-rs-rtt`
  Downloaded smawk v0.3.1
  Downloaded parking_lot_core v0.8.3
  Downloaded textwrap v0.13.4
  Downloaded goblin v0.3.4
  Downloaded 4 crates (248.7 KB) in 0.90s
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling libc v0.2.86
   Compiling serde v1.0.123
   Compiling serde_derive v1.0.123
   Compiling cfg-if v1.0.0
   Compiling autocfg v1.0.1
   Compiling memchr v2.3.4
   Compiling value-bag v1.0.0-alpha.6
   Compiling log v0.4.14
   Compiling cc v1.0.67
   Compiling pkg-config v0.3.19
   Compiling version_check v0.9.2
   Compiling bitflags v1.2.1
   Compiling lazy_static v1.4.0
   Compiling crc32fast v1.2.1
   Compiling futures-core v0.3.13
   Compiling ryu v1.0.5
   Compiling once_cell v1.7.0
   Compiling pin-project-lite v0.2.4
   Compiling take_mut v0.2.2
   Compiling rle-decode-fast v1.0.1
   Compiling futures-io v0.3.13
   Compiling adler32 v1.2.0
   Compiling linked-hash-map v0.5.4
   Compiling proc-macro-hack v0.5.19
   Compiling static_assertions v1.1.0
   Compiling cache-padded v1.1.1
   Compiling radium v0.5.3
   Compiling hashbrown v0.9.1
   Compiling unicode-width v0.1.8
   Compiling dtoa v0.4.7
   Compiling cfg-if v0.1.10
   Compiling waker-fn v1.1.0
   Compiling byteorder v1.4.2
   Compiling parking v2.0.0
   Compiling smallvec v1.6.1
   Compiling event-listener v2.5.1
   Compiling fallible-iterator v0.2.0
   Compiling scopeguard v1.1.0
   Compiling base64 v0.13.0
   Compiling adler v1.0.2
   Compiling fastrand v1.4.0
   Compiling stable_deref_trait v1.2.0
   Compiling ucd-trie v0.1.3
   Compiling anyhow v1.0.38
   Compiling funty v1.1.0
   Compiling lexical-core v0.7.5
   Compiling proc-macro-nested v0.1.7
   Compiling rusb v0.7.0
   Compiling wyz v0.2.0
   Compiling tap v1.0.1
   Compiling serde_json v1.0.63
   Compiling regex-syntax v0.6.22
[... 156 lines cut ...]
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^ doesn't have a size known at compile-time
    | 
   ::: /Users/Jack/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/9ae0e98/probe-rs/src/flashing/download.rs:119:35
    |
119 | pub fn download_file_with_options<P: AsRef<Path>>(
    |                                   - required by this bound in `download_file_with_options`
    |
    = help: within `Path`, the trait `Sized` is not implemented for `[u8]`
    = note: required because it appears within the type `Path`

error[E0063]: missing fields `do_chip_erase`, `dry_run` in initializer of `DownloadOptions<'_>`
   --> src/main.rs:452:17
    |
452 |                 DownloadOptions {
    |                 ^^^^^^^^^^^^^^^ missing `do_chip_erase`, `dry_run`

error: aborting due to 4 previous errors

Some errors have detailed explanations: E0063, E0277.
For more information about an error, try `rustc --explain E0063`.
error: failed to compile `cargo-embed v0.10.1 (https://github.com/probe-rs/cargo-embed#a2f82165)`, intermediate artifacts can be found at `/var/folders/dd/lk8h_t2x0vbb_cgsrj97f02m0000gn/T/cargo-installieDTSs`

Caused by:
  could not compile `cargo-embed`
```

Yatekii, 2021-02-26:
Sorry, I should have been more specific: You need `--branch master-fix` :)

noppej, 2021-02-26:
Oops, sorry. I missed that ... the build for cargo-embed worked after pointing to master-fix
```
   Compiling cargo-embed v0.10.1 (/Users/Jack/.cargo/git/checkouts/cargo-embed-7664de7e4a0e8cc2/258db00)
    Finished release [optimized] target(s) in 2m 55s
   Replacing /Users/Jack/.cargo/bin/cargo-embed
    Replaced package `cargo-embed v0.10.1` with `cargo-embed v0.10.1 (https://github.com/probe-rs/cargo-embed?branch=master-fix#258db00f)` (executable `cargo-embed`)
```

but now I get a segmentation fault with cargo-embed also
```
   Compiling sstp-rs v0.8.0 (/Users/Jack/dev/sstp-rs)
    Finished dev [optimized + debuginfo] target(s) in 0.96s
      Config default
      Target /Users/Jack/dev/sstp-rs/target/thumbv7em-none-eabihf/debug/sstp_compute_and_adjust
zsh: segmentation fault  cargo-embed --bin sstp_compute_and_adjust
```

Yatekii, 2021-02-26:
Hmm `rusb` was v0.6.0 in 0.10.1 and on master it is v0.7.0. You could try downgrading that and see if it works :)

noppej, 2021-02-26:
That was it ... what would you like me to do with this ticket?
PS. Big thanks for helping me get un-stuck.

```
   Compiling probe-rs-cli v0.10.0 (/Users/Jack/dev/probe-rs/cli)
    Finished release [optimized + debuginfo] target(s) in 1m 21s
   Replacing /Users/Jack/.cargo/bin/probe-rs-cli
    Replaced package `probe-rs-cli v0.10.0 (/Users/Jack/dev/probe-rs/cli)` with `probe-rs-cli v0.10.0 (/Users/Jack/dev/probe-rs/cli)` (executable `probe-rs-cli`)
Jack@JackiMac cli % probe-rs-cli list
The following devices were found:
[0]: STLink V3 (VID: 0483, PID: 374e, Serial: 0039002D3137510E39383538, STLink)
```

Yatekii, 2021-02-27:
It was a pleasure!

Next steps for this would be:

- [ ] Check if the 0.8.0 version of `rusb` that was released yesterday still contains this regression.
- [ ] If so, report the issue on the [rusb](https://github.com/a1ien/rusb) repo and create a PR with a downgrade to v0.6.0.
- [ ] If not, create a PR with an upgrade to v0.8.0.

noppej, 2021-02-27:
Here's what I've done so far ...

- [x] Verified that the problem exists in `rusb` 0.7 and 0.8, but not in 0.6
- [x] Reported it to `rusb` repo as https://github.com/a1ien/rusb/issues/53

- [ ] I cannot create a PR at this time, because I have too many changes against the same repo that is not ready for PR yet, and git (probably user error) keeps trying to add those changes to the PR. Any chance @Yatekii that you or someone can do the PR?

Yatekii, 2021-02-27:
Sure, I can do a PR :) But let's wait for a minute to see what @a1ien has to say about this =)

mcuee, 2021-03-04:
From the rusb ticket.

To know the low-level libusb debug info, run your program with LIBUSB_DEBUG=4 set, then from the debug log, you can see the low level libusb function which failed. Then it is easier to see if it is a problem with libusb or rusb or probe-rs.

Please also take a look at the following libusb issues.

Segfault at libusb_exit() -- libusb crashing on the wrong context. This could be the issues here. It is expected behavior but there is an improvement already (not yet merged but should be soon).
libusb/libusb#855

libusb_set_interface_alt_setting returns kIOUSBPipeStalled on macOS (fixed in latest git)
libusb/libusb#838

noppej, 2021-03-19:
@Yatekii, I've just tested this against the latest master, and the problem is resolved.

noppej, 2021-03-24:
Closing it as 'resolved'

closed 2021-03-24 by noppej