probe-rs/probe-rs #559: Hide specific features of `object`

Urhengulas, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
I'd like to decrease the dependency-chain of [`probe-run`](github.com/knurling-rs/probe-run) by not pulling in all dependencies of the default features of `object`, e.g. `wasmparser` in our case.

**Describe the solution you'd like**
I'd like `probe-rs` to hide the more niche features of `object` behind a feature flag. I am thinking of specific file support here.

**~~Describe alternatives you've considered~~**

**~~Additional context~~**

Urhengulas, 2021-03-09:
Hi @Yatekii, I am happy to do these changes, if you are generally interested in this. What do you think?

Yatekii, 2021-03-09:
Hi @Urhengulas

I'll gladly accep a PR for this. I am not a dependency fanatic myself so I wont put my time into it, but a slimmer dependency count is a good thing :)

I do not even think that probe-rs needs a feature flag for this because we do not flash wasm.

I would like to have no feature flags at all. If we have those, we have the same issue that OpenOCD suffers from. A fragmented set of the same tool with very distinct featuresets.
The two flags we currently have are absolutely necessary unfortunately. `ftdi` because there is troubles with windows still and `sentry` because some folks are not okay with "tracking".

I hope that answers the question :) Ping me if not :)

commit cead0274 (2021-03-10): Merge #580

580: Minimize deps r=Yatekii a=Urhengulas

This PR removes the sub-dependencies `flate2`, `indexmap`, `wasmparser` by disabling all expect required features for `gimli` and `object`. Also it ~~bumps `rusb` to `0.8.0` and~~ sorts the dependencies.

Fixes #559 

## compile time comparison
The compile time decreases by `~6s` in release mode and `~3s` in dev mode. All of this is only tested on my laptop, but should still give a good enough estimate.
* `master` (17c71ef72a4b1d7635517626af5966a9aeb730e3)
  * `cargo clean && cargo build`
    * 35.45s
    * 36.62s
    * 34.91s
  * `cargo c [... cut]
PR #580 (2021-03-10): Minimize deps
This PR removes the sub-dependencies `flate2`, `indexmap`, `wasmparser` by disabling all expect required features for `gimli` and `object`. Also it ~~bumps `rusb` to `0.8.0` and~~ sorts the dependencies.

Fixes #559 

## compile time comparison
The compile time decreases by `~6s` in release mode and `~3s` in dev mode. All of this is only tested on my laptop, but should still give a good enough estimate.
* `master` (17c71ef72a4b1d7635517626af5966a9aeb730e3)
  * `cargo clean && cargo build`
    * 35.45s
    * 36.62s
    * 34.91s
  * `cargo clean && cargo build --release`
    * 1m 0 [... cut]
closed 2021-03-10 by bors[bot]