probe-rs/probe-rs #561: Jlink: Register DPIDR not supported by debug port version DPv0

allexoll, 2021-03-01:
**Describe the bug**
probe-rs fails to use the debugger on NXP's LPCxpresso54608 board

**To Reproduce**
Steps to reproduce the behavior:
1. program the JLINK firmware on the board using LPCscrypt > ./program_JLINK
2. run cargo flash

**Expected behavior**
cargo flash programs the chip, probe-rs can be used for debugging, etc.

**Stacktrace**
```
../cargo-flash/target/debug/cargo-flash --release --example blinky --chip-description-path ../probe-rs/probe-rs/targets/LPC546xx\ Series.yaml  --chip LPC54608J512ET180 --features="rt mcu-lpc54608j512et180"  --log trace
       DEBUG cargo_flash > Changed working directory to .
       DEBUG probe_rs_cli_util > Running 'cargo' in directory /Users/allexoll/Desktop/lpc546xx-hal
    Finished release [optimized + debuginfo] target(s) in 0.15s
    Flashing /Users/allexoll/Desktop/lpc546xx-hal/target/thumbv7em-none-eabi/release/examples/blinky
       DEBUG probe_rs::config::registry > Searching registry for chip with name LPC54608J512ET180
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x4
        INFO probe_rs::flashing::download > Matching section: ".vector_table"
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x00000124, virtual address: 0x00000124, flags: 0x5
        INFO probe_rs::flashing::download > Matching section: ".text"
        INFO probe_rs::flashing::loader   > Found 2 loadable sections:
        INFO probe_rs::flashing::loader   >     .vector_table at 00000000 (292 bytes)
        INFO probe_rs::flashing::loader   >     .text at 00000124 (488 bytes)
       DEBUG probe_rs::flashing::loader   > Adding data: ExtractedFlashData { name: [".vector_table"], address: 0, data: "[..] (292 bytes)" }
       DEBUG probe_rs::flashing::loader   > Adding data at address 0x00000000 with size 292 bytes
       DEBUG probe_rs::flashing::loader   > Adding data: ExtractedFlashData { name: [".text"], address: 124, data: "[..] (488 bytes)" }
       DEBUG probe_rs::flashing::loader   > Adding data at address 0x00000124 with size 488 bytes
       DEBUG jaylink                      > libusb 1.0.24.11584
       DEBUG jaylink                      > libusb has capability API: true
       DEBUG jaylink                      > libusb has HID access: false
       DEBUG jaylink                      > libusb has hotplug support: true
       DEBUG jaylink                      > libusb can detach kernel driver: false
       DEBUG jaylink                      > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                      > scanning 3 interfaces
       TRACE jaylink                      > active configuration descriptor: ConfigDescriptor {
    bLength: 0x9,
    bDescriptorType: 0x2,
    wTotalLength: 0x62,
    bNumInterfaces: 0x3,
    bConfigurationValue: 0x1,
    iConfiguration: 0x4,
    bmAttributes: 0x80,
    bMaxPower: 0x32,
    extra: Some(
        [
            0x8,
            0xb,
            0x0,
            0x2,
            0x2,
            0x2,
            0x0,
[... 204 lines cut ...]
        "Unspecified(\"LPC54608J512ET180\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "0",
    ),
    release: "0.10.2",
    commit: "71e0526",
}
       DEBUG cargo_flash::diagnostics                             > 0: An error with the usage of the probe occured
       DEBUG cargo_flash::diagnostics                             > 1: An error specific to the selected architecture occured
       DEBUG cargo_flash::diagnostics                             > 2: Register DPIDR not supported by debug port version DPv0
       Error Connecting to the chip was unsuccessful.

        Hint A hard reset during attaching might help. This will reset the entire chip. Run with `--connect-under-reset` to enable this feature.
```

**Desktop (please complete the following information):**
 - macOS BigSur 11.2.2

**Additional context**
The jlink firmware is somewhat limited [according to their website](https://www.segger.com/products/debug-probes/j-link/models/other-j-links/lpc-link-2/#firmware-version-history-and-limitations). 
That may make this a non-issue if there is no plan to support that kind of niche debugger. I tried with the CMSIS-dap as well, without success, but i'll open an other issue to track it.

Dirbaio, 2021-03-01:
I'm seeing the same error message with using a stm32f401 blackpill with the WeAct CMSIS-DAP firmware to debug a stm32f411 blackpill. No idea if it's related.

https://gist.github.com/Dirbaio/cc7608b2b2d4f24579159d1089e616d2

sparky8251, 2021-03-02:
Here's my logs for this error as well. My probe is this little one here: https://www.adafruit.com/product/3571

```
Running `target/debug/probe-rs-cli info`
 DEBUG jaylink > libusb 1.0.24.11584
 DEBUG jaylink > libusb has capability API: true
 DEBUG jaylink > libusb has HID access: true
 DEBUG jaylink > libusb has hotplug support: true
 DEBUG jaylink > libusb can detach kernel driver: true
 DEBUG jaylink > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
 DEBUG jaylink > scanning 1 interfaces
 TRACE jaylink > active configuration descriptor: ConfigDescriptor {
    bLength: 0x9,
    bDescriptorType: 0x2,
    wTotalLength: 0x20,
    bNumInterfaces: 0x1,
    bConfigurationValue: 0x1,
    iConfiguration: 0x4,
    bmAttributes: 0x80,
    bMaxPower: 0x32,
    extra: None,
}
 TRACE jaylink > interface #1 descriptors:
 TRACE jaylink > InterfaceDescriptor {
    bLength: 0x9,
    bDescriptorType: 0x4,
    bInterfaceNumber: 0x0,
    bAlternateSetting: 0x0,
    bNumEndpoints: 0x2,
    bInterfaceClass: 0xff,
    bInterfaceSubClass: 0xff,
    bInterfaceProtocol: 0xff,
    iInterface: 0x5,
}
 TRACE jaylink > endpoint descriptors: [
    EndpointDescriptor {
        bLength: 0x7,
        bDescriptorType: 0x5,
        bEndpointAddress: 0x81,
        bmAttributes: 0x2,
        wMaxPacketSize: 0x40,
        bInterval: 0x1,
    },
    EndpointDescriptor {
        bLength: 0x7,
        bDescriptorType: 0x5,
        bEndpointAddress: 0x2,
        bmAttributes: 0x2,
        wMaxPacketSize: 0x40,
        bInterval: 0x1,
    },
]
 DEBUG jaylink > J-Link interface is #0
 DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 001: ID 1d6b:0003
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
[... 143 lines cut ...]
 TRACE jaylink                                > read 4 bytes: [83, 0, 0, 0]
 DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
 INFO  probe_rs::probe::jlink                 > J-Link: S/N: 801024252
 DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
 TRACE jaylink                                > write 1 bytes: [1]
 TRACE jaylink                                > read 2 bytes: [70, 0]
 TRACE jaylink                                > read 112 bytes: [4a, 2d, 4c, 69, 6e, 6b, 20, 45, 44, 55, 20, 4d, 69, 6e, 69, 20, 56, 31, 20, 63, 6f, 6d, 70, 69, 6c, 65, 64, 20, 4f, 63, 74, 20, 32, 32, 20, 32, 30, 31, 39, 20, 31, 36, 3a, 32, 38, 3a, 34, 35, 0, 43, 6f, 70, 79, 72, 69, 67, 68, 74, 20, 32, 30, 31, 36, 2d, 32, 30, 31, 36, 20, 53, 45, 47, 47, 45, 52, 3a, 20, 77, 77, 77, 2e, 73, 65, 67, 67, 65, 72, 2e, 63, 6f, 6d, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 INFO  probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Oct 22 2019 16:28:45
 TRACE jaylink                                > write 1 bytes: [f0]
 TRACE jaylink                                > read 4 bytes: [10, 27, 0, 0]
 INFO  probe_rs::probe::jlink                 > J-Link: Hardware version: JLink 1.0.0
 TRACE jaylink                                > write 1 bytes: [7]
 TRACE jaylink                                > read 8 bytes: [0, 0, 0, 0, 0, 0, 1, ff]
 WARN  probe_rs::probe::jlink                 > J-Link: Target voltage (VTref) is 0.00 V. Is your target device powered?
 TRACE jaylink                                > write 24 bytes: [cf, 0, 50, 0, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, 9e, e7]
 TRACE jaylink                                > read 11 bytes: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 DEBUG probe_rs::probe::jlink::swd            > Performing line reset!
 TRACE jaylink                                > write 30 bytes: [cf, 0, 62, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
 TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 DEBUG probe_rs::probe::jlink::swd            > Unexpected response from target, does not conform to SWD specfication (ack=[false, false, false])
 TRACE jaylink                                > write 30 bytes: [cf, 0, 62, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
 TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 DEBUG probe_rs::probe::jlink::swd            > Unexpected response from target, does not conform to SWD specfication (ack=[false, false, false])
 WARN  probe_rs_cli::info                     > Error identifying target using protocol SWD: An error with the usage of the probe occured
```

dbrgn, 2021-04-29:
@Dirbaio same issue here, with an STM32F411 blackpill from WeAct. Did you resolve your issue?

dbrgn, 2021-04-29:
Ohh, never mind, I'm an idiot. I accidentally swapped SWDIO and SWCLK pins on my JLink 🙈🤦🏻‍♂️ With the proper connections I can flash the MiniF4 / Black Pill.