probe-rs/probe-rs #562: CMSIS-dap:  Failed to read register DRW at address 0x0000000c

allexoll, 2021-03-01:
**Describe the bug**
 probe-rs fails to use the debugger on NXP's LPCxpresso54608 board

**To Reproduce**
Steps to reproduce the behavior:
1. program the CMSIS-dap fw on NXP's LPCXpresso54608 dev kit using LPCscypt > ./program_CMSIS
2. run cargo flash

**Expected behavior**
device is programmed

**Stacktrace**
[https://gist.github.com/allexoll/4fa74ce4f2c1e1af6e5b1e4cc484ab3f](https://gist.github.com/allexoll/4fa74ce4f2c1e1af6e5b1e4cc484ab3f)

**Desktop (please complete the following information):**
 - macOS BigSur 11.2.2

**Additional context**
I opened a similar issue with the JLINK firmware, while both fail, they do not fail the same, which is why I'm opening 2 issues.

Yatekii, 2021-03-01:
Could it be that your chip is locked of some kind? Also @Tiwalun was working on #483 for the LPC55s69. Not sure if your LPC requires something similar.

allexoll, 2021-03-02:
Both the probe and the target work when i debug them with segger software or with MCUXpresso. I glossed over #483 and i don't think this is it, neither the datasheet or the user manual say much about debugging, except that is is SWD, but i don't know where @Tiwalun found information about his debug sequence, maybe that might help me say if it is related

Yatekii, 2021-03-02:
Just because MCUXpresso can debug it, it doesn't mean it doesn't just unlock the MCU :)

allexoll, 2021-03-02:
True, but i'm able to use it with vscode/cortex-debug/jlink software when it is programmed with jlink probe fw. but again, that doesn't mean that jlink doesn't just unlock the probe...

Yatekii, 2021-03-02:
Yeah, I am not claiming this to be the case here, but we had cases where this was thea ctual problem, which is why I wanted to make sure :)

ianrrees, 2022-02-21:
@allexoll, I wonder if https://github.com/probe-rs/probe-rs/pull/1019 might help with this?