probe-rs/probe-rs #575: Don't ignore non-existent algorithms in target YAMLs

Dirbaio, 2021-03-07:
In target yamls, you reference flash algos by name in `/variants/flash_algorithms`.  If you specify an algo that doesn't actually exist in `/flash_algorithms`, it is silently ignored instead of erroring. This can cause no algos to be available for the core, in which case flashing fails with a `no suitable flash loader algorithm` error. This should fail instead at YAML loading time.

This is due to the filtermap here
https://github.com/probe-rs/probe-rs/blob/9812975398e5a364d8f36191d2f6c49d91d4ad83/probe-rs/src/config/registry.rs#L262

jukrieger, 2024-04-03:
@bugadani can you assign this to me? Seems simple enough.

bugadani, 2024-04-03:
Maybe, but the issue is open so who knows, someone needs to check ;)

bugadani, 2024-04-04:
The check has been added to `Target::new` and it's also present in `ChipFamily::validate` albeit not by calling the same code. IMO `validate` could be somewhat simplified by trying to create a `Target` out of each chip `variant` besides the extra checks that it contains currently. After that I believe we will be able to close this issue.

bugadani, 2024-04-04:
Another thing we should do is run the validation logic in `build.rs` so probe-rs doesn't even build.

Dirbaio, 2024-04-04:
iirc we didn't put it in build.rs to not make the build slower for users using probe-rs as a lib (either from git or crates.io), since the target yamls there are supposed to be good since they're checked by CI. No need to spend effort at build time checking something that we know should be good.

Yatekii, 2024-04-04:
Ah I forgot about that :D Yes, we should not put them in build.rs then. Absolutely right!

jukrieger, 2024-04-04:
I agree :)

commit a9dcd430 (2021-08-22): Validate all flash algos exist on target creation. Fixes #575
commit d155c5ad (2021-08-22): Validate all flash algos exist on target creation. Fixes #575
commit 07da89dd (2021-08-23): Validate all flash algos exist on target creation. Fixes #575
commit db4a5eb3 (2021-08-23): Validate all flash algos exist on target creation. Fixes #575
closed 2025-05-24 by Yatekii