probe-rs/probe-rs #581: probe-rs-cli: Panic while flashing ELF executable to nRF52840

Javier-varez, 2021-03-14:
**Describe the bug**
Flashing an elf using probe-rs-cli panics with the following message:
```
thread 'main' panicked at 'Overlap in data, address 0x00000974 was already written earlier. This is a bug, please report it.', probe-rs/src/flashing/builder.rs:250:25
```

Reproduced with a nRF52840 DK board using the master branch of probe-rs (Commit ID: `2445d24ce58788d658e1f51c1fffc25b1aac9064`).

The same elf executable can be flashed with version `0.10.1` of probe-rs.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect nRF52840 DK
2. Run the following command to trigger the issue:
```
RUST_LOG=debug cargo run --bin=probe-rs-cli -- download --chip nRF52840_xxAA ../ate_repo/build/targets/stack_overflow_app
```

**Expected behavior**
The ELF program can be flashed without panicking.

**Stacktrace**
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `target/debug/probe-rs-cli download --chip nRF52840_xxAA ../ate_repo/build/targets/stack_overflow_app`
 DEBUG jaylink > libusb 1.0.23.11397
 DEBUG jaylink > libusb has capability API: true
 DEBUG jaylink > libusb has HID access: true
 DEBUG jaylink > libusb has hotplug support: true
 DEBUG jaylink > libusb can detach kernel driver: true
 DEBUG jaylink > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x1015,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
 DEBUG jaylink > scanning 4 interfaces
 DEBUG jaylink > J-Link interface is #2
 DEBUG probe_rs::probe::daplink::tools > Device Bus 006 Device 001: ID 1d6b:0003
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 005 Device 001: ID 1d6b:0002
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 001: ID 1d6b:0003
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 003: ID 0b05:18f3
 DEBUG probe_rs::probe::daplink::tools > No device matches
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 024: ID 1038:12b3
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 023: ID 1366:1015
 DEBUG probe_rs::probe::daplink::tools > No device matches
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 001: ID 1d6b:0002
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 001: ID 1d6b:0003
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 003: ID 8087:0029
 DEBUG probe_rs::probe::daplink::tools > No device matches
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 002: ID 05e3:0610
 DEBUG probe_rs::probe::daplink::tools > Device open error Access
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 039: ID 413c:2113
 DEBUG probe_rs::probe::daplink::tools > No device matches
[... 241 lines cut ...]
 INFO  probe_rs::flashing::download                         > Matching section: ".text"
 INFO  probe_rs::flashing::download                         > Matching section: ".ARM.exidx"
 INFO  probe_rs::flashing::download                         > Found loadable segment, physical address: 0x00000984, virtual address: 0x20000000, flags: 0x6
 INFO  probe_rs::flashing::download                         > Matching section: ".data"
 INFO  probe_rs::flashing::download                         > Matching section: ".uninit"
 INFO  probe_rs::flashing::download                         > Found loadable segment, physical address: 0x00000974, virtual address: 0x00000974, flags: 0x4
 INFO  probe_rs::flashing::download                         > Matching section: ".ARM.exidx"
 INFO  probe_rs::flashing::loader                           > Found 3 loadable sections:
 INFO  probe_rs::flashing::loader                           >     Multiple sections at 00000000 (2436 bytes)
 INFO  probe_rs::flashing::loader                           >     Multiple sections at 00000984 (1256 bytes)
 INFO  probe_rs::flashing::loader                           >     .ARM.exidx at 00000974 (16 bytes)
 DEBUG probe_rs::flashing::loader                           > Adding data: ExtractedFlashData { name: [".root_section", ".text", ".ARM.exidx"], address: 0, data: "[..] (2436 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000000 with size 2436 bytes
 DEBUG probe_rs::flashing::loader                           > Adding data: ExtractedFlashData { name: [".data", ".uninit"], address: 984, data: "[..] (1256 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000984 with size 1256 bytes
 DEBUG probe_rs::flashing::loader                           > Adding data: ExtractedFlashData { name: [".ARM.exidx"], address: 974, data: "[..] (16 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000974 with size 16 bytes
thread 'main' panicked at 'Overlap in data, address 0x00000974 was already written earlier. This is a bug, please report it.', probe-rs/src/flashing/builder.rs:250:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**Desktop (please complete the following information):**
 - Linux Ubuntu, 20.04 LTS

**Additional context**

Yatekii, 2021-03-14:
I do not have my DK with me today, but I can grab it from the office tomorrow. Could you maybe provide the test ELF? I have a theory what could be wrong. Could you try with this commit: https://github.com/probe-rs/probe-rs/commit/8159a2c938a737ddad278a0df62f472384aee5e6?

Javier-varez, 2021-03-14:
Thanks for the quick reply! I just tested with https://github.com/probe-rs/probe-rs/commit/8159a2c938a737ddad278a0df62f472384aee5e6, but unfortunately the problem is still reproducible there. Here's the elf file:
https://github.com/Javier-varez/stack_overflow_dwt_protection/suites/2254698361/artifacts/46906242

Yatekii, 2021-03-14:
Oh, I actually would have found the culprit in your debug output before too:

```
> Adding data: ExtractedFlashData { name: [".root_section", ".text", ".ARM.exidx"], address: 0, data: "[..] (2436 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000000 with size 2436 bytes
> Adding data: ExtractedFlashData { name: [".data", ".uninit"], address: 984, data: "[..] (1256 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000984 with size 1256 bytes
 DEBUG probe_rs::flashing::loader                           > Adding data: ExtractedFlashData { name: [".ARM.exidx"], address: 974, data: "[..] (16 bytes)" }
 DEBUG probe_rs::flashing::loader                           > Adding data at address 0x00000974 with size 16 bytes
 ```

0x00000000 + 2436 == 0x00000984. We can also see that it tries to include the `.ARM.exidx` section twice. Why that happens is a mystery to me. Do you know what that section does? I have never heard of it before.

Also, why this could have changed is because we went from goblin to object for parsing the ELF.

Javier-varez, 2021-03-14:
That is quite interesting, As far as I know 'ARM.exidx' is used for stack unwinding on C/C++ in ARM processors. I build with -fno-exceptions and -fno-rtti so this section is almost empty. 

It seems probe-rs is trying to load a segment that is not marked as loadable:

```
$ arm-none-eabi-readelf -l build/targets/stack_overflow_app
Elf file type is EXEC (Executable file)
Entry point 0x891
There are 4 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x001000 0x00000000 0x00000000 0x00984 0x00984 R E 0x1000
  LOAD           0x002000 0x20000000 0x00000984 0x004e8 0x004e8 RW  0x1000
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0
  EXIDX          0x001974 0x00000974 0x00000974 0x00010 0x00010 R   0x4

 Section to Segment mapping:
  Segment Sections...
   00     .root_section .text .ARM.exidx
   01     .data .bss .uninit
   02
   03     .ARM.exidx
```

I have created a change with a potential fix. Thanks for looking into this!

Yatekii, 2021-03-14:
Yup, this has happened in the goblin -> ELF port and I have not spotted this in the review. I am very sorry!

commit 0f913f64 (2021-03-14): Only load segments with type PT_LOAD

We need to make sure that the only loadable sections are actually
downloaded to the core. This check seems to be missing currently.

Fixes #581
commit 6a086f11 (2021-03-14): Merge #582

582: Only load segments with type PT_LOAD r=Yatekii a=Javier-varez

We need to make sure that the only loadable sections are actually
downloaded to the core. This check seems to be missing currently.

Fixes #581
PR #582 (2021-03-14): Only load segments with type PT_LOAD
We need to make sure that the only loadable sections are actually
downloaded to the core. This check seems to be missing currently.

Fixes #581
closed 2021-03-14 by bors[bot]