probe-rs/probe-rs #593: stlink v3 mini locks up after first run

evils, 2021-04-15:
**Describe the bug**
ctrl-c on probe-run leaves stlink v3 locked up

**To Reproduce**
Steps to reproduce the behavior:
1. probe-run via stlink v3 mini
2. ctrl-c
3. probe-run again
4. observe failure

**Expected behavior**
a subsequent run after stopping probe-run with `ctrl-c` to work

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - Linux (nixos-unstable)

**Additional context**
project in question: [here](https://gitlab.com/evils/waad/-/tree/54ad99dfc9da3a5f56c163f24b8a2ddddeaf1332) (src/bin/test.rs)
first run of `probe-run` (seems to work), ctrl-c left stlink blinking red/green [probe-run_stlinkv3_lockup_first_log.txt.gz](https://github.com/probe-rs/probe-rs/files/6320743/probe-run_stlinkv3_lockup_first_log.txt.gz)
second run, fails [probe-run_stlinkv3_lockup_second_log.txt](https://github.com/probe-rs/probe-rs/files/6320744/probe-run_stlinkv3_lockup_second_log.txt)

i was directed to open an issue here from `#rust-embedded` by @dirbaoi

evils, 2021-04-23:
any hope of getting some help with this?
i'm only doing a hobby thing, but this is a bit of a blocker for progress

evils, 2021-04-28:
@Dirbaio cargo-embed works, so i suspect this is a probe-run issue rather than probe-rs

Yatekii, 2021-04-28:
This is fixed on master. `probe-run` does not build off master.

evils, 2021-04-28:
interesting, do you know what the issue was?

Yatekii, 2021-04-28:
I cannot tell specifically, but handling of USB devices is tricky an APIs are easy to get wrong. If I remember correctly we made some changes that fixed this. It was a problem with ST-Links in general.

evils, 2021-04-28:
cargo-embed v0.8.0 works as well
though i suppose it's entirely possible the bug got introduced after that

evils, 2021-04-29:
upgrading my stlink v3 from V3J3M2 to V3J6M2 to V3J7M3 (their utility is annoying..., i didn't try on the intermediate version) didn't help

upgrading my stlink v2 from V2J17S0 to V2J37S0 did enable my use of that one in both cargo-embed and probe-run without issue :D

evils, 2021-04-29:
the behaviour did change with the firmware update
the first use of `cargo run` results in a successful flashing and connection to RTT (with the probe blinking red-green throughout)
ctrl-c after `cargo run` (with probe-run) results in
```
^CError: An error with the usage of the probe occured

Caused by:
    0: USB Communication Error
    1: Operation timed out
```

and leaves the probe's LED on solid green (previously blinking red-green)

a second `cargo run` invocation result in a brief red-green blinking followed by solid green and
```
Error: USB Communication Error

Caused by:
    Operation timed out
```

a third run of `cargo run` however results in the same as the first (successful flashing and RTT connection but bad stopping)

(this is all with probe-run version 0.2.1)

is there a way to move this issue to probe-run?

Tiwalun, 2021-04-29:
I think this repo is the right place for it. In the end, probe-rs is responsible for the USB communication, so we should be able to recover from USB issues. 

Not sure if we can do anything about the first error, but we should probably try harder to reset the USB device. With that we should be able to fix the second error.

Yatekii, 2021-04-29:
Well this is an old issue. Newer versions of probe-rs seem to fix this. So it would be a probe-run isse. I had this issue with older versions too. It definitely works for me with the ST-Link v3 here and master cargo-embed.