probe-rs/probe-rs #595: Improve build and install instructions for Windows.

perlindgren, 2021-04-18:
In the README.md it is described how you can compile probe using `vcpkg`. I did not read that first (my bad), instead I installed the visual studio community edition with the suggested packages for working with Rust. Admittedly I'm a complete novice to Win10 programming, so even if I eventually "happened" to get it compiling I think the process could be better explained for Win10 novices.

At this point I while is was able to compile `probe` it still did not work at run-time.

```
Error: Probe could not be created

Caused by:
    0: Entity not found
    1: Entity not found
```
As I understand, the problem at hand here is that Win10 does not provide a built in driver for the usb device (in this case it was a Nucleo with an on board `stlink` programmer). After googling around I found a working solution by installing `WinUSB` using the Zadig tool. 

Also here I'm not really sure what is going on, but it seems that Win10 associates a single driver to a specific usb device (in my case it was none was installed at that point.)

All in all, it was not terrible, I got it working (as far as I can tell) in the end. Nevertheless, if somebody with better understanding could have a look at the text regarding the compilation under Win10 and also add a section regarding the Win10 driver requirement (and installation process), it would be great for Win10 novices like me.

The text does not need to be that elaborate I think, perhaps just a set of links to appropriate sources for further reading.

/Per

Yatekii, 2021-04-28:
Thanks for reporting this, Per.

The information you are/were looking for is here: https://probe.rs/guide/2_probes/j-link/ .
The question is how we can make this more visible. Do you have any idea? Should we put this into the README or maybe make the guide more prominent in the READMEs.

perlindgren, 2021-04-29:
First of all its not a j-link specific problem, it is general to whatever programmer that you use with `probe`, so I guess its better to place it elsewhere in the doc. 

Then its also a general matter of compiling `probe` under win10. I got it working somehow, but I don't have a clear picture on what I did and why, what alternatives there are regarding tool-chains, what the pros/cons are and so forth. It would be very useful to newcomers (and for developers lacking Win10 experience) to provide informative links to guides/tutorials for Rust/Win10, as well as to highlight the `probe` specific dependencies (and how `probe` interfaces to the underlying libraries), to what end the choice of environment and toolchain matters (wsl, wsl2, msys, mingw, native microsoft through visual studio, etc. etc., to me its a jungle very hard to understand/navigate). 

If there is no good source of information regarding tool-chain options for Win10, maybe it would make sense to do such a write-up (not necessarily as part of the `probe` doc, but could be a separate repo (README.md), or a blog, or something else.

PR #2077 (2024-01-23): Remove libftdi & libusb
Blocked on #2075

Closes #595 (`cargo install probe-rs --features cli` should be all it takes on Windows, too)
closed 2024-01-23 by bugadani