probe-rs/probe-rs #599: Black Magic Probe support

mryndzionek, 2021-04-24:
I would like to have [Black Magic Probe](https://github.com/blacksphere/blackmagic) support to use all the auxiliary goodness probe-rs provides, like defmt via RTT. I'm willing to start adding the support, but I need some clarifications. BMP talks GDB protocol directly on a USB serial port. To connect to ARM chips I need a `try_get_arm_interface` and that means implementing `ArmProbeInterface`. This trait however needs some low-level stuff that I won't have access to via BMP. Maybe I should forget about implementing `DebugProbe` and somehow transparently relay GDB commands from GDB server directly?

Yatekii, 2021-04-28:
Hi!

The BMP is not fit for this task. It does everything we do in probe-rs on board. Which has several pluses and several minuses.
We chose to not go that route.
It is not possible to implement the necessary interfaces for low level operations. You could try and implement the higher level ones. But I think it's fundamentally incompatible.
What hinders you from using an aequivalent probe to the BMP if you want to use probe-rs? :)

GDB is terrible when it comes to it's protocol and it's documentation. You can go down that route, but it's an agony; speaking from experience.

mryndzionek, 2021-04-29:
Okay, fair enough. I was suspecting that this route might be very difficult and is not worth the effort. Thanks for the response.

closed 2021-04-29 by mryndzionek

mfiumara, 2023-09-05:
Hi, I was trying to figure out to get the BMP to work specifially with defmt over RTT. In case anyone else finds this issue it's possible to directly pipe the RTT output of the BMP into `defmt-print` like so:
```
(stty speed 115200 >/dev/null && cat) </dev/tty.usbmodem98B724953 | defmt-print -e target/thumbv8m.main-none-eabihf/debug/my-app
```

Thanks to James Munns for the solution:
"The reason for the tricky stty and piping stuff in the first half is to avoid cat line-buffering, which doesn't make sense with the binary data output by defmt over rtt" (https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org/$-SFtETYL7wZX_HIp2LQzlJ__XfoqwwEzkjdsfzS0omo?via=matrix.org&via=catircservices.org&via=tchncs.de)

mfiumara, 2023-09-11:
Hi @Yatekii,

I was looking at this again and I believe the BMP support could be added without having to interface with GDB.

The BMP has an option to be built as a hosted application (see https://black-magic.org/hardware.html#black-magic-debug-app): in this way the GDB server is built into an application that is ran on the host. The host application then talks to the BMP using another protocol (bmp_remote, see here https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/hosted/bmp_remote.h). This is a more lower level API and I believe if this protocol were to be implemented the BMP could be fully supported by probe-rs.

I think the following would need to be done to achieve this:
- Be able to recognize a BMP probe when listing serial devices
- Implement the bmp_remote protocol
- Implement the necessary traits like ArmProbeInterface

I'm open to help getting this to work but it seems like quite a significant piece of work, so before I start would like to know if this sounds like a good approach to you?
 co

Yatekii, 2023-09-27:
This sounds great! I would love to go GDB free where possible.

While I really don't like GDB I try to go back a bit on my hard stance on it to be more open for more probes and devices.

Someone else implemented a GDB based interface here: https://github.com/probe-rs/probe-rs/pull/746/files. I wonder if you could use the same. Ofc the PR needs some cleanup before others could use it.

Either way I am happy to get BMP support. I prefer the GDB free approach, but we can support both :)

chrysn, 2024-04-19:
> more lower level API

It may make sense to coordinate a bit with the blackmagic team here, to assess whether that API is supposed to be stable enough for this use. (It's not like we can't just update probe-rs when that interface changes, but it'll still helpful to have awareness of each other on both sides).

> Either way I am happy to get BMP support

Should this issue be re-opened, then?

Yatekii, 2024-04-19:
If someone wants to actively work on it. If not, I would like to keep it closed :)

mfiumara, 2024-04-21:
For everyone looking this, I have had some discussions with the guys at bmp's discord and did some initial work here:

- https://github.com/mfiumara/blackmagic-sys -> Rust bindings for the bmp protocol
- https://github.com/probe-rs/probe-rs/pull/2397

The amount of work involved here and knowledge required on the bmp protocol is too much for me to do on my free time so I have decided not to continue work on this. If anyone is interested in picking this up further feel free to continue from here.

xobs, 2024-09-02:
I've started work on this at https://github.com/xobs/probe-rs/tree/bmp-support

It works as a low-level probe, and once it's merged we can start working on adding support for things like `DapAccess` to improve performance.