probe-rs/probe-rs #608: Using `gdbstub` for enhanced GDB server functionality?

daniel5151, 2021-05-05:
_Disclaimer:_ I am the author of the `gdbstub` crate, so this is a totally shameless plug.

Right at the tail end of one of Ferrous Systems most [recent blog posts](https://ferrous-systems.com/blog/gdb-and-defmt/#gdb-in-probe-run), it mentioned that `probe-rs`'s GDB server functionality is somewhat immature.

Might I suggest swapping out `probe-rs`'s home-grown GDB Remote Serial Protocol implementation with the [`gdbstub`](https://github.com/daniel5151/gdbstub) crate? It implements substantially more of the GDB RSP than `probe-rs`'s current in-tree implementation, and is already being used by [several other projects](https://github.com/daniel5151/gdbstub#real-world-examples).

I wasn't sure if `gdbstub` was on your radar, so I thought I'd file an issue just in case!

Oh, and please let me know if there are any glaring issues with the `gdbstub` API / functionality that would preclude it from being used in `probe-rs`. I'm very open to feedback, since my long-term goal is to make `gdbstub` the "go to" GDB RSP crate in the Rust ecosystem!

Tiwalun, 2021-05-06:
Thanks for reaching out!

Using gdbstub for the protocol sounds interesting, and would make our life easier.

From a quick look at the API, it seems that the `Target` trait is usually implemented for a single architecture. The architecture used in our GDB server is only determined at runtime, is it possible to implement this with `gdbstub`?

daniel5151, 2021-05-06:
Thanks for the quick response!

Given that the most common use-case of `gdbstub` up until now has been as part of full-system-emulators and/or hypervisors, the fact that `Target` was tied to a single `Arch` hasn't been too much of an issue. In fact, by having this tight compile-time coupling, the `gdbstub` API is able to abstract a way a lot of the underlying protocol's warts and details, which makes it a lot more ergonomic to use. For example, the `Target`-level [`read_addr`](https://docs.rs/gdbstub/0.4.5/gdbstub/target/ext/base/singlethread/trait.SingleThreadOps.html#tymethod.read_addrs) and `write_addr` handers don't expose the raw register data byte-string, instead having `gdbstub` quietly marshal/unmarshal the data into much more user-friendly arch-dependent structs, which makes reading/writing registers [a breeze](https://github.com/daniel5151/gdbstub/blob/dev/0.5/examples/armv4t/gdb/mod.rs#L109-L124).

That said, you're not the first to bring up the need for more dynamic `Arch`/`Target` implementations! In fact, I just popped open a tracking issue on daniel5151/gdbstub that summarized a past discussion I had about this exact problem: https://github.com/daniel5151/gdbstub/issues/53

In a nutshell, the current `gdbstub` API is in-fact flexible enough to support runtime configurable `Arch`/`Target` implementations, albeit with some caveats. 

Most of these caveats are relatively minor perf-related issues (which should be totally acceptable unless you _really_ care about squeezing every last drop of performance out of the GDB RSP). The most unfortunate caveat is that you'll be losing those ergonomic benefits I mentioned above, and will require that GDB raw register data is de/serialized at the `target` level. Not ideal, but without a substantial rewrite of the `Arch` API (which is not out of the question!), that's pretty much unavoidable.

bet4it, 2022-04-23:
> From a quick look at the API, it seems that the `Target` trait is usually implemented for a single architecture. The architecture used in our GDB server is only determined at runtime, is it possible to implement this with `gdbstub`?

I use `gdbstub` on multi architectures which is determined at runtime in my project [udbserver](https://github.com/bet4it/udbserver). You could check the [code](https://github.com/bet4it/udbserver/tree/master/src/reg) to see how to implement it.

commit 911aaa08 (2022-06-07): Port gdb-server to use the gdbstub crate

* Replace existing GDB parsing logic with gdbstub
* Add support for multi-core debugging
* Fix session setup for multi-core

Closes #608
commit 9b717373 (2022-06-07): Merge #1125

1125: Port gdb-server to use the gdbstub crate r=Yatekii a=thefaxman

* Replace existing GDB parsing logic with gdbstub
* Add support for multi-core debugging via the multi-threaded gdbstub interfaces
* Fix session setup to run DebugCoreStart on each core instead of just the first one

Closes #608
PR #1125 (2022-06-07): Port gdb-server to use the gdbstub crate
* Replace existing GDB parsing logic with gdbstub
* Add support for multi-core debugging via the multi-threaded gdbstub interfaces
* Fix session setup to run DebugCoreStart on each core instead of just the first one

Closes #608
closed 2022-06-07 by bors[bot]