probe-rs/probe-rs #617: Support Raspberry Pi Pico

ctron, 2021-05-11:
**Is your feature request related to a problem? Please describe.**

I would like to see support for the Raspberry Pi Pico.

**Describe the solution you'd like**

`cargo embed` for the Raspberry Pi Pico.

**Describe alternatives you've considered**

I guess there are other ways, using `probe-rs` is just plain simple, has a nice ecosystem, and JustWorks™ … but not for the Pico.

**Additional context**
Add any other context or screenshots about the feature request here.

Tiwalun, 2021-05-11:
There is a probe-rs fork which supports the Pico: 

https://github.com/rp-rs/probe-rs

Based on that fork, there is also a [probe-run](https://github.com/knurling-rs/probe-run) variant:

https://github.com/rp-rs/probe-run

The goal is to merge these back upstream, but for now you can use them to work with the Pico.

ctron, 2021-05-12:
This is good to know, thanks! But I am fine, waiting a few more days :)

Dirbaio, 2021-08-20:
Solved by #720

closed 2021-08-20 by Dirbaio