probe-rs/probe-rs #627: Allow debug stepping past hardware AND software breakpoints

noppej, 2021-05-14:
**Describe the bug**
Currently, a step and/or run instruction will get stuck on breakpoints. 

**Expected behavior**
Once a breakpoint is hit, the user should be able to step/run past that instruction without having to unset the breakpoint.

**Additional context**
I have implemented a partial solution for hardware breakpoints in `probe-rs/src/architecture/arm/core/m4.rs`, by disabling the hardware breakpoints, then stepping one instruction, then re-enabling the hardware breakpoints, before executing the user's run instruction. 

@Tiwalun has correctly pointed out that this will not work for software breakpoints.  Also, it should be implemented and tested for all supported architectures.

Tiwalun, 2021-05-15:
This is not really fixed by #629, software breakpoints are still not handled correctly.

noppej, 2021-05-15:
> This is not really fixed by #629, software breakpoints are still not handled correctly.

Understood, but it does help with making hardware breakpoints better ... thanks for that fix :)

noppej, 2021-05-16:
I am de-prioritizing the support of SW breakpoints for myself, while working on supporting the evaluation of additional data types in the debugger.   If no one picks this issue up, I will get back to it later.

sszilvasi, 2022-03-26:
@Tiwalun and @noppej: I have recently been experimenting with RAM-only projects and [ran into issues with `probe-run`](https://github.com/knurling-rs/probe-run/issues/160#issuecomment-1077717595). Presumably, those are due to the lack of support for _software_ breakpoints in `probe-rs`.

Could you summarize the current state of the _software_ breakpoint implementation in `probe-rs` and provide references to any related prior and ongoing work?

noppej, 2022-03-26:
@sszilvasi ... have a look at PR #1056 ... hat is the latest work I have ongoing in this are.  It uses hardware breakpoints to implement stepping at statement level.  

I don't believe this will help what you are looking for.

noppej, 2022-03-27:
@sszilvasi When I was looking into software breakpoints originally, I found [this very useful article](https://interrupt.memfault.com/blog/cortex-m-breakpoints#software-breakpoint). 

From what I understand of the probe-rs code, it should be reasonably straightforward to add this functionality, and not that hard to warn the user about situations where they won't work. If someone wants to PR this, and has knowledge of RISCV, then I would be happy to work with them on the ARM implementation.

sszilvasi, 2022-03-28:
@noppej That article is indeed very well written.

Does "reasonably straightforward" translate to "good first task"? 🙂 

Given that I am still trying to come up to speed with both `probe-rs` and Rust itself, the only help I can confidently offer at this point is testing the ARM implementation.

Also, is the RISC-V implementation out-of-the-box strictly necessary?

noppej, 2022-03-28:
@sszilvasi Yes, I think it will be good learning opportunity and if you think you are open to invest future contributions to probe-rs, then I'd be willing to help. If you want,  DM me on @jack-n:matrix.org so that we can discuss in more detail.

@Tiwalun What do you think? If we have an ARM-only PR that allows software breakpoints for RAM code, would you accept that? Or do you know someone who understands RISCV enough that can collaborate with us?

Tiwalun, 2022-03-29:
Just doing it for ARM is fine, for RISCV I would assume that it's more or less the same anyway, just with a different breakpoint instruction.