probe-rs/probe-rs #630: Invalid Error: 'Unable to set hardware breakpoint, all available breakpoint units are in use'

noppej, 2021-05-15:
**Describe the bug**
When setting and clearing breakpoints, a few changes will eventually result in `Unable to set hardware breakpoint, all available breakpoint units are in use` error, even though fewer than the max (8 in my case) are being set.

** Discussion **
I traced this down to the fact that the `Core::state::breakpoints` is not cleared by [clear_all_hw_breakpoints()](https://github.com/probe-rs/probe-rs/blob/92475db67728bbf4924f12351fb1f58d4abd11a7/probe-rs/src/core/mod.rs#L470)

Before I fix it, I'd like to understand why&if `Core::state::breakpoints` vector is necessary. Is there something that that can't be done without it, or can we replace it with hardware queries, in favour of simplicity and reliability?

** Additional Question**
I noticed that  [clear_all_hw_breakpoints()](https://github.com/probe-rs/probe-rs/blob/92475db67728bbf4924f12351fb1f58d4abd11a7/probe-rs/src/core/mod.rs#L470) and other methods call `self.inner.clear_breakpoint()`, while there is also  a local [clear_hw_breakpoint()](https://github.com/probe-rs/probe-rs/blob/92475db67728bbf4924f12351fb1f58d4abd11a7/probe-rs/src/core/mod.rs#L443) - this has a direct impact on the lack of maintenance of the `Core::state::breakpoints`. Is this a redundancy that I can cleanup, or is this there for a purpose?

Tiwalun, 2021-05-15:
The breakpoints in `Core::state::breakpoints` are only tracked to reduce the amount of queries we have to do.

I think it's a good idea to remove the list breakpoints to increase reliability. If we discover later that the performance suffers too much, we can add some caching again. For now reliability is more important.

I don't think the redundancy is there on purpose.

noppej, 2021-05-15:
@Tiwalun ... I have an additional question. `CoreState` only has an `id` and `breakpoints`, so it would seem to make sense to remove `CoreState` as well ... BUT ... that would alter `Core` API in a breaking way. Any recommendations?

Tiwalun, 2021-05-15:
I think the `id` is still useful to have, so I wouldn't remove the `CoreState`.

noppej, 2021-05-15:
@Tiwalun : Additional question. [set_breakpoint()](https://github.com/probe-rs/probe-rs/blob/a02cf5d4d641edf65e98b2a54c31339a3a85e3d8/probe-rs/src/architecture/arm/core/m4.rs#L592) will set a hw breakpoint, irrespective of whether one already is set at a different index. Should incur the register read overhead to look for existing hw breakpoints at the same address before we set the new one?

noppej, 2021-05-15:
@Tiwalun and @Yatekii  ... I am not sure who to ask, but I've fixed this for the m4.rs. It requires a trait `CoreInterface::get_hw_breakpoints` to be implemented for the other architectures. I can 'guess' my way around, but have no way of testing these. Is there someone who can help me with those?

Yatekii, 2021-05-15:
We can test it for M0 for sure, M33 is trickier as it is not well tested anyways. The linked branch does not seem to contain any commits for this tho?

noppej, 2021-05-16:
@Yatekii  ... the commit is there now. The new trait has an implementation for `m4.rs`, and a default `todo!()` implementation for the other architectures.

Tiwalun, 2021-05-16:
@noppej Can you please make a PR? Can just be as a draft, but that would make it easier to discuss, then we can also add comments to the changes directly.

closed 2021-06-02 by bors[bot]