probe-rs/probe-rs #634: Support for flashing read protected flash

azerupi, 2021-05-18:
**Is your feature request related to a problem? Please describe.**
When setting the read protection level 1 (rdp = 1) on stm32 chips, it makes flashing them with probe-rs / probe-run fail. 

Segger's JLink and openocd are both able to detect that the flash is read protected. Segger's JLink si the only flashing software I used that can erase and flash the readout protected chips.

[image]

With openocd and the particular chip I am using I get:
```
Info : STM32L flash size is 64kb, base address is 0x8000000
Error: Flash protection is not supported.
```

**Describe the solution you'd like**
It would be great if probe-rs could detect readout protection and set the rdp level to 0 so that flashing is possible.

Yatekii, 2021-05-18:
Thanks for opening a tracking issue for this :)

This will be possible, once we have sequences implemented. Unfortunately it's not easy to get them right, so they have not landed yet and I think it will take another while.

azerupi, 2021-05-18:
I'm glad to hear this is on the roadmap! :+1: 

>once we have sequences implemented

Is that what is proposed in #242 ? Has there been any progress since the last comment on that issue?

Yatekii, 2021-05-18:
Yes, basically that.

A concept implementation has been started in #483 which implements the sequences given in the standard by ARM: https://arm-software.github.io/CMSIS_5/Pack/html/debug_description.html#pdsc_SequenceNameEnum_pg it requires some changes to the probe-rs internals and all of them are kinda unsatisfying so far, which is why it makes slow progress.
The idea here is that we implement the ARM defined ones and define our own on top for functionality we need additionally and is not in the standard yet.

azerupi, 2021-05-19:
Thanks for the information!

I'm leaving some extra information for future reference:
In order to unlock the flash you need to reset the read-out protection to level 0. This is done by writing specific values into the option bytes. In the case of the `stm32l0xx` this is `0xaa 0x00 0x55 0xff`. This triggers a mass erase and after that you can flash the chip as usual.

At the moment I found a workaround that works for me, I use `st-flash` to unlock the flash and then I can use probe-rs / probe-run

```
$ xxd option_bytes_unlock.bin
00000000: aa00 55ff                                ..U.

$ st-flash write option_bytes_unlock.bin 0x1FF80000
st-flash 1.7.0
2021-05-19T12:31:29 INFO common.c: L0x3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
file option_bytes_unlock.bin md5 checksum: 297be22d5c6b0a198ee68999330c6b9, stlink checksum: 0x000001fe
2021-05-19T12:31:29 WARN common.c: Writing option bytes 0xff5500aa to 0x1ff80000
2021-05-19T12:31:29 INFO common.c: Wrote 4 option bytes to 0x1ff80000!
2021-05-19T12:31:29 INFO common.c: Go to Thumb mode

$ probe-run --chip stm32l0 target/thumbv6m-none-eabi/release/firmware
  (HOST) INFO  flashing program (18.71 KiB)
  (HOST) INFO  success!
```

I took a look at the code of probe-rs and found the [stm32l0xx_opt](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/STM32L0%20Series.yaml#L2708) flashing algorithm. This references the address `536346624` which is `0x1ff80000`. So I assume that it does something with the option bytes?

Maybe this could be adapted to write the values to the option byte? I don't know if this would have any negative impact on chips that are already unlocked. Anyways, I have no idea how the code actually works so this might be a really stupid idea :sweat_smile: 

## Screenshots from the [reference manual](file:///tmp/mozilla_azerupi0/dm00108282-ultralowpower-stm32l0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf):
[image]

[image]

Yatekii, 2021-05-19:
I am against unlocking automatically as this can lead to inconvenient experiences when you do not want to erase the entire flash. I'd add a sequence that does unlocking and be done with that :)

johannesneyer, 2024-02-22:
This can disable the readout protection for stm32wl chips: https://github.com/newAM/stm32wl-unlock