probe-rs/probe-rs #655: Failed to flash nrf9160 dk

Sheap, 2021-05-25:
When I try to flash anything to my nrf9160 dk using cargo-flash or other tools which use it (e.g. a knurling application template app), I receive the following error:

```
$ cargo flash --chip nrf9160_xxAA --target thumbv8m.main-none-eabihf
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
    Flashing /home/stuart/nrf-hal/target/thumbv8m.main-none-eabihf/debug/hello-world
     Erasing sectors ✔ [00:00:00] [----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]       0B/ 20.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]       0B/ 20.00KB @       0B/s (eta 0s )
       Error failed to flash /home/stuart/nrf-hal/target/thumbv8m.main-none-eabihf/debug/hello-world

Caused by:
    0: Error while flashing
    1: Something during memory interaction went wrong
    2: A core architecture specific error occured
    3: Failed to write register DRW at address 0x0000000c
    4: An error specific to the selected architecture occured
    5: Target device did not respond to request.
```

**To Reproduce**
Steps to reproduce the behavior:
1. clone https://github.com/nrf-rs/nrf-hal/tree/master
2. navigate to examples/hello-world
3. (it seems like this one shouldn't be required - but my compiler didn't like it. maybe I'm doing something wrong here) - remove the feature gates for the non-nrf9160 board, rewrite dependencies such that they work
```diff
--- a/examples/hello-world/Cargo.toml
+++ b/examples/hello-world/Cargo.toml
@@ -10,21 +10,8 @@ publish = false
 [dependencies]
 cortex-m = "0.6.2"
 cortex-m-rt = "0.6.12"
+nrf9160-hal = { version = "0.12.1", features = ["rt"] }

 [dependencies.embedded-hal]
 version = "0.2.3"
 features = ["unproven"]
-
-[dependencies.nrf9160-hal]
-features = ["rt"]
-path = "../../nrf9160-hal"
-optional = true
-
-[dependencies.nrf52840-hal]
-features = ["rt"]
-path = "../../nrf52840-hal"
-optional = true
-
-[features]
-9160 = ["nrf9160-hal"]
-52840 = ["nrf52840-hal"]
```

and

```diff
--- a/examples/hello-world/src/main.rs
+++ b/examples/hello-world/src/main.rs
@@ -3,9 +3,6 @@

 // Simple UART example

-#[cfg(feature = "52840")]
-use nrf52840_hal as hal;
-#[cfg(feature = "9160")]
 use nrf9160_hal as hal;

 use core::fmt::Write;
@@ -15,20 +12,6 @@ use hal::{gpio, uarte, uarte::Uarte};
 fn main() -> ! {
     let p = hal::pac::Peripherals::take().unwrap();

-    #[cfg(feature = "52840")]
-    let (uart0, cdc_pins) = {
-        let p0 = gpio::p0::Parts::new(p.P0);
-        (
-            p.UARTE0,
-            uarte::Pins {
-                txd: p0.p0_06.into_push_pull_output(gpio::Level::High).degrade(),
-                rxd: p0.p0_08.into_floating_input().degrade(),
-                cts: Some(p0.p0_07.into_floating_input().degrade()),
-                rts: Some(p0.p0_05.into_push_pull_output(gpio::Level::High).degrade()),
-            },
-        )
-    };
-    #[cfg(feature = "9160")]
     let (uart0, cdc_pins) = {
         let p0 = gpio::p0::Parts::new(p.P0_NS);
         (
```

**Expected behavior**
flash to complete without error

**Desktop (please complete the following information):**
Linux (manjaro)

**Additional context**
I am able to successfully flash to the board using the nrf-connect app, suggesting there is nothing wrong with the board itself. My colleague gets the same error on a separate board using a separate host.

Sheap, 2021-05-25:
It looks like this worked thanks!

For anybody with the same issue - the given command doesn't quite get you there. It gives the following error:
```
Caused by:
  no matching package named `probe-rs` found
  location searched: https://github.com/probe-rs/probe-rs
  prerelease package needs to be specified explicitly
  probe-rs = { version = "0.11.0-alpha.5" }
  required by package `cargo-flash v0.10.2
  ```
  
  however, cloning cargo-flash, and explicitly setting the dependencies to the prerelease versions, then `cargo install --path .` does the job.

Yatekii, 2021-05-25:
Uff, I am very sorry! We worked on automatic releases on the weekend and we messed something up. We will fix this asap, thanks for reporting!

Dirbaio, 2021-08-20:
Fixed in #619

closed 2021-08-20 by Dirbaio