probe-rs/probe-rs #665: Attaching to GD32VF103CBT6 occurs "The version '13' of the debug transport module (DTM) is currently not supported" error.

alekseysidorov, 2021-06-02:
**Is your feature request related to a problem? Please describe.**

The https://github.com/probe-rs/probe-rs/commit/2efc3fbaf1dcd3e3ad85672c1d8c2609c5c9566f version of `probe-rs` doesn't support Debug Transport Mode 13 of the `GD32VF103CBT6`:

```
$ RUST_LOG=trace cargo flash --release --chip GD32VF103CBT6 --connect-under-reset --protocol jtag

... some lines omitted ...

       DEBUG probe_rs::architecture::riscv::dtm     > Dtmcs: Dtmcs { .0: 268457533, idle: 5, dmistat: 1, abits: 35, version: 13 }
        INFO cargo_flash                            > Metadata {
    chip: Some(
        "Unspecified(\"GD32VF103CBT6\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "0",
    ),
    release: "0.10.2",
    commit: "a055665",
}
       Error 0: An error specific to the selected architecture occured
             1: The version '13' of the debug transport module (DTM) is currently not supported.
```

**Describe the solution you'd like**

I wonder how difficult it is to add support for this protocol version, or how I can switch to the supported DTM version?

Disasm, 2021-06-02:
I believe probe-rs supports this chip, at least debugging worked through an FTDI probe.

alekseysidorov, 2021-06-03:
I use JLINK programmer.

Tiwalun, 2021-06-04:
I think this is most likely a bug in our JTAG implementation for the J-Link. Unfortunately, it is not the same as for the FTDI based probes. 

The error message is a bit misleading, but the DTM version 13 does not make any sense, the only allowed versions according to the spec are 0 and 1. See section 6.1.4 of the [RISCV Debug Spec].

[RISCV Debug Spec]: https://raw.githubusercontent.com/riscv/riscv-debug-spec/release/riscv-debug-release.pdf

alekseysidorov, 2021-06-04:
I.e my probe reports a weird DTM version bit, therefore `probe-rs` doesn't know how to deal with it. Unfortunately, I use the Chinese JLINK clone. So, I can try to disable version check and see what happens.

Tiwalun, 2021-06-04:
The version check is not the problem. I now rembember why this is not working.

For the FTDI probes, we support multiple TAPs in a JTAG chain, and can select the appropriate one for the `GD32VF103CBT6`.  For J-Link probes (original and clones), we don't support this yet. This means that we don't support that chip with J-Link right now.

To fix it, we have to add support for JTAG chains to our J-Link probe implementation.

rmsyn, 2023-11-04:
> For the FTDI probes, we support multiple TAPs in a JTAG chain, and can select the appropriate one for the GD32VF103CBT6

I'm currently working on support for the `JH7110` family of SoCs, and experiencing the same issue (through an FTDI FT2232D probe). It uses an `U74` core complex, and `E24` supervisor core. I think it's reporting version `13` because the `U74` supports version `0.13` of the RISC-V Debug Specification (according to Chapter 14 of [U74 Core Complex Manual](https://starfivetech.com/uploads/u74_core_complex_manual_21G1.pdf)). Looks to be the same for GD32VF103 according to Chapter 10 of the [User Manual v1.2](https://dl.sipeed.com/LONGAN/Nano/DOC/GD32VF103_User_Manual_EN_V1.2.pdf).

I've got past the version check part, and will be adding the generalization for multiple DTM versions by adding the invariants for register addresses, offsets, etc.

Maybe some of the hardware manufacturers are using a `u16` scheme for versioning where `0x000d` means `v0.13`, which is being interpreted by the current parser as `13`? So, the major version would still be `0`, technically falling under spec?

Yatekii, 2023-11-04:
How does `0xd` correspond with the spec? It's a value that is not listed? But I am happy to make this more resilient in code :)

rmsyn, 2023-11-04:
> How does `0xd` correspond with the spec? It's a value that is not listed? But I am happy to make this more resilient in code :)

Loosely because the `1` bit is set, but looking more closely at the code, it would fall into the "Unknown" category. Also because this is present on a core released by SiFive makes me a little more generous to the implementation.

I am working on reworking parts of the code to split the register definitions now, also. There is a very large difference between the `v1.00` stable release of the debug spec the code looks to be based on, and the `v0.13` version of the spec.

Breaking out some of the data structures and matching against the `v0.13` version bits of the `dtmcs` register. One annoying thing is the backwards-incompatible register definitions for determining version.

The `dmstatus` register in `v1.00` is at address `0x11`, and the `dtmcs` register in `v0.13` is at `0x10`. The bitfield definitions are also different. So, other than doing a lookup based on known-supported debug protocol version, there doesn't seem to be a general, reliable way to determine protocol version from reading registers.

Yatekii, 2023-11-05:
Sounds good, thanks a lot! We can probe the different registers in succession I guess and take the first hit :)