probe-rs/probe-rs #666: TODO: Need to test and cleanup RISCV implementation of `get_hw_breakpoints()`

noppej, 2021-06-02:
**Describe the bug**
The PR #632 introduced a new trait `get_hw_breakpoints()` which is essential for setting new breakpoints. It requires further testing for the RISCV architecture.

@Tiwalun ... feel free to list additional challenges that you encountered during testing for that architecture.

Tiwalun, 2021-06-02:
Please avoid this in the future, master should always be in a working state. The whole point of PRs is that we can test stuff before it is merged..

closed 2021-06-02 by Tiwalun

Yatekii, 2021-06-02:
Actually it was my bad merging this. I did not understand that this would be in broken state. Sorry about that.

noppej, 2021-06-02:
> Actually it was my bad merging this. I did not understand that this would be in broken state. Sorry about that.

Same here. Without being able to test that platform it was hard to know, but I did not have reason to believe it would leave it in broken state. Sorry about that. 

PS. I'd be happy to invest in additional testing boards if we can agree on a 'basic' common set of boards required to validate core functionality.