probe-rs/probe-rs #682: Support for TI Stellaris In-Circuit Debug Interface (ICDI)

vipierozan99, 2021-06-05:
I have the [TM4C123](https://www.ti.com/tool/EK-TM4C123GXL) board here and have been testing out some Rust on it. I can flash it with lm4flash and debug it with OpenOCD/GDB.

It would be very cool to use probe-rs with it, but when I run `probe-rs-cli list` I get `No devices were found.`.
Running it with sudo `sudo $(which probe-rs-cli) list` gives the same result.

The board entry in `lsusb` is `Bus 001 Device 011: ID 1cbe:00fd Luminary Micro Inc. In-Circuit Debug Interface` and the logs from `dmesg` are the following: 

```
[29909.347464] usb 1-4: new full-speed USB device number 12 using xhci_hcd
[29909.475677] usb 1-4: New USB device found, idVendor=1cbe, idProduct=00fd, bcdDevice= 1.00
[29909.475685] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[29909.475688] usb 1-4: Product: In-Circuit Debug Interface
[29909.475691] usb 1-4: Manufacturer: Texas Instruments
[29909.475693] usb 1-4: SerialNumber: 0E2298FB
[29909.477768] cdc_acm 1-4:1.0: ttyACM0: USB ACM device
```

I tried researching terms like `SWD` and `JTAG` but I'm pretty inexperienced with hardware so that lead nowhere. Also, I managed to get the OpenOCD/GDB setup working a while ago so I don´t really remember my thought process.

But after digging, I realised that the OpenOCD config file for my board at `/usr/share/openocd/scripts/board/ek-tm4c123gxl.cfg` seems to include another file at `/usr/share/openocd/scripts/interface`, which contains the following comment:

```
#
# TI Stellaris In-Circuit Debug Interface (ICDI) Board
#
# This is the propriety ICDI interface used on newer boards such as
# LM4F232 Evaluation Kit - http://www.ti.com/tool/ek-lm4f232
# Stellaris Launchpad - http://www.ti.com/stellaris-launchpad
# http://www.ti.com/tool/ek-lm4f232
#
```

So this leads me to think that this family of boards use a different proprietary interface, not JTAG, but I don't really know.
So, knowing all this:

1. Can `probe-rs` implement support for this debug interface?
2. How hard would it be?
3. I could do a pull request but I have no idea on how to start implementing such a thing, so how can I help?

Yatekii, 2021-06-09:
Hi @vipierozan99!

1. I am not sure how exactly the ICDI looks in terms of the API, but if it implements basic operations such as reading/writing memory and perfoming CPU operations such as run/halt/step/breakpoints, it should be no issue to implement it :)

2. Hard to say. We are still actively fixing bugs and performance issues and perfroming general maintenance on the ARM interface which is working since over a year. Yet we still have a lot to implement. For RISC-V it's even more. So for a basic implementation, IF you have good docs on ICDI, could be a matter of a day or two for someone experienced. If you have no real docs but maybe a reference implementation it might take longer. If you have to reverse engineer it yourself, expect a lot of work.

3. We are very happy to give you advice on how to work on this and on probe-rs in general. Though, be aware that ICDI is not a priority for the active devs (if aynone is, please correct me :P) so we wont work on it ourselves for now most likely, as we have many open features and bugs already :)
Basically what you need to do is implement this trait: https://docs.rs/probe-rs/0.10.1/probe_rs/trait.DebugProbe.html for your debug probe.
We are also present on Matrix and can have a bit more of an interactive chat there if you'd like to :)
It would be very cool if you worked on this!

Best,
Noah

luksan, 2021-07-06:
Hi,

I have spent a few days looking at this, and while the ICDI interface supports debugging and flashing firmware it maps poorly to a JTAG interface. Thus, it is very hard to implement support in probe-rs for it since the DapAccess and SwoAccess traits cant be implemented.

The best reference to the interface is the openocd source code, look at the hla driver and ti_icdi_usb.c.

Yatekii, 2021-07-06:
It looks quite similar to the ST-Link interface.

Sadly the `ti_icdi_usb.c` is very badly documented, but from what I can judge, we can take the same approach as with the ST-Link. It wont be able to do sequences, but it should be possible with medium effort :)

Check: https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/stlink/mod.rs

luksan, 2021-07-06:
I have spent quite some time looking at the st driver, trying to map the ICDI to it. The biggest issue is that there is no way to read AP and DP registers, so the ArmProbeInterface can't be implemented as I understand it. I have made some notes about the avaliable ICDI commands at https://github.com/luksan/lm4flash-rs/wiki

Yatekii, 2021-07-07:
Yeah, indeed, that's troublesome :/

I think we would have to make a special case for this. The general Probe would work, but not the Core API.
We are currently in the process of simplifying a lot of things internally. Maybe this will be easier afterwards. Or harder :P

luksan, 2021-07-09:
Actually, it seems that the ICDI probe implements (a subset) of the GDB remote server protocol.
https://sourceware.org/gdb/onlinedocs/gdb/Remote-Protocol.html#Remote-Protocol

Perhaps a way forward would be to add GdbServer as a new type of ArchitectureInterface in session.rs. Then probe-rs could be used with any target that acts like a GDB remote debug server, including openocd I guess.

I could take a stab at this, but maybe I should wait if there is a lot of refactoring planned in this area?

Yatekii, 2021-07-09:
I don't think this is a good idea :/

The sole purpose of probe-rs is to provide a stable, single, well integrated, nicely API'd solution to talk to any probe or chip in the same fashion. Adding GDB to the mix does not mesh well with that. In fact, one of the main purposes is getting rid of GDB.

Why not add an ArchitectureInterface for ICDI instead? I am not sure this is the right solution, but GDB surely isn't. Is ICDI ARM specific or is it even another architecture it targes? I am not too familiar with TI prorietary things :/

We are not refactoring the archtecture entry point currently, so that would be good to go, but I will not just merge GDB in :/

luksan, 2021-07-09:
It's not gdb, just the protocol that gdb uses to communicate with remote debug servers. An architecture interface for ICDI would be an interface to a gdb remote debug target, so I'm saying that an interface that supports ICDI could be made more generic so it could be used with other devices that implement the protocol. See for example the gdbstub crate.

ICDI is a debug probe found on embedded ARM devkits from TI. Completely proprietary and undocumented, it seems, but it matches the GDB remote protocol. The debug probe controls the target via JTAG, but it does not expose the JTAG interface.

It would of course be easier to just consider the ICDI probe initially, if another debug probe with a similar interface is found later the code could be refactored at that point, if supporting generic gdb targets is out of scope for probe-rs.

adamgreig, 2021-07-09:
Feelings about even just GDB's protocol aside, I guess this could then also support the [blackmagic](https://github.com/blacksphere/blackmagic) probe which similarly uses the GDB protocol, over USB serial.

luksan, 2021-07-09:
@adamgreig I guess so. Interestingly, according to https://github.com/blacksphere/blackmagic/wiki the Blackmagic firmware can be flashed to the Stellaris ICDI, turning it into a Blackmagic probe.

luksan, 2021-07-24:
I created pull request #746 with support for the ICDI probe. I have tested that upload/download and `cli info` works with an LM4F120 eval board.