probe-rs/probe-rs #684: The new DownloadOptions API is confusing

Dirbaio, 2021-06-09:
you can either do the "struct style"

```rust
let mut opts =  DownloadOptions::new();
opts.keep_unwritten_bytes = true;
flash(opts)
```

or the "builder style"

```rust
flash(DownloadOptions::new().keep_unwritten_bytes(true));
```

IMO there should be only one way of doing things. Makes the API easier to understand, makes code easier to read since it all does it the same way.

Also, builders typically allow you to not use chaining, but not this one:

```rust
let mut opts =  DownloadOptions::new();
opts.keep_unwritten_bytes(true); // consumes instead of mutating
flash(opts) // opts is moved out
```

Dirbaio, 2021-06-09:
More on the line of "there is two ways of doing the same thing": `::new()` and `::default()` do the exact same thing. IMO there should be only `::default()`.

Yatekii, 2021-06-09:
Personally I prefer `new` over `default`, tho `default` would be the proper rustic style; so I am ok with having just `default`. Tho I see no harm in having `new` as an alias. Most libs do it afaik :)

As for the public fields vs builder style, yes, I believe, if we choose the builder style, we should not make the fields public. In any case, I wont be in favor of either.

Concerning the consuming setters: This is actually how a proper builder works and so do most libs. So you can chain all the builder methods which is the big adavantage of the pattern :) You'd use it like so:

```rust
flash(
    DownloadOptions::new()
        .keep_unwritten_bytes(true)
)
```

You can now see it's main advantage of using a PORS. Tho, with a PORS implementing default you could do:

```rust
flash(
    DownloadOptions {
        keep_unwritten_bytes: true,
        ..Default::default()
    }
)
```

which basically achieves the same. In the end it boils down to whether the options should be easily mutable or not.

To get back to the original point of this last discourse: If the builder had non-chainable methods (`&mut self`) it looses its entire purpose and it's not really a builder imo. Of course, Rust is kind of a special case here, because in GCd languages, you get both in one. With Rust, this is unfortunately not possible :)

Anyways, long story short, I do not really any problems with the builder here, but I also feel your opinion. I do not see what `non_exhaustive` has to do with this tho, as you always have to well-define all fields of a struct, even with `non_exhaustive`; or what bit of the story do I miss?

Dirbaio, 2021-06-09:
> Personally I prefer new over default, though default would be the proper rustic style; so I am ok with having just default. Though I see no harm in having new as an alias. Most libs do it afaik :)

No strong opinion on this one, I guess it's fine as long as `new` documents it returns the same as `default.

> Concerning the consuming setters: This is actually how a proper builder works and so do most libs. 

Builders usually take `&mut self` and return `&mut Self` so you *can* chain if you want to, but still have the option to not chain. If you're forced to chain, conditionally setting fields becomes super awkward, while it's super easy with a PORS.

```rust
let mut opts = DownloadOptions::new();
opts.keep_unwritten_bytes = true;
if nuke_it {
    opts.do_chip_erase = true;
}
flash(opts)
```

> I do not see what non_exhaustive has to do with this 

When rust didn't have `non_exhaustive`, the way to do "options" structs while allowing adding new fields in a backwards-compatible way was to do a builder. This is one of the reasons builders have become popular.

Now that `non_exhaustive` is a thing, a `non_exhaustive` PORS with default/new and public fields is more idiomatic IMO. Builders don't have any advantages. Being able to chain setters is a matter of taste, some like it, some don't. It's not intrinsically better.

commit 1316b839 (2021-06-21): Simplify DownloadOptions, fix #684
commit 2a4f6bf9 (2021-06-21): Merge #705

705: Simplify DownloadOptions, fix #684 r=Yatekii a=Tiwalun

Simplifying DownloadOptions, the lazy way ;).
closed 2021-06-21 by bors[bot]