probe-rs/probe-rs #703: Debugging when wfe is used doesn't appear to work

huntc, 2021-06-20:
I am unable to observe my code stopping at breakpoints when `wfe()` is used.

Steps to reproduce the behavior:

1. Clone the https://github.com/titanclass/microrust-start repo
2. Open the project in VS Code
3. Install the probe-rs debugger and VS Code extension
4. Set a breakpoint on the loop statement of the main.rs
5. Debug from within VS Code

I expect the debugger to stop at the breakpoint set. If I comment out the `wfe()` call then all is well i.e. I observe the debugger stopping at the breakpoint.

For further context, please note that if I debug via `pyocd` with the following configuration then all appears well:

```
{
    /* 
     * Requires the Rust Language Server (RLS) and Cortex-Debug extensions
     * https://marketplace.visualstudio.com/items?itemName=rust-lang.rust
     * https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug
     */
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (pyocd)",
            "servertype": "pyocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToMain": true,
            "executable": "./target/thumbv7em-none-eabihf/debug/microrust-start",
            "device": "nRF52833_xxAA",
            "boardId": "9904360259994e45001a401100000064000000009796990b",
            "targetId": "nrf52"
        }
    ]
}
```

Disasm, 2021-06-20:
Have you tried this? https://github.com/probe-rs/probe-rs/issues/350#issuecomment-740550519

huntc, 2021-06-20:
> Have you tried this? [#350 (comment)](https://github.com/probe-rs/probe-rs/issues/350#issuecomment-740550519)
> UPD: Oops, sorry, apparently you have a completely different chip.

Thanks for replying. Yes, different chip. Plus the other debugger works with no code change required.

noppej, 2021-06-21:
Hi @huntc  ... I will look into this today. Thanks for taking the time to create such a detailed bug report!

Yatekii, 2021-06-21:
Some hint maybe: I think other toolkits automatically set debug flags on the core which enable debugging on wfe/wfi. This would be part of the upcoming sequences as it is different for each chip.
I believe having a look in the pyOCD source for the nRF51 might help :)

noppej, 2021-06-21:
@huntc ... I was secretly hoping that I'd be able to reproduce this on my STM32H7 chip, but your very simple main.rs debugs(and halts on breakpoint) just fine on my chip, so I will need you to test something for me please.

My theory is that one of two things are happening ... 
1) Your board is not halting on the breakpoint. If that is the case, then I think we can easily test if the upcoming 'sequences' work mentioned above will alleviate the problem.
2) Also a possibility, is that we have a case that your board is halting on the breakpoint, and the debugger is not detecting this correctly. The debugger has to poll the board periodically to detect if it has chanced status, so that it can notify the VSCode client. 

To eliminate no 2 above, can you please try to hit the halt button in VSCode when you expect the breakpoint to hit but nothing is happening. If your board has indeed halted, then the VSCode interface should update to show that it is halted at the breakpoint. If NOT, then I'd appreciate if you can give me some details on the VSCode stack trace at that point.

huntc, 2021-06-22:
> To eliminate no 2 above, can you please try to hit the halt button in VSCode when you expect the breakpoint to hit but nothing is happening. If your board has indeed halted, then the VSCode interface should update to show that it is halted at the breakpoint.

Here's the breakpoint that I set:

[image]

When I execute via the debugger and then subsequently pause, the stacktrace is as expected i.e. the device is within `wfe`:

[image]

These micro:bit devices are really nice. I'm happy to pay and have one shipped to you for working on this; it is the least I can do! I really appreciate you looking into this. Also, if it helps, I'd be happy to jump onto a video and help you diagnose what's going on. I can be found on the Matrix as @huntc.

noppej, 2021-06-22:
Hi @huntc ... from your call stack, the breakpoint is definitely not being hit. It might be that it doesn't get set appropriately.  I will ping you on matrix so we can do a remote working session. I suspect debugging it that way will be faster than sending around hardware.

noppej, 2021-06-23:
Update: After a remote session with @huntc, I confirmed the following:
- The breakpoint is being set correctly.
- The chip halts on the breakpoint.
- The debugger does not correctly detect the change in status from 'running' to 'halted'. 

Workaround: In VSCode, click on the 'Pause' (F6) button. The debugger will show that the code is halted on the breakpoint.

Note: Further investigation is needed, but it appears the synchronization of run/halt status between the debugger and the chip is related to a sequence of events where the breakpoint is being set while the chip is in a `wfe()` instruction.

noppej, 2021-06-28:
**Update**: After another remote session with @huntc I can now reproduce this on my own hardware, so that will help a lot with resolution. The 'trick' was that the problem only happens when VSCode is using a 'launch' configuration, but does not happen when using an 'attach' configuration against a running debug instance of the `probe-rs-debugger`.  That surprised me, but I will build in additional run-time logging to try to narrow down the root cause.

noppej, 2021-06-30:
**Update**: I have identified the root cause. The debugger fails to correctly set the hardware breakpoint while the target is in sleeping. The cortex-m architecture requires the debugger to update the DHCSR register's C_HALT value to '1' before it attempts certain core operations during sleep mode. I will make the appropriate updates and after my testing, will send you a branch to test against, but it likely won't be until Jul 13th.

noppej, 2021-07-06:
@huntc Are you able to test something for me?   If you install from this branch with 

`cargo install --force --git https://github.com/probe-rs/probe-rs --branch debug_during_wfe_wfi probe-rs-debugger`

then run it in server mode with `probe-rs-debugger debug --dap --port=50001` and `attach` to it from VSCode.

With my test case I get consistent results in this mode, but when I switch to VSCode `launch` mode, it still fails to signal the VSCode client when the target hits breakpoints. I'm working on the latter, but it would help to know how the `attach` workflow behaves on your hardware.

huntc, 2021-07-06:
> @huntc Are you able to test something for me? If you install from this branch with
>
> `cargo install --force --git https://github.com/probe-rs/probe-rs --branch debug_during_wfe_wfi [...]

Success! Nice work.

[image]

noppej, 2021-07-29:
Re-opening, as the current fix for #724  only solves the problem when using VSCode's `attach` mode. The fix for `launch` mode will be done in a separate PR for the probe-rs/vscode repo.

noppej, 2021-10-13:
@huntc I've checked in additional fixes which tests successfully on my side (using your code and similar Micro:Bit board. 

Please can you test with ...
- `cargo install --force --git https://github.com/probe-rs/probe-rs --branch debugger_fixes probe-rs-debugger`

This should allow you to test with the VSCode 'launch' configuration, and if you feel adventurous, you can add `rtt_enabled:true` also :)

huntc, 2021-10-14:
> @huntc I've checked in additional fixes which tests successfully on my side (using your code and similar Micro:Bit board.
> 
> Please can you test by installing `probe-rs-debugger` from this branch ...
> 
> * `cargo install --force --git https://github.com/probe-rs/probe-rs --branch debugger_fixes probe-rs-debugger`
> 
Works fabulously! Thanks!

> This should allow you to test with the VSCode 'launch' configuration, and if you feel adventurous, you can add `rtt_enabled:true` also :)

Thanks - where will the RTT output go? Not seeing it...

noppej, 2021-10-14:
@huntc Thanks. I struggled with where to route the output. I would have liked to route it to the debug console, but then if you have more than one RTT channel, it becomes a bit difficult. VSCode has an HTML view that opens a new window in the code section. If we opened windows there, rather than new Terminal windows, would that make it easier?

commit ff8fd20b (2021-07-29): Merge #724

724: If SLEEP'ing target, assert C_HALT for certain debug commands r=Yatekii a=noppej

This is a (as yet untested by the user) fix to resolve #703 

`probe-rs-debugger` will now assert the DHCSR:C_HALT register flag before certain debug actions, such as `set_breakpoints`.
PR #724 (2021-07-29): If SLEEP'ing target, assert C_HALT for certain debug commands
This is a (as yet untested by the user) fix to resolve #703 

`probe-rs-debugger` will now assert the DHCSR:C_HALT register flag before certain debug actions, such as `set_breakpoints`.