probe-rs/probe-rs #704: Operation timed out with STM32L432KCUx

ctron, 2021-06-20:
**Describe the bug**

I am trying to get "blinky" of https://github.com/embassy-rs/embassy/tree/master/examples/stm32l4 running with an STM32L432KCUx developer board.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/embassy-rs/embassy`
2. `cd embassy/examples/stm32l4`
3. `cargo run --release --bin blinky`

**Expected behavior**

Should work.

**Stacktrace**

~~~
    Finished release [optimized] target(s) in 0.06s
     Running `probe-run --chip STM32L432KCUx target/thumbv7em-none-eabi/release/blinky`
(HOST) WARN  insufficient DWARF info; compile your program with `debug = 2` to enable location info
(HOST) INFO  flashing program (8.50 KiB)
(HOST) INFO  success!
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out

~~~

**Desktop (please complete the following information):**
- Linux, Fedora 34

**Additional context**
Add any other context about the problem here.

Dirbaio, 2021-06-20:
`probe-run` is stuck on a quite old `probe-rs` version. Can you try again with `cargo install --git https://github.com/Dirbaio/probe-run`? 

If it still happens, can you paste the full trace logs? `cargo run --release --bin blinky -- -vv`

ctron, 2021-06-21:
Awesome! Yes, that fixed the issue!

I guess it may be beneficial to release a newer version of probe-run then.

Yatekii, 2021-06-21:
That's not their mistake tho, we didn't release in a long time, which makes it impossible for them to update :)

A release on our side is iminent tho.

closed 2021-06-23 by Yatekii