probe-rs/probe-rs #710: Allow `smoke-tester` to be run without a custom configuration file

noppej, 2021-06-23:
This may just be a documentation shortcoming, but it appears as if the [smoke-tester](https://github.com/probe-rs/probe-rs/pull/690) always requires a configuration file to specify the hardware against which to test.

As a contributor to this repository, I would like to be able to run the smoke tests against my currently connected probe, without having to pollute the repo with custom config files.   That way I can be sure I don't break basic stuff before I submit code for a new PR.

Is it possible to pass configuration such as `chip`, `probe_selector`, and `flash_test_binary` on the command line?

PS. What does 'dut' in 'dut_configuration' stand for?

Yatekii, 2021-06-23:
The idea is that you put all the config into a directory that is not tracked by git, so you do not have to put all the config flags every time you run the smoke tester :)
The idea is that you can run these on many boards, not just one. So it's not very sensible to add flags for this.

Maybe we could add flags nonetheless, if you just wanna test for one board :)

Tiwalun, 2021-06-23:
*dut* stands for *Device under test*.

There are already flags for chip and probe-selection, for the flash binary there is no flag yet.

noppej, 2021-06-23:
> The idea is that you put all the config into a directory that is not tracked by git

That helps ... it wasn't clear to me from the docs, but I can definitely work with that. Thanks.

closed 2021-06-23 by noppej