probe-rs/probe-rs #717: Accessing memory on BL602 RISCV Chip via FTDI/JTAG is very slow

bjoernQ, 2021-06-28:
**Describe the bug**
I'm trying to access memory on a BL602 RISCV chip via Sipeed JTAG adapter (https://www.seeedstudio.com/Sipeed-USB-JTAG-TTL-RISC-V-Debugger-p-2910.html) 

It generally works fine but e.g. dumping memory via CLI is very slow compared to OpenOCD (same for writing memory programmatically)

**To Reproduce**
Steps to reproduce the behavior:

1. in CLI folder `cargo run --features ftdi -- dump 22020000 1000 --chip "Riscv"`

This takes _6.599846518s_ while it's very fast with OpenOCD (OpenOCD config for this chip: https://github.com/sipeed/bl602-rust-guide/blob/main/openocd.cfg )

I tried on two different host machines (Windows 11 and MacOS) and I see the exact same behavior on both. So I guess it's not a hardware / driver thing.

**Expected behavior**
I'd expected this to be speedier - even if it wouldn't be as fast as OpenOCD this is too slow to be useful.

**Stacktrace**
n/a

**Desktop (please complete the following information):**
 - macOS BigSur
 - Windows 10

**Additional context**
I tried to set the FTDI baudrate in code and also tried to set the JTAG clock divider in code. I can get it slower this way but I'm not able to speed things up - so I'm clearly looking into the wrong direction here.

I'd really like to help with this but I don't even have an idea where to look at.

Tiwalun, 2021-06-28:
The RISCV / JTAG code is not really optimized yet, so we should certainly be able to improve our speed here.

From experience with the ARM chips, there are two main areas where we can probably improbe:

- Currently, the code for the FTDI uses one USB transfer to read / write a single JTAG register. USB transfers have quite a big latency, so one way to improve performance is to perfom multiple reads / writes with a single USB transfer.
- The code which handles more high level access, so translating a memory access in the correct JTAG reads and writes, is probably not optimized for access of larger memory areas, but only for access of a single value.

If you're interested in working on this, I can give you some more pointers, depending on which part would interest you more.

bjoernQ, 2021-06-29:
Thanks for the detailed explanation.

My guess is that reducing the number of USB transfers might be the best bet to speed things up.

A brief look into the RISCV TAP spec makes me wonder a bit how such a bulk transfer can be done. An additional look into the OpenOCD code increased my confusion. 

However, I will definitely look into this - not really confident if I can get this done but I'll try

Tiwalun, 2021-06-29:
If you look at the `dmi` register description, in appendix A.2 of the RISCV debug spec, you can see that batching is mentioned there.

The basic idea is that you write this register multiple times, and then you can check afterwards if an error occured. The `op` field of the `dmi` register is sticky, so as soon as an error occurs it will always return an error until you clear it. 

If we want to write multiple registers in the debug module, we can now create a single USB transfer to write all these registers, and will get back the result of all these writes in the same USB transfers. The most tricky bit is probably the error handling, because we have to figure out where the error occured in this case, and then repeat the failed writes.

If you want we can also discuss this in Matrix, in the [probe-rs channel] or privately.

[probe-rs channel]:     https://matrix.to/#/#probe-rs:matrix.org

bjoernQ, 2021-06-30:
Thanks again! Since it seems I have to dive a bit deeper into the topic I started by extracting the relevant code into an own test application so I can make some experiments and get an idea how to speed things up.

Once I'm successful with that I will look into the actual code to see how I can integrate it into the probe-rs code base.

Probably at that point I will come back with additional questions here or on matrix.

Hope I can come up with something useful here

bjoernQ, 2021-07-10:
Here is an update: I'm still on it and it's still more work to do until I can create a PR but batching seems the way to go!

Current _master_: Read 1000 words in 6.6491235s
My fork with batching: Read 1000 words in 252.6971ms

closed 2021-07-20 by bjoernQ