probe-rs/probe-rs #756: Bump bitvec to 0.22

tmplt, 2021-07-29:
The problem I'm trying to solve is that when building https://github.com/rtic-scope/cargo-rtic-scope/tree/bitvec-build-fail, bitvec 0.19 fails to build for some reason I do not understand. This seems to be because `cargo-rtic-scope` depends on a `itm-decode 0.4` alpha which depends on `bitvec 0.22`.

At the moment I need to maintain a fork of `probe-rs` that bumps its `bitvec` dep to 0.22, but this now breaks ftdi, which I do not require (yet).

_Originally posted by @tmplt in https://github.com/probe-rs/probe-rs/issues/755#issuecomment-889362780_

tmplt, 2021-07-29:
The issue can be replicated in `probe-rs` by bumping `itm-decode` to `0.4.0-alpha`. [A fix is to pin `funty = "=1.1.0`](https://github.com/bitvecto-rs/bitvec/issues/105) but this then breaks dependency resolution: `probe-rs` requires `funty v1.1.0` and `itm-decode` requires `~1.2` (currently `1.2.0`).

`itm-decode` will reach a `0.4` release eventually, so we'll have to deal with this eventually also.

Tiwalun, 2021-07-29:
ftdi should not be an issue, we just have to update probe-rs to use the newer bitvec api. The bigger problem is the `gdb-server`:

```
bitvec v0.19.4
└── nom v6.1.1
    └── gdb-server v0.11.0 (/Users/tiwalun/code/probe-rs/gdb-server)
```

Tiwalun, 2021-07-29:
`cargo tree` is very helpful if you need to find out why a package is used: `cargo tree -p bitvec:0.19.4 -i` will produce the output above.

Tiwalun, 2021-07-29:
Using version `7.0.0-alpha1` of nom should fix this, see https://github.com/Geal/nom/issues/1330#issuecomment-867706743.

closed 2021-07-29 by bors[bot]

Geal, 2021-08-21:
you can now upgrade to nom 7, which was just released, and moves bitve input types to the nom-bitvec crate

Yatekii, 2021-08-21:
Thanks for the headsup and all the hard work!

commit e1528ee7 (2021-07-29): Merge #757

757: cargo: bump bitvec r=Yatekii a=tmplt

Closes #756.
PR #757 (2021-07-29): cargo: bump bitvec
Closes #756.