probe-rs/probe-rs #781: Fail to connect to RISCV core over jlink

MabezDev, 2021-08-24:
**Describe the bug**
I am working on a [probe-rs flash loader](https://github.com/MabezDev/esp-flash-loader) for the [esp32c3](https://www.espressif.com/en/products/socs/esp32-c3), however I haven't got as far as testing because probe-rs is failing to connect to the chip.

**To Reproduce**
Steps to reproduce the behavior:
1. Wire up JLINK to esp32c3
2. Run `cargo flash --connect-under-reset --chip-description-path output/esp32c3.yaml --log debug --protocol jtag --speed 1000 --chip esp32c3 --reset-halt`
3. Failure.

**Expected behavior**
Correctly connect to the on chip DTM.

**Stacktrace**
```rust
DEBUG cargo_flash > Changed working directory to .
       DEBUG probe_rs_cli_util > Running '/home/mabez/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo' in directory /home/mabez/development/rust/embedded/experiments/esp32c3
   Compiling esp32c3 v0.1.0 (/home/mabez/development/rust/embedded/experiments/esp32c3)
    Finished dev [unoptimized + debuginfo] target(s) in 0.19s
    Flashing /home/mabez/development/rust/embedded/experiments/esp32c3/target/riscv32imc-unknown-none-elf/debug/esp32c3
       DEBUG probe_rs::config::registry > Searching registry for chip with name esp32c3
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x40380000, virtual address: 0x40380000, flags: 0x5
        INFO probe_rs::flashing::download > Matching section: ".text"
        INFO probe_rs::flashing::download > Found loadable segment, physical address: 0x40381d00, virtual address: 0x40381d00, flags: 0x4
        INFO probe_rs::flashing::download > Matching section: ".rodata"
        INFO probe_rs::flashing::loader   > Found 2 loadable sections:
        INFO probe_rs::flashing::loader   >     .text at 40380000 (7412 bytes)
        INFO probe_rs::flashing::loader   >     .rodata at 40381D00 (1484 bytes)
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.24.11586
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: true
       DEBUG jaylink                          > libusb can detach kernel driver: true
       DEBUG jaylink                          > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                          > scanning 1 interfaces
       DEBUG jaylink                          > J-Link interface is #0
       DEBUG jaylink                          > legacy caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx
       DEBUG jaylink::capabilities            > unknown ext. capability bits: 0x354FB17C1DB9FF7BBF truncated to 0x1B9FF7BBF (Reserved0 | GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite)
       DEBUG jaylink                          > extended caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
[... 32 lines cut ...]
       DEBUG probe_rs::probe::jlink                 > Write IR: [16], len=5
       DEBUG probe_rs::probe::jlink                 > Read 32 bits from DR
       DEBUG probe_rs::probe::jlink                 > Read from DR: [0, 0, 0, 0]
       DEBUG probe_rs::architecture::riscv::dtm     > Dtmcs: Dtmcs { .0: 0, idle: 0, dmistat: 0, abits: 0, version: 0 }
        INFO cargo_flash                            > Metadata {
    chip: Some(
        "Unspecified(\"esp32c3\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "1000",
    ),
    release: "0.11.0",
    commit: "crates.io",
}
       Error 0: An error specific to the selected architecture occured
             1: The version '0' of the debug transport module (DTM) is currently not supported.
```

**Desktop (please complete the following information):**
 - macOS Mojave
 - Windows 10
 - Linux (Ubuntu, Gentoo, Debian, Fedora, etc.)

Yatekii, 2021-08-24:
Yes, I think this is expected.

RISC-V has several ways of accessing memory etc. We have only implemented the ones for specific chips we already have :)
Someone has to implement this.

MabezDev, 2021-08-24:
Thanks @Yatekii, I'd be interested in taking a look and implementing the missing features. However, I am new to the probe-rs project, do you have any pointers on how you would go about solving this?

Tiwalun, 2021-08-25:
It already fails earlier, to access the debug module memory access is not required yet. This only comes into play later.

The JTAG IDCODE should not be zeroes, so my suspicion would be that something in the JTAG access is not working correctly.
It could be that there are multiple JTAG TAPs, and it fails to find the correct one.

If you have an FTDI based debugger, it would be great to check it works with that one, because the JTAG code there has support for multiple TAPs.

Yatekii, 2021-08-25:
Ok, pardon my ignorance then :/ I should have consulted @Tiwalun first :)

MabezDev, 2021-08-25:
> If you have an FTDI based debugger, it would be great to check it works with that one, because the JTAG code there has support for multiple TAPs.

Unfortunately I don't have an FTDI adapter around right now - is the feature for searching for taps missing in probe-rs or jaylink?

If I have the tap id (`0x00005c25`) can I forcefully choose that one some how?

Tiwalun, 2021-08-25:
It's missing in probe-rs, there is only a very basic interface provided by jaylink / J-Link.

For the FTDI, there is a scan function which tries to find the correct TAP: https://github.com/probe-rs/probe-rs/blob/e8b6af46daeae2b84a70481a43f879f2a99c360d/probe-rs/src/probe/ftdi/mod.rs#L246

This is missing for the J-Link, so there is no way of selecting a TAP right now.

MabezDev, 2021-08-25:
Thanks for the tip @Tiwalun, I started implementing the scan feature (it copy and pasted pretty easily across) and found that there is only one tap device on the esp32c3. However, in my step debugging, I found it would correctly read the idcode when stepping through the code. 

I added a 100ms delay between the `jtag_io` to reset the jtag chain & `let idcode_bytes = self.read_dr(32)?;` and it seems to work reliably. I don't know enough about the jtag spec itself to know where there is a allowed delay period, or whether this is just a quirk of this silicon - how do you suggest we proceed?

commit a32fdef0 (2021-09-08): Merge #784

784: Read idcode bytes until non-zero bytes appear, or 1 second passes. r=Yatekii a=MabezDev

Closes #781