probe-rs/probe-rs #790: Operation Status = Some(RequestInProgress)

MabezDev, 2021-08-28:
**Describe the bug**
Trying to using cargo flash with the current probe-rs master not longer works with a jlink and esp32c3, v0.11.0 works fine.

**Stacktrace**
```rust
Error 0: Something during the interaction with the core went wrong
      1: An error specific to a probe type occured
      2: Operation Status = Some(RequestInProgress)
```

**Additional context**
I think this was introduced in https://github.com/probe-rs/probe-rs/pull/735, specifically this error comes from  https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/src/architecture/riscv/dtm.rs#L115

cc : @bjoernQ perhaps you might have a better understanding of what might be going on?

bjoernQ, 2021-08-28:
Which adapter are you using? Asking because a user reported that it worked fine with the Sipeed JTAG adapter (which I also use, the older one - not the BL702 based) but not with the Pine64 (I ordered one but it's still on its way - hope to get it soon)  (see https://github.com/bjoernQ/blash/issues/1#issuecomment-900601199 )

The reported problem looks different but it is maybe adapter dependent.

My first guess is that it might be caused by the value at https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/src/probe/ftdi/mod.rs#L612

MabezDev, 2021-08-28:
I am using a J-LINK debug probe. A quick skim of that issue you linked this problem doesn't seem related.

bjoernQ, 2021-08-28:
Ah sorry Jlink - not Ftdi ... Then I'm not too sure it's caused by #735

MabezDev, 2021-08-28:
Are you sure? The changed files in #735 modify `probe-rs/src/probe/jlink/mod.rs` and others?

bjoernQ, 2021-08-28:
Yes it's a big PR but I really hoped to not break Jlink since I don't have a Jlink adapter to verify it. I hope I mentioned that in the PR

So I can not be sure it's not caused by #735 - just hoping it

MabezDev, 2021-08-28:
Hi @bjoernQ, I just spent a bit of time `git bisect`ing and have narrowed it down to exactly this commit:  37714e898412652082b88f792f7ef6e24451a20f.

Looks like that was included in your PR - any idea why the Jlink changes would break? Any ideas you'd like me to test?

bjoernQ, 2021-08-29:
I guess I see the problem:
https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/src/architecture/riscv/dtm.rs#L112-L117

returns an `Err`, while the non-batch implementation:

https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/src/architecture/riscv/dtm.rs#L153-L155

returns an `Ok` wrapping an `Err` 

So, I guess it's not only JLink but also FTDI - it just never happens with my BL602s

Got two ESP32C3 yesterday - will try to use them

bjoernQ, 2021-08-29:
Still trying to get the ESP32C3 JTAG working - burnt DIS_USB_JTAG e-fuse but JTAG doesn't seem to work
Also just ordered a JLINK adapter

Anything else needed to get ESP32C3 to enable JTAG?

MabezDev, 2021-08-29:
Hmm, that _should_ work, is the wiring is correct?

FYI you can set the `JTAG_SEL_ENABLE` efuse, then use GPIO10 as a strapping pin (Pulling low for external jtag, high for usb jtag), if you don't want to permanently disable the usb jtag.

We're working on improving the docs for this.

bjoernQ, 2021-08-29:
I have the AI Thinker ESP-C3-32S-Kit and connected the JTAG adapter like this

IO4 -> TMS
IO5 -> TDI
IO6 -> TCK
IO7 -> TDO
GND -> GND

with DIS_USB_JTAG burnt there is nothing else needed, right?

bjoernQ, 2021-08-29:
ok JTAG seems to be activated on the chip - just probe-rs master doesn't work for FTDI for me currently

MabezDev, 2021-08-30:
I hope you managed to sort the FTDI issues!

In the mean time I've been thinking about the core issue - `DmiOperationStatus::RequestInProgress` is a "soft error" that can be recovered from. The problem is that by the time the probe is `execute()`ing the batch of commands, its not possible to recover from this error and these are processed at probe level, not the dtm level.

Should the batching be done at the dtm level, not the probe level?

Perhaps batching should be an extension trait for a DebugProbe, along with a method for the communication interface to test whether the given probe supports batching, and execute in batch mode or single mode.

bjoernQ, 2021-08-30:
Seems that the FTDI problem is just a problem of the CLI and Debugger which I use to quickly test things. Need to look into that and create an issue for that (if it's really just a problem there)

Yes `DmiOperationStatus::RequestInProgress` triggered a re-try before so it's really a problem introduced with my PR unfortunately.

An easy / quick solution would be to define an error like `RetrySuggested` and the batch executor can decide to retry (until a timeout occurs). Not sure if there is something I didn't took into consideration. Certainly, just a solution for the emulated batching - not for FTDI but I need to investigate if that can really happen with FTDI and if yes there are probably other ways to solve it there.

At least that is what I'd like to try when my ordered JLink arrives. But no idea how long the delivery will take - but it's delivered from a shop located in my home-country and they said it's already in the mail.

In my first commits in the PR the probes had a function to query if it supports batching or not and the bulk-memory operations checked that and had two branches of execution. That was definitely safer but also uglier

bjoernQ, 2021-08-30:
You are certainly right and retries are not that easy

The RV Debug Spec contains some hints:

_The still-in-progress status is sticky to accommodate debuggers that batch together a number of
scans, which must all be executed or stop as soon as there’s a problem.
For instance a series of scans may write a Debug Program and execute it. If one of the writes
fails but the execution continues, then the Debug Program may hang or have other unexpected
side effects._

_3: An operation was attempted while a DMI re-
quest is still in progress. The data scanned into
dmi in this access will be ignored. This status is
sticky and can be cleared by writing dmireset in
dtmcs. If a debugger sees this status, it needs to
give the target more TCK edges between Update-
DR and Capture-DR. The simplest way to do that
is to add extra transitions in Run-Test/Idle._

bjoernQ, 2021-08-31:
So the FTDI problem I mentioned got already fixed but now trying to connect to the ESP32C3 doesn't work (with FTDI):

I see this in the logs
```
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] attaching...
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] tap found: 00005c25
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] tap 0 irlen: 2
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] tap found: 00005c25
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] tap 0 irlen: 2
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] Target chain params: ChainParams { irpre: 0, irpost: 0, drpre: 0, drpost: 0, irlen: 2 }
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] read_register(0x10, 32)
[2021-08-31T16:48:32Z DEBUG probe_rs::probe::ftdi] target_transfer error: Custom { kind: InvalidInput, error: "invalid register address" }
Error: Error attaching to the probe: Probe(ProbeSpecific(Custom { kind: InvalidInput, error: "invalid register address" })).
```

I'm a bit confused why it reports two TAPs with same IDcode

Tiwalun, 2021-08-31:
The FTDI problem happens because the detected `irlen` is wrong, it should detect a length of 5. 

This is from another tool:

```
jtag> detect
IR length: 5
Chain length: 1
Device Id: 00000000000000000101110000100101 (0x00005C25)
Unknown manufacturer! (11000010010) (/usr/share/urjtag/MANUFACTURERS)
```

But I think this is a separate issue, not related to the operation status.

bjoernQ, 2021-08-31:
Thanks! When I hardcode the `irlen` it gets a bit further and times out in `execute_abstract_command`  after executing some commands without timeout - no idea why that apparently works better using Jlink

bjoernQ, 2021-08-31:
That happens in `get_available_breakpoint_units` ... if I just return `Ok(0)` for testing I can actually (apparently) dump memory - quite fast and without the `RequestInProgress` problem

bjoernQ, 2021-09-01:
So, my Jlink arrived and after soldering a suitable cable (guess it won't last long) I can reproduce the problem with the `debugger` tool (`cargo run -- dump 3ff00000 10 --chip riscv --protocol jtag`) while it works with my oddly patched FTDI version.

Quite interesting ....
I can make it work by making https://github.com/probe-rs/probe-rs/blob/f8c1df1cfbe3d4b3fb0bac767cb7c54daff968e0/probe-rs/src/probe/jlink/mod.rs#L672

unconditionally (i.e. remove the surrounding `if`). Would be interesting if that also works for you (while that is certainly not a real solution - but would be interesting)

BTW what do you guys use to connect the Jlink to the DevKit? I have seen JTAG breakout boards but they are either ridicules expensive or will take weeks to ship from China.

bjoernQ, 2021-09-01:
BTW that optimization I removed to make it work for me seems actually counterproductive.

_0.11.0_ without changes:  Read 1000 words in 2.4009798s
_master_ with my change (the removed `if`): Read 1000 words in 1.2770868s

So maybe not that much of a problem to remove it? (All number by using ESP32C3)

MabezDev, 2021-09-01:
> Quite interesting ....
I can make it work by making

Interestingly I can reproduce that here. By unconditionally calling `write_ir` it seems to work.

However I do not see the performance gains, in fact I see a regression of program speed (writing to flash) in the order of 10 seconds.

> BTW what do you guys use to connect the Jlink to the DevKit? I have seen JTAG breakout boards but they are either ridicules expensive or will take weeks to ship from China.

I just use some dupont cables hooked up from the jlink 20pin directly to the esp32c3 pins.

bjoernQ, 2021-09-02:
> However I do not see the performance gains, in fact I see a regression of program speed (writing to flash) in the order of 10 seconds.

Just to double check: that is comparing _0.11.0_ vs _master_ including the change, right? Interesting the results differ so much. My figures suggested me that the penalty of always doing `write_ir` outweighs the penalty of repeating the command - but even if it's the case it might be MCU dependent

MabezDev, 2021-09-02:
Yes that was comparing v0.11 and master.

Do you get the same results when you write 1000 words to memory, instead of reading?

bjoernQ, 2021-09-02:
I can confirm the performance regression by a factor of 1.5 / 1.6 unfortunately

bjoernQ, 2021-09-03:
If you like you can try my attempt to fix this: https://github.com/bjoernQ/probe-rs/tree/fix_esp32c3_memory_access

It looks quite encouraging to me:

`cargo run --example ram_download -- --address 0x40380000 --size 4000 --chip riscv --protocol jtag`

_0.11.0_
```
Wrote 16000 bytes in 1.7331137s ( 9231.94 bytes/s)
Read  16000 bytes in 9.9603429s ( 1606.37 bytes/s)
Verification succesful.
```

_master plus my fixes_
```
Wrote 16000 bytes in 1.3800337s (11593.92 bytes/s)
Read  16000 bytes in 3.828124s ( 4179.59 bytes/s)
Verification succesful.
```

But I only tested Jlink + ESP32C3 ... need to also test with BL602 and also check it doesn't break FTDI

MabezDev, 2021-09-03:
Thanks @bjoernQ!

I tried out your branch whilst it does seem to work, I don't see your gains; at least for flashing.

v0.11.0:

```rust
Erasing sectors ✔ [00:00:00] [#####################################################################################################################################################################################] 16.00KiB/16.00KiB @  9.98KiB/s (eta 0s )
Programming pages   ✔ [00:00:14] [#####################################################################################################################################################################################] 12.25KiB/12.25KiB @     803B/s (eta 0s )
Finished in 15.506s
```

master + your changes:

```
Erasing sectors ✔ [00:00:00] [#####################################################################################################################################################################################] 16.00KiB/16.00KiB @ 11.35KiB/s (eta 0s )
Programming pages   ✔ [00:00:23] [#####################################################################################################################################################################################] 12.25KiB/12.25KiB @     506B/s (eta 0s )
Finished in 24.68s
```

Looking at the changes though, they do seem like reasonable optmisations so I'm not sure why I'm seeing such poor performance.

Yatekii, 2021-09-03:
Flashing speed seems to be defined by something else. Is the algorithm that bad (no offense) or is the flash really that slow (can't imagine)?
I guess it has to be one of those two because general data transfers seem to be faster (Even tho those seem to be terribly slow as well!).

MabezDev, 2021-09-04:
> Is the algorithm that bad (no offense)

None taken! The algorithm is definitely slow, I'm just getting stuff working atm; however I'm suprised to see reduced performance of the algo with these changes. Maybe I messed something up applying the patch, I'll try again tomorrow.

Regardless of performance, this only seems to patch the symptoms, not the root cause. Currently in batched mode, there is now way to recover from a Dtm busy error, I opened #802 to explore trying to fix that. However if there are performance gains to be had, @bjoernQ should definitely PR them!

bjoernQ, 2021-09-04:
@MabezDev  any insights what causes the totally different experience on your side? Maybe you can run the ram_download example to see if you have similar bad figures there?

I'd really like to PR this since after that I can try to improve the FTDI implementation based on this. But if it makes things worse for someone certainly not.

I also think it's independent from your PR since yours is more an architectural change - both make sense on their own

MabezDev, 2021-09-04:
I haven't figured out whats going on wrt flashing, but I do see your gains running the ram example!

> both make sense on their own

I agree, I think your should PR your changes. I don't think its worth blocking on just one experimental target.

bjoernQ, 2021-09-05:
Thanks for testing. I did some more tests on my side and unfortunately it degrades performance for BL602/Jlink slightly. So, I won't create a PR for that.

Your PR seems to be the way to go

commit 91004107 (2021-09-10): Merge #802

802: DTM batching r=Dirbaio a=MabezDev

The main goal here is to avoid having to emulate batching for every JTAG probe we add; instead queue commands within the DTM and selectively batch or write sequentially based on probe capabilities.

The extension trait technique (coined as "IDET"'s by `@daniel5151,` see the [gdbstub docs](https://docs.rs/gdbstub/0.5.0/gdbstub/target/ext/index.html#how-protocol-extensions-work---inlineable-dyn-extension-traits-idets)) seems to work quite well here, probes can opt into the batching functionality, and there is a single branching path in `execute [... cut]
PR #802 (2021-09-10): DTM batching
The main goal here is to avoid having to emulate batching for every JTAG probe we add; instead queue commands within the DTM and selectively batch or write sequentially based on probe capabilities.

The extension trait technique (coined as "IDET"'s by @daniel5151, see the [gdbstub docs](https://docs.rs/gdbstub/0.5.0/gdbstub/target/ext/index.html#how-protocol-extensions-work---inlineable-dyn-extension-traits-idets)) seems to work quite well here, probes can opt into the batching functionality, and there is a single branching path in `execute` to determine what to do with the probe. [... cut]
closed 2021-09-10 by bors[bot]