probe-rs/probe-rs #792: Add ATSAMG targets

mcbridejc, 2021-08-30:
I'd like to support the SAMG55J19 for my project, so would love to see the microchip/atmel SAMG pack pulled into probe-rs.

I attempted to create a pull request using target-gen, but I ran into an issue about a missing 'cores' field in the yaml description. If I've understood correctly from reading commit logs and issues, it appears there's some work ongoing to support multiple cores, and this involves a change to the yaml schema which hasn't yet found its way into target-gen, and I've stumbled in at a bad time to add a part! 

In the mean time, I'm working with a [fork of 0.11.0](https://github.com/mcbridejc/probe-rs) to which I added the SAMG parts.

Dirbaio, 2021-08-30:
> it appears there's some work ongoing to support multiple cores, and this involves a change to the yaml schema which hasn't yet found its way into target-gen,

Yes, that's unfortunately correct!  :(

It should be easy to manually fix up the YAML to work, if the chip is singlecore.

1. add `cores` with a single core named "main", like this:

https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/targets/SAMD51.yaml#L6-L12

2. Add `cores: [main]` to all memory map regions

https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/targets/SAMD51.yaml#L19

2. Add `cores: [main]` to all flash algos

https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/targets/SAMD51.yaml#L262

Yatekii, 2022-01-30:
This has been improved as much as possible :)

Closing.

closed 2022-01-30 by Yatekii