probe-rs/probe-rs #793: Regression: "connect-under-reset" no longer works for STLINK

noppej, 2021-08-30:
**Describe the bug**
When using "connect-under-reset" on an STLINK device, [this code](https://github.com/probe-rs/probe-rs/blob/bbe3bfc4604a7529f6ff3f7a6cf9b1b02cd6cb4f/probe-rs/src/session.rs#L180) will cause an unrecoverable *"ERROR: Some functionality was not implemented yet: ST-Links do not support raw SWD access."*

This used to work prior to [#483](custom sequences)

Yatekii, 2021-08-30:
Yeah this is my mistake. @Tiwalun warned me and I forgot to test that it still works like a pro ... ofc. Maybe @Tiwalun or @Dirbaio has an idea how to get around this off the top of their head. Anyways, I'll have a look in the coming days :)

noppej, 2021-10-18:
There is a PR #850 waiting to be reviewed. Once that is merged the `halt_after_reset` can be used with RTT again.