probe-rs/probe-rs #797: Publicly expose `enum StlinkError`

Urhengulas, 2021-09-01:
**Is your feature request related to a problem? Please describe.**
In `probe-run` we are not able to cleanly handle a `StlinkError::CommandFailed(JtagNoDeviceConnected)`, because the `enum StlinkError` isn't publicly exposed.

https://github.com/probe-rs/probe-rs/blob/b7e7cfd667a0a85fe012017f3cf865038001970e/probe-rs/src/probe/stlink/mod.rs#L1109-L1131

**Describe alternatives you've considered**
Currently we transform the error into a string and then check if it contains a specific string. This is not optimal.

```rust
if e.to_string().contains("JtagNoDeviceConnected") {
    // ...
}
```

**Describe the solution you'd like**
We'd like to be able to handle this directly and therefore need this type to be exposed.

**Additional context**
This issue came up in https://github.com/knurling-rs/probe-run/pull/253#discussion_r698359959.

Tiwalun, 2021-09-01:
Exposing this shouldn't be a problem.

Maybe it might be better to add a more generic error for this case? This error indicates that no device seems to be connected, which can also occur with other probes. I'm not sure what kind of error is returned for other probes right now.

Do you need to handle the ST-Link differently, or would a more generic error for this kind of problem also be suitable?

Urhengulas, 2021-09-01:
> Exposing this shouldn't be a problem.

Great! 😄 

> Maybe it might be better to add a more generic error for this case? This error indicates that no device seems to be connected, which can also occur with other probes. I'm not sure what kind of error [...]

A more generic error would probably be even better to be able to handle more probes.

Is there a way to get the specific probe from the `probe_rs::Probe`, in case we want to differentiate?

bugadani, 2024-03-30:
The enum was made publicin #2027

closed 2024-03-30 by bugadani