probe-rs/probe-rs #798: Error occured during execution of an abstract command: HaltResume

MabezDev, 2021-09-01:
**Describe the bug**
Occasionally programming will be interrupted by the error in the title.

**To Reproduce**
Steps to reproduce the behavior:
1. Repeatedly flash until the error is hit.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Stacktrace**
```rust
Error 0: Something during the interaction with the core went wrong
      1: A core architecture specific error occured
      2: Error occured during execution of an abstract command: HaltResume
```

**Additional context**
Tested on a jlink & an esp32c3.

I looked at how [openocd handles it](https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c), and it seems they don't even check it... No reference to `CMDERR_HALT_RESUME` is found except for its initial definition. 

I modified the probe-rs to not exit when `HaltResume` is detected, and it seems to just work...

Tiwalun, 2021-09-02:
The implementation in probe-rs is based on the [RISCV Debug Spec], and from my understanding the `HaltResume` error can happen when you try to execute an abstract command, but the state of the hart is incorrect.

I think probe-rs should ensure the hart is in the correct state before executing an abstract command, so this error shouldn't happen at all. Do you know which abstract command causes this error?

[RISCV Debug Spec]: https://raw.githubusercontent.com/riscv/riscv-debug-spec/release/riscv-debug-release.pdf

MabezDev, 2021-09-07:
It seems to be erroring on a `AccessRegisterCommand`, I managed to capture a (crappy, sorry! anyhow backtrace wasn't working for me) backtrace:

```rust
execute_abstract_command (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:991)
abstract_cmd_register_read<&probe_rs::core::RegisterDescription> (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:1050)
perform_memory_write_multiple_progbuf<u8> (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:879)
write_multiple<u8> (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:1230)
write_8 (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:1705)
write_8 (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/architecture/riscv/mod.rs:550)
write_8 (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/core/mod.rs:215)
load_page_buffer (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/flasher.rs:810)
{closure#0} (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/flasher.rs:401)
run_program<i32, probe_rs::flashing::flasher::{impl#3}::program_double_buffer::{closure#0}> (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/flasher.rs:204)
program_double_buffer (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/flasher.rs:397)
program (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/flasher.rs:285)
commit (/home/mabez/development/rust/embedded/tools/probe-rs/probe-rs/src/flashing/loader.rs:330)
run_flash_download (/home/mabez/development/rust/embedded/tools/cargo-flash/src/main.rs:668)
main_try (/home/mabez/development/rust/embedded/tools/cargo-flash/src/main.rs:480)
main (/home/mabez/development/rust/embedded/tools/cargo-flash/src/main.rs:285)
call_once<fn(), ()> (@core::ops::function::FnOnce::call_once::h818d69ef6aa4ff63:6)
__rust_begin_short_backtrace<fn(), ()> (@std::sys_common::backtrace::__rust_begin_short_backtrace::h8fa5fb41f40f9822:6)
{closure#0}<()> (@std::rt::lang_start::_$u7b$$u7b$closure$u7d$$u7d$::h785d4945e9dcbc5c:7)
core::ops::function::impls::_$LT$impl$u20$core..ops..function..FnOnce$LT$A$GT$$u20$for$u20$$RF$F$GT$::call_once (@std::rt::lang_start_internal::hca14fb2549072049:174)
std::panicking::try::do_call (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::panicking::try (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::panic::catch_unwind (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::rt::lang_start_internal::_$u7b$$u7b$closure$u7d$$u7d$ (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::panicking::try::do_call (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::panicking::try (@std::rt::lang_start_internal::hca14fb2549072049:172)
std::panic::catch_unwind (@std::rt::lang_start_internal::hca14fb2549072049:172)
lang_start_internal (@std::rt::lang_start_internal::hca14fb2549072049:172)
lang_start<()> (@std::rt::lang_start::h2db905bf6adaed1c:13)
main (@main:10)
__libc_start_main (@__libc_start_main:53)
_start (@_start:14)
```

MabezDev, 2021-09-21:
So I actually thing this is related to #826, I think in some cases it was taking _so_ long to load the program data I would run into this error. I haven't seen this error once since #826, trying multiple page sizes. Closing.

closed 2021-09-21 by MabezDev