probe-rs/probe-rs #801: Speed estimation of CLI progress bars seems wrong

Dirbaio, 2021-09-02:
```
[dirbaio@mars firmware]$ probe-rs-cli download --format hex --chip nrf52840 softdevice/s140_v7.2.0.hex --speed 30000
     Erasing sectors ✔ [00:00:03] [#####################################################################################################] 156.00KiB/156.00KiB @ 45.02KiB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [#####################################################################################################] 156.00KiB/156.00KiB @ 25.14KiB/s (eta 0s )
    Finished in 5.332s
```

The first bar takes 3s, and the speed estimate says `45.02KiB/s`. The second takes 1s, so it's *faster*, but the speed estimate is `25.14KiB/s` which is slower...

bugadani, 2024-01-21:
Is this still an issue?

Yatekii, 2024-01-22:
I think the @ was always just a rolling speed and not the average, so this is actually correct althought counter-intuitive I believe. I wonder if we can make it more intuitive.

bugadani, 2024-03-30:
I do not have an nrf52840 but after a bunch of trials I haven't been able to repro, so I have to assume this has been fixed. Please reopen if you encounter the issue again.

closed 2024-03-30 by bugadani