probe-rs/probe-rs #827: Implement compressed transfers

MabezDev, 2021-09-21:
It seems when flashing, loading data is the most time consuming process (just may experience, may not always be the case). Reducing the number of bytes we have to transfer should improve flashing speeds.

Compressing pages of data, copying to target ram, then decompressing on the target ready for writing to flash should net improved flashing performance.

A quick look on crates.io yielded this crate: https://github.com/alexkazik/lzss which may be of use.

Dirbaio, 2021-09-21:
On many ARM chips the bottleneck is the actual flash write speed (copying from RAM to flash), not the copy from host to target RAM, given a decent probe (ie not jlink) and decent SWD frequency.

How common is for jtag/swd to be the bottleneck? Maybe there is some low-hanging-fruit optimizing the jtag memory transfers?

MabezDev, 2021-09-21:
I should note that I'm not running into this bottleneck right now, that's for sure!

However, in the case of some Espressif chips (and other chips I image), the spi flash fitted is capable off tens of Megabytes write speed, when in quad io mode at full speed. I'm getting no where near that speed (10kb/s write) with the `ram_download` example but I am running a jlink so it could be that.

Side note, whats the jtag support like on the hs-probe - I can't lie I am quite tempted by the lil guy :D

adamgreig, 2021-09-22:
> Side note, whats the jtag support like on the hs-probe - I can't lie I am quite tempted by the lil guy :D

I can answer this at least! It supports the `JTAG_Sequence` CMSIS-DAP command, so it can do arbitrary JTAG sequences using something like openocd, at reasonably high throughputs: my [ecpdap](https://github.com/adamgreig/ecpdap) tool writes to an FPGA over JTAG at 820KB/s, and reads attached SPI flash memory at 750KB/s, for example. The firmware doesn't yet support the JTAG commands used to debug ARM JTDP debug ports, though that's not relevant to the riscv use cases.

> spi flash fitted is capable off tens of Megabytes write speed

Wow, tens of MB/s write? What SPI flash is that?

MabezDev, 2021-09-22:
> Wow, tens of MB/s write? What SPI flash is that?

My bad, on further inspection although the chips I was looking at support quad writes, thats only at low clock rates only.

> I can answer this at least! It supports the JTAG_Sequence CMSIS-DAP command, so it can do arbitrary JTAG sequences using something like openocd, at reasonably high throughputs: my ecpdap tool writes [...]

Thanks for the info adam! Unfortunately is seems like JTAG functionality [isn't actually supported in probe-rs](https://github.com/probe-rs/probe-rs/blob/0538e3e39961cda6d02d520dd2ca0830030ff84f/probe-rs/src/probe/cmsisdap/mod.rs#L479) just yet, I'll keep an eye out :)

Speaking of JTAG support, at some point I'd like there to be support for the usb jtag on the c3(and other upcoming chips); given that this usb interface is Usb Full Speed (1.5MB/s) compression could help here.

Yatekii, 2021-09-22:
1.5MB/s is well enough for any flash algo. The fastest flashes I have seen are in the 100kb/s speed wise. so 15x the overhead is well enough :) But if your flash is really that fast, we can consider it :)

ianrrees, 2022-02-21:
Can the flash algos typically handle flashing and receiving data over JTAG/SWD at the same time?  I've only looked at them for some weird old chips at work, but those don't do anything that fancy.  So, those steps are done in serial, so I think that even if the flashing itself is the slower part of the process there may still be an advantage in compressing over JTAG/SWD.

PR #1947 (2023-12-16): Support different transfer encodings and implement Miniz (+ length prefix)
Closes #827.

The encoding is the property of the flash algorithm, because there is no good standard API of setting it in runtime. Also, the flash algorithms is preferably small, so selection should be on a per-flash-algo level anyway, each flash algorithm supporting only one.

When not specified, we fall back to the raw encoding, so additional encodings are completely opt-in.

This PR also removes `use ProgressEvent::*;` patterns as it is a refactoring footgun, as demonstrated, well, here...
closed 2023-12-16 by MabezDev