probe-rs/probe-rs #828: [RFC] Unify all binaries

Dirbaio, 2021-09-23:
Currently the probe-rs project produces 6 binaries, with overlapping functionality:

- cargo-embed
  - flashing
  - RTT, with optional defmt
  - GDB server
- cargo-flash
  - flashing
- probe-rs-cli
  - flashing
  - erasing
  - debugger CLI
- probe-rs-debugger
  - flashing
  - debugger CLI
  - MS-DAP server
  - RTT, with optional defmt
- rtthost
  - RTT
- gdb-server
  - gdb server

I propose merging them all into a single binary, named `probe-rs`. It would have the union of all the above features, organized in the following subcommands:

- `download`: download (flash) a binary
- `erase`: erase all target flash memory
- `run`: Flash binary, then run it showing RTT/defmt output (like cargo-embed, probe-run)
- `rtt`: Attach to rtt input and/or output channels
- `list`: list all connected debug probes
- `reset`: Reset the target
- `debug`: debug CLI
- `gdb-server`: GDB server
- `dap-server`: MS DAP server

and perhaps others (dump, trace..)

## Advantages

- Faster installation: just one binary to install. Especially faster if building from source with `cargo install`, as it builds everything from scratch for each binary.
- Discoverability: a user may have read about `cargo flash` in some tutorial, and does `cargo-flash --help` and doesn't discover there's additional awesome functionality but in other binaries.
- Visibility for the project: Having `probe-rs` being the binary name is nice :) 
- Less code duplication, more consistent UI

## Cargo integration

I propose to *not* make the new binary a [cargo custom subcommand](https://doc.rust-lang.org/cargo/reference/external-tools.html#custom-subcommands), ie it will not be runnable as `cargo probe-rs` or similar, unlike `cargo embed` and `cargo flash`.

Instead, the integration can be achieved by registering `probe-rs` as a [Cargo runner](https://doc.rust-lang.org/cargo/reference/config.html#targettriplerunner) in `.cargo/config.toml`:

```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-rs run --chip nRF52840"
```

With that, `cargo run` will build the binary, then flash+run it with `probe-rs`.

It is still possible to pass custom arguments to probe-rs: `cargo run -- --connect-under-reset`.

Reasons: 

- We want `probe-rs` to be useful for any embedded developer, not just Rust developers. A binary named `cargo-foo` is not welcoming for non-Rust developers. We could distribute `probe-rs` and `cargo-probe-rs` but that would be two binaries.
- Multiple commands require multiple binaries: `cargo-embed` vs `cargo-flash`.
- The Rust integration is not that seamless; cargo subcommands must closely track and add support for all arguments Cargo supports, or some might not work. For example, `cargo embed -Zbuild-std` doesn't work today.

With Cargo runners we can keep the same great UX for Rust developers, while having a single binary, and having it be appealing to non-Rust developers.

MabezDev, 2021-09-23:
I love this! 

Another advantage is that its easier to spot breakages in other tools. For instance, a PR removing a 'unused' argument handling module broke `cargo flash` & `cargo embed`, a change that could have been fixed in the PR if it failed to build all the tools in CI.

I guess the only downside is for newbie Rust dev's its hard to tell what `cargo run` is doing, without prior knowledge of the `.cargo/config.toml` file.

MabezDev, 2021-09-23:
Possibly not an issue, but its worth discussing the name and usage vs [`probe-run`](https://github.com/knurling-rs/probe-run). I.e the difference between the two is _very minimal_ (`s/-/ /g`) and may be confusing.

Additionally, a lot of work has been put into [`probe-run`](https://github.com/knurling-rs/probe-run), perhaps we want to absorb some of that functionality in our `run` command.

noppej, 2021-09-26:
@Dirbaio Thank you for doing such a great job on writing this up. I believe you've addressed the major issues (that I can see), and I have nothing to add. 

@MabezDev makes a good point about `probe-run`, but I don't know how the [project owners](https://knurling.ferrous-systems.com/) would feel about that (plus we'd have to figure out what to do about funds from project sponsorships). Do you know any of the people/org involved in `probe-run`?

How do we move this to the next step? I'd be willing to dedicate some time to working on this.

jonathanpallant, 2021-09-28:
I understand the major feature of `probe-run` to be support for decoding their `defmt` wire protocol, so that embedded MCUs can 'print' formatted strings but have the formatting done in `probe-run` on the PC instead of on the embedded device. The name of their tool is unfortunate though.

noppej, 2022-02-21:
@Dirbaio , @Yatekii , @Tiwalun ... I noticed that there is some good work going on in the `probe-rs-cli`. I am thinking I'd like to remove the CLI interface from `probe-rs-debugger`.  Any objections?

Yatekii, 2022-02-21:
No, I think that is a good course of action :) Best would be if everything is integrated in probe-rs-cli, even tho that was initially intended as Tiwaluns hack-board :D

noppej, 2022-03-04:
I am going ahead and removing CLI from 'probe-rs-debugger'.  Henceforth, the only command line capabilities will be the `list` (list connected probes) and `list-chips` (list supported chips) that the other tools share. I left these in because it is kinda useful for people getting started with the debugger to be successful, without the need to install a different tool. 

As part of this process, I have noticed that a LOT of the actual logic around debugging, (e.g. flash, unwind, breakpoints, rtt, gdb), etc. are actually coded in other 'shared' parts of probe-rs (e.g. probe-rs::debug, probe-rs-cli-util, etc.  

Based on that, I am changing my opinion a bit about consolidating the tools. I now feel better about having a separate probe-rs-debugger, probe-rs-cli, cargo-embed, rtt-host, etc. If each of these are front-ends for custom use cases, and we are diligent about coding the heavy-lifting logic in shared modules (and not duplicating that part), then all we need is some good automated testing, and some good documentation to help users understand the different use cases.

There are a few potential benefits to keeping the front-ends as custom tools:
- Forces us to think about and refine the api's to be re-usable by multiple consumers. A lot of this work has already been done by early creators of probe-rs, but time and experience always reveals new 'opportunities' to improve.
- Hopefully it will help motivate investment in automated testing of the api's. This is a tricky problem to solve, but it may very well differentiate probe-rs from other embedded toolsets.
- Keeps the front-end tools for specific use cases nice and light (and isolated from changes in code for other use cases). Also makes it more attractive for the community to experiment with niche use cases/applications.

Just a thought :)

bjorn3, 2025-04-07:
What is the status of this? It took me quite a while to figure out that there was a way to show defmt log messages while at the same time having a debugger attached as I was only using the probe-rs command and didn't know cargo-embed existed and that it supports this functionality unlike the probe-rs command.

Yatekii, 2025-04-15:
This is already completed :) cargo embed and cargo flash are the same binary as probe-rs under the hood :) They are just different tools :)

closed 2025-04-15 by Yatekii

bjorn3, 2025-04-15:
This issue seemed like was also about unifying the exposed user interface, not just the executable files stored on the disk. So cargo embed and cargo flash would be subsumed entirely by probe-rs subcommands if my understanding was correct.