probe-rs/probe-rs #877: Tracking Issue: Unwinding support for RISCV

Tiwalun, 2021-11-10:
Track the progress until unwinding support for RISCV is on the same level as for ARM. Further work can be done to improve unwinding in general, this issue only tracks tasks which relate to problems where ARM specific assumptions are currently hard-coded.

## Open issues
- [ ] Proper register names in debugger, currently ARM registers are hardcoded https://github.com/probe-rs/probe-rs/blob/5e5a4307ef8f1fee2f09da63a2d85b0720471182/debugger/src/debug_adapter.rs#L740
- [ ] Use architecture-specific register values when unwinding https://github.com/probe-rs/probe-rs/blob/5e5a4307ef8f1fee2f09da63a2d85b0720471182/probe-rs/src/debug/mod.rs#L387

noppej, 2021-11-12:
I'd like to add one please ...  
- [ ]  Create architecture specific `capstone` in the debugger. Currently it assumes `arm` 
https://github.com/probe-rs/probe-rs/blob/545267b0065e5242a84826dc3de2dbd6e6dd4fd3/debugger/src/debugger.rs#L320-L324

Tiwalun, 2021-11-12:
Thanks, I've added it.

Feel free to edit the issue directly next time, then the progress is tracked.

MabezDev, 2021-11-17:
Perhaps not specific to RISCV, but perhaps the unwinder could be more robust when debug info is missing for a particular PC (i.e, ROM code that doesn't have debug symbols). At the moment, the `StackFrameIterator` returns `None` which stops the unwinding, maybe we could add `Unknown frame @ 0xDEADBEEF` and continue the unwinding.

noppej, 2021-11-18:
> Perhaps not specific to RISCV, but perhaps the unwinder could be more robust when debug info is missing for a particular PC (i.e, ROM code that doesn't have debug symbols). At the moment, the [...]

@MabezDev ... This should be easy enough to do, and I'm willing to take it on. However, it really doesn't  belong in this Issue, so maybe you can create a new one (enhancement)? It would help me if you could give a bit of an explanation of how the `Unknown frame @ 0xDEADBEEF` (yes, I see what you did there ... lol), would be useful in a debug situation.

MabezDev, 2021-11-18:
I've opened https://github.com/probe-rs/probe-rs/issues/896 :)

noppej, 2022-09-15:
Closing this, now that all the known items have been implemented, and debug functionality is now considered on-par with Arm.

I suggest we treat new discoveries as independent issues.

closed 2022-09-15 by noppej

MabezDev, 2022-09-15:
Thank you @Tiwalun & @noppej for all your hard work on this! It's very pleasant to use probe-rs on RISCV cores now :)

Yatekii, 2022-09-15:
Oh man, this is awesome! Thanks so much!