probe-rs/probe-rs #883: Flashing of STM32L071KBTx with J-Link broken with 0.11.0 but works with 0.10.1

rnestler, 2021-11-11:
**Describe the bug**

When I try to flash my binary with `cargo embed` 0.11.0 my chip just hangs indefinitely. If I do the same with 0.10.1 it works. I also tried latest master which fails as well

**To Reproduce**
Steps to reproduce the behavior:
1. Flash with cargo embed 0.11.0
2. Broken

**Expected behavior**
It should work with 0.11.0 as well

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - Linux (ArchLinux)

**Additional context**

J-Link:
```
J-Link: Firmware version: J-Link V9 compiled May  7 2021 16:26:12
J-Link: Hardware version: J-Link 9.40.0
```

rnestler, 2021-11-11:
> Maybe I can drop by the coredump sometime? ;)

Sure thing :slightly_smiling_face:

berkowski, 2022-02-28:
I'm having the same issue as @rnestler with a STM32L071RTZx and disabling double buffering per his patch seems to fix it for me as well.  This problem is forcing me to stick with `defmt` 0.2 and older versions of `cargo-flash` and `probe-run` to monitor RTT traffic.

If disabling double-buffering is a fix at the cost of speed could it be made an option for upstream users to opt out of it?

Yatekii, 2022-02-28:
Yeah so initially we had a flag in the target yamls that specified if a chip would support double buffering. we could reintroduce that. Double buffering only gives like 20% perf for most chips.

Maybe we have an issue in the calculation of required RAM. Maybe we are too stingy and the algo actually requires more RAM. We should also check if that would solve the issue.

berkowski, 2022-02-28:
Is it possible to have it as a library option so that upstream crates like `cargo-flash` could add a command-line argument like `--disable-double-buffering` that would flip the switch?  Having it set in a YAML file seems great for platforms where support is known, and seems like it's the easiest fix but if there's a bug in that file then we'd be back in this same situation.

I'd be happy to take a swing at a feature PR with some direction where to start.

Yatekii, 2022-02-28:
Sure, I would very much accept that :) I think initially we even had a flag for that internally. Could easily be on the flash options :)

bugadani, 2024-03-30:
Can someone please retest this issue with the current master branch? There were a number of changes in the last month or two alone that could make a difference, and the last activity here is quite old.

rnestler, 2024-04-02:
This is quite long ago, I'm not even sure anymore on which project I encountered this...