probe-rs/probe-rs #896: Make stack unwinding more robust

MabezDev, 2021-11-18:
**Is your feature request related to a problem? Please describe.**
When unwinding the stack, if a given PC does not have any debug info associated with it [the unwinder will stop](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/debug/mod.rs#L393-L403) regardless of whether there are more frames to unwind.

**Describe the solution you'd like**
The unwinder should report that the frame isn't decodable, then continue unwinding.

Tiwalun, 2021-11-19:
@MabezDev Do you know how the frame pointer is handled for RISCV targets? For the thumbv6 / thumbv7 targets, the framepointer is explicitly enabled in the rustc target spec:

```json
{
  "abi": "eabi",
  "arch": "arm",
  "c-enum-min-bits": 8,
  "data-layout": "e-m:e-p:32:32-Fi8-i64:64-v128:64:128-a:0:32-n32-S64",
  "emit-debug-gdb-scripts": false,
  "executables": true,
  "frame-pointer": "always",
  "is-builtin": true,
  "linker": "rust-lld",
  "linker-flavor": "ld.lld",
  "llvm-target": "thumbv7em-none-eabi",
  "max-atomic-width": 32,
  "panic-strategy": "abort",
  "relocation-model": "static",
  "target-pointer-width": "32"
}
```

For the RISCV targets this doesn't seem to be the case. If the frame pointer gets optimized away by rustc / LLVM, we will need the DWARF information to do unwinding, so I don't think we can improve our behaviour in that case. Of course, frame pointers can always be [enabled manually](https://doc.rust-lang.org/rustc/codegen-options/index.html#force-frame-pointers).

MabezDev, 2021-11-19:
This wasn't always the case for thumb targets, see this PR: https://github.com/rust-lang/rust/pull/69248. Perhaps we should PR something similar for the bare metal RISCV targets.

MabezDev, 2021-11-19:
Oh actually it used to the case for RISCV too, https://github.com/rust-lang/rust/pull/61675, but it was removed via https://github.com/rust-lang/rust/pull/69890.

I guess to improve stack unwinding we can note that frame pointer elimination should be disabled. However my original reason for opening this is debugging whilst calling into binary blobs, it should be possible to continue unwinding even if no debug info is available for that function.

noppej, 2022-01-19:
@MabezDev I implemented a fix that will unwind as long as I can determine a valid LR and PC, and then if no other information is available, will at least show you the Register values for that frame.   Are you able to try this with the new branches I mention in https://github.com/probe-rs/vscode/issues/26#issuecomment-1016870363, and let me know how you would like to enhance based on that solution?

MabezDev, 2022-01-24:
FYI this is what the same call stack looks like with openocd:

[image]

noppej, 2022-01-25:
@MabezDev Thank you for taking the time to test this. I will continue to work on this and keep you posted.

noppej, 2022-01-25:
@MabezDev A quick question. Do you like seeing the namespace name as part of the function name in the call stack (as per your openocd example)? Or do you prefer just the function name (as in the probe-rs example)?

MabezDev, 2022-01-25:
Namespace can be useful at a glance, but it's not strictly necessary as it's possible to figure out where it comes from with the filename and line number. The `@` following by address is quite handy though.

noppej, 2022-06-28:
@MabezDev ... I've just created a draft PR #1147 with my first attempt to resolve this.  The logic I used was as follows:

We cannot do stack unwinding if we do not have debug info. However, there is one case where we can continue. When the following conditions are met:
1. The current frame is the first frame in the stack, AND ...
2. The frame registers have a valid return address/LR value.

If both these conditions are met, we can push the 'unknown function' to the list of stack frames, and use the LR value to calculate the PC for the calling frame.
- The current logic will then use that PC to get the next frame's unwind info, and if that exists, will be able to continue unwinding.
- If the calling frame has no debug info, then the unwinding will end with that frame.

This works for my test case, but I'd really appreciate if you can validate that it works for you too.

noppej, 2022-06-29:
Had confirmation from @MabezDev that this now works, so will close this with PR #1147 

[image]

closed 2022-07-08 by bors[bot]